raspberrypi/linux #2167: Support for AARCH64

codekatana, 2017-08-18:
Hi, 
I would like to know if there exists any plan to support 64-bit kernel on Raspberry Pi 3?
I found that despite having Cortex A53, current kernel + userland is 32-bit.

closed 2017-08-18 by codekatana

pelwell, 2017-08-18:
This is a VFAQ. We are reluctant to add a third kernel build option to the existing v6 and v7 builds, so until somebody provides us evidence of a clear benefit (which they haven't) we won't be supporting a 64-bit kernel. Adding an entire 64-bit Raspbian userspace is a massive undertaking and very unlikely in the near future.

codekatana, 2017-08-18:
I am so sorry to have raised this issue here. Apologies, I should have posted on Forum instead. However, 64-bit support is truly for very specific use-case and given the current hardware state and application areas of R-PI3, I agree that 32 bit is enough.

I, however, want to experiment out with 64 bit kernel (just for fun)! Hence would try out unofficially. 

Thank you for your clarifications.