raspberrypi/linux #2168: Is the hauppauge wintv dualhd fully supported (both tuners)?

AlBundy33, 2017-08-19:
It seems that there are some patches to get both tuners to work. 
https://patchwork.kernel.org/patch/8741651/

AlBundy33, 2017-08-20:
Found this #2091 - does this mean that the tuner is currently not supported or only the second tuner is not supported?

JamesH65, 2017-08-20:
We don't tend to give support for specific third party peripherals, and I
don't believe we have this particular device in house to test. @Popcornmix
may have some ideas as he works with Kodi.

Google is probably your best bet to see if anyone has tried this setup.

6by9, 2017-08-20:
There are the sets of patches you linked to kicking around.
The first is merged in 4.11 to support a single tuner on the WinTV-dualHD - you're free to build that tree from here, or IIRC OpenElec is using a more recent kernel so may already be suitable.
The second set of patches have not been pushed for review upstream, so as pelwell and popcornmix state on the PR, it's unlikely they'll get merged into the Pi tree.

Having said that, taking the two sets of patches does seem to give a system with WinTV-dualHD working (I have a pair of them that seem to work OK with tvheadend and the Pi 4.9 kernel. I'm currently having issues with it on Ubuntu 16.04 and rebuilding top-of-tree media_build tree)

6by9, 2017-09-13:
To relay info on what's happening upstream - http://www.spinics.net/lists/linux-media/msg121493.html

My experience is that I have a pair of them on an Ubuntu 17.04 HP Microserver now for about a fortnight, and it's been stable.
Whilst I had them running on the Pi, tvheadend was misbehaving. However I think that was down to trying to save everything over NFS to the NAS, and it was spinning up the RAID array too slowly. I might have another play at a later date using local storage.

JamesH65, 2017-12-04:
@6by9 Any thing/progress to report on this?

JamesH65, 2018-06-27:
@6by9 Did this ever make it through upstreaming?

6by9, 2018-06-27:
It's all upstream - can't recall which release it will have hit, although there have been a number of followups.

#2434 was me attempting to backport, but whilst early revisions worked the current state is broken, and I haven't had the energy to go messing in the evenings (no TV aerial in the office that I'm aware of).

closed 2024-10-31 by AlBundy33