raspberrypi/linux #2172: i2s clock starts before data is ready

Gadgetoid, 2017-08-23:
I've been investigating the oft-reported issue of pops and clicks when using our various DAC+Speaker products with the i2s audio output on the Pi.

I have boiled it down to what I believe is an issue with the driver itself, which is prone to leaving too much time between initialising the i2s clock and actually outputting audio data.

My reading of the i2s bus specification (https://www.sparkfun.com/datasheets/BreakoutBoards/I2SBUS.pdf) may be incorrect, but it seems to suggest a delay between clock start and data of ~300ns, whereas the delay on the Pi can be anything from 1ms (when playing a wav file using aplay directly to the audio device) and 860ms+ (when playing an online audio stream through PulseAudio from Mopidy).

This delay appears to be the cause of the pops, since the clock is started with no valid audio data, but I'm unsure if it's something that can be fixed in the driver level. Intuitively I'd expect the driver not to start the clock until the data is ready to follow, but in this case the clock seems to start the instant the driver is initialised.

Is there any way to change this behaviour? Either through configuration, or from sane tweaks to the driver itself? I'm fairly confident, from some crude testing, that closing the gap between clock and data start would reduce or eliminate popping.

An alternate solution would be to start the i2s clock at boot, and leave it continuously running and untouched by the i2s audio driver. Is this possible? This would cause the same issue, of course, but only once upon boot, and not during any subsequent track changes or audio stops/starts.

Thank you.

pelwell, 2017-08-24:
When you say "the clock is started", is that just SCLK or also WS? Without WS you can't know where the sample starts, so the sample doesn't start until WS changes.

I don't read any part of the spec as saying you have to deliver data within a certain interval of the clock starting, rather it sets limits on timing to ensure correct operation when delivering data. Can you be more specific about the bit of the spec you are referring to?

P33M, 2017-08-24:
The timings in the datasheet are unrelated to the issue you are experiencing. They refer to the position of data with respect to clock edges.

Turning the question on its head, what delay is necessary for slave-mode codecs to initialise internal state (DAC outputs, unmute functions) when the clock starts prior to outputting the first audio sample in a stream?

P33M, 2017-08-24:
e.g.: page 12 of this datasheet: http://www.ti.com/lit/ds/symlink/pcm1748.pdf

Specifies that the internal logic will remain in reset for 1024 SCLK clocks after power-on reset.

Gadgetoid, 2017-08-25:
> Turning the question on its head, what delay is necessary for slave-mode codecs to initialise internal state (DAC outputs, unmute functions) when the clock starts prior to outputting the first audio sample in a stream?

That's a fair point, and it's possible I could be barking up the wrong tree and my reading of the datasheet is horribly wrong.

To take another approach- one that doesn't implicate the driver which, frankly, is somewhat at the fringes of my level of understanding- I've been chasing the sourcecode up the chain in an attempt to find out how I might circumvent or mitigate the problem without resorting to hardware changes.

I've found that running Pulse as a system service (something which is quite expressly caveated and warned against) with a very long idle timeout will fix the problem by keeping the clock running- albeit rather inelegantly. Is it possible to, instead, spin up the clock and leave it running as if it were an external clock reference?

My limited understanding of the code seems to suggest the driver supports this, but I can't figure out how to take advantage of it.

Thank you for your responses!

closed 2017-09-06 by Gadgetoid