raspberrypi/linux #2175: MCP2515 Hangs When Set to IRQF_TRIGGER_FALLING

Chris-W-Weir, 2017-08-24:
https://github.com/raspberrypi/linux/blob/dbe9b03bef2b949093b4393d5f4efd8ef5cdb772/drivers/net/can/spi/mcp251x.c#L939

I'm working with a Raspberry Pi 3 and a PiCAN Duo, and my can0/can1 interfaces will hang after a period of time ranging from 30 seconds to 15 minutes. 

I hooked up a logic analyzer to the MCP2515 and saw that the interrupt line was getting stuck low (active) and the can interface would resume if the line was pulled high again. I suspect that the falling edge interrupt is being missed, though I don't understand why.

I changed line 939 to be IRQF_TRIGGER_LOW (level triggered low) and the issue has gone away. I haven't had a chance to test if this causes the CAN interface to miss messages or not, but it seems to be working as far as I can tell.

If the falling edge trigger is required for some reason, or I'm doing something foolish, please let me know.

Chris

pelwell, 2017-08-24:
This is a duplicate of https://github.com/raspberrypi/linux/issues/1490, a long-standing issue for which up to now we don't have a fix. I'd like to read up a bit to verify that your change is safe, but I would always prefer a level-sensitive interrupt to an edge-sensitive one for this reason, so I'm hopeful that we may be able to close both issues soon thanks to your suggestion.

Chris-W-Weir, 2017-08-24:
I have a 4 channel MCP2515 board I built using SPI 0.0, 0.1, 1.0, and 1.1, which stresses the Pi further, let me know if I can be of any assistance with research into the issue.

mypiandrew, 2017-11-13:
Hi

I'm busy debugging my dual can adapter and seeing some weird initialization side effects of using the flip-flip gate array that albertodesouza provided.

I much prefer this approach 

If we're not sure about making a sweeping change why don't we add a module option to this to specify whether you want a edge or level trigger and default it to the edge as standard?

I'm not the greatest of C coders, but I can hack together a patch if that helps move things forward 

It would be nice to clear this up before the holiday season.

Chris-W-Weir, 2017-11-13:
I've seen on other overlays the trigger type as an option, but I'm not sufficiently familiar with how they work to implement it myself.

mypiandrew, 2017-11-13:
Hey Chris,

Did you have any problems with CAN interfaces not automatically coming up (even if respective entries are there /etc/network/interfaces) or issues with CAN devices not initializing when rebooting via command line.

Got a weird bug and it's not making a lot of sense :(

Chris-W-Weir, 2017-11-13:
I typically do

```
ip link set can0 type can bitrate <bitrate>
ip link set can0 up
```

in a script when I want to bring the interface. I haven't tried to have them automatically come up via other methods. What sort of hardware are you using?

mypiandrew, 2017-11-13:
OK, so this might look like I'm hijacking your thread, sorry!

It's my own (4 layer) HAT board, I've had a single can card work fine with no problems (that one always works as expected)

I've used two MCP2515s with can0 wired to SPI0-CE0/GPIO25 and can1 to SPI0-CE1/GPIO26

The interrupt GPIO lines go through the flip-flop logic gate array as per #1490 that forces the interrupts to act synchronously, which works really well and I've not managed to fault it in that regard.

This is running with Raspbian Jessie 

```
root@raspberrypi:~# uname -a
Linux raspberrypi 4.9.61-v7+ #1049 SMP Fri Nov 10 15:32:51 GMT 2017 armv7l GNU/Linux
```

I'm using this in config.txt
```
dtparam=spi=on
dtoverlay=spi0-hw-cs
dtoverlay=mcp2515-can1-overlay,oscillator=16000000,interrupt=26
dtoverlay=mcp2515-can0-overlay,oscillator=16000000,interrupt=25
```

And these to /etc/network/interfaces  :

```
auto can0
iface can0 inet manual
    pre-up ip link set $IFACE type can bitrate 500000 listen-only off
    up /sbin/ifconfig $IFACE up
    down /sbin/ifconfig $IFACE down

auto can1
iface can1 inet manual
    pre-up ip link set $IFACE type can bitrate 500000 listen-only off
    up /sbin/ifconfig $IFACE up
    down /sbin/ifconfig $IFACE down

```

From a cold boot (or hard reset) I *always* get :

```
[root@raspberrypi:~# dmesg 
<snip>
[    5.192646] CAN device driver interface
[    5.212054] mcp251x spi0.0 can0: MCP2515 successfully initialized.
[    5.229011] mcp251x spi0.1 can1: MCP2515 successfully initialized.
<snip>
```

The can interfaces don't auto start (which is weird as the single can card I have always does) but once I've done ifup can0  and ifup can1 the system works fine.

I've done testing using cangen's normal and high load test facilities by wiring up two cards (one dual can card doing Tx and the other doing both Rx) and running for hours without freezing up.

The problem I see is when I have a situation where I've brought up either/both interfaces and then/sbin/reboot is run. What happens is on the next reboot I get one of the below in dmesg :

```
[    4.903004] mcp251x spi0.0 can0: MCP2515 successfully initialized.
[    4.914614] mcp251x spi0.1: Cannot initialize MCP2515. Wrong wiring?
[    4.914645] mcp251x spi0.1: Probe failed, err=19
```

or 

```
[    4.992682] mcp251x spi0.0: Cannot initialize MCP2515. Wrong wiring?
[    4.992713] mcp251x spi0.0: Probe failed, err=19
[    5.003091] mcp251x spi0.1: Cannot initialize MCP2515. Wrong wiring?
[    5.003116] mcp251x spi0.1: Probe failed, err=19
```

or

```
[    4.847508] mcp251x spi0.0: Cannot initialize MCP2515. Wrong wiring?
[    4.847543] mcp251x spi0.0: Probe failed, err=19
[    4.859358] mcp251x spi0.1 can0: MCP2515 successfully initialized.
```

As I say this does not happen at initial power up or if I do a hardware reset (reset pins shorted)

if when I see this I reboot the system again via the command line it comes good again.

if I do the below in the fault mode I can sort of fix it :

```
root@raspberrypi:~# rmmod spi_bcm2835
root@raspberrypi:~# modprobe spi_bcm2835
root@raspberrypi:~# dmesg 
<snip>
[    4.992682] mcp251x spi0.0: Cannot initialize MCP2515. Wrong wiring?
[    4.992713] mcp251x spi0.0: Probe failed, err=19
[    5.003091] mcp251x spi0.1: Cannot initialize MCP2515. Wrong wiring?
[    5.003116] mcp251x spi0.1: Probe failed, err=19
[    5.006362] sd 0:0:0:0: Attached scsi generic sg0 type 0
[    5.071467] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[    5.426320] systemd-journald[146]: Received request to flush runtime journal from PID 1
[    6.100255] Adding 102396k swap on /var/swap.  Priority:-1 extents:2 across:110588k SSFS
[    6.471973] smsc95xx 1-1.1.1:1.0 eth0: hardware isn't capable of remote wakeup
[    6.472336] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    8.050410] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[    8.051578] smsc95xx 1-1.1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xC1E1
[   15.081718] random: crng init done
[   93.414869] mcp251x spi0.0 can0: MCP2515 successfully initialized.
[   93.427267] mcp251x spi0.1 can1: MCP2515 successfully initialized.
```

but then you run into some problems with ifup initially :

```
root@raspberrypi:~# ifup can0
SIOCSIFFLAGS: No such device
Failed to bring up can0.
root@raspberrypi:~# ifup can0
root@raspberrypi:~# ifup can1
SIOCSIFFLAGS: No such device
Failed to bring up can1.
root@raspberrypi:~# ifup can1
root@raspberrypi:~# ifconfig
can0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          UP RUNNING NOARP  MTU:16  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:10
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

can1      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          UP RUNNING NOARP  MTU:16  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:10
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
<snip>
```

I though this might be to do with the logic gate, but I've tried removing that this afternoon and wiring it straight across but that's not made any difference to the reboot problem.

I will try tomorrow removing the second MCP2515 chip and see if I can get the card to act "normal" like the single can card.

If you have any insights on this I would welcome them as I'm tearing my hair out with this...

pelwell, 2017-11-13:
Perhaps this forum thread may help with the initialisation problem: https://www.raspberrypi.org/forums/viewtopic.php?t=190868

mypiandrew, 2017-11-13:
Hi Phil

Thanks for the swift attention, much appreciated

Sadly I have tried each of the below and it's not made any difference, first time is good then subsequent soft reboots give intermittent results

```
dtoverlay=mcp2515-can0,oscillator=16000000,interrupt=25,spimaxfrequency=500000
dtoverlay=mcp2515-can1,oscillator=16000000,interrupt=26,spimaxfrequency=500000

```
```
dtoverlay=mcp2515-can0,oscillator=16000000,interrupt=25,spimaxfrequency=50000
dtoverlay=mcp2515-can1,oscillator=16000000,interrupt=26,spimaxfrequency=50000

```
```
dtoverlay=mcp2515-can0,oscillator=16000000,interrupt=25,spimaxfrequency=10000
dtoverlay=mcp2515-can1,oscillator=16000000,interrupt=26,spimaxfrequency=10000
```

I'll lift the chip attached to CE1 tomorrow and see if that gives any more clues

pelwell, 2017-11-14:
See https://github.com/raspberrypi/linux/pull/2267.

mypiandrew, 2017-11-15:
Just to update the issue with initialization above was a crystal issue, swapped them for the usual type I use and the interfaces came up as expected. 

Happy to test the patch referred to above if you can point me on how to do this?

pelwell, 2017-11-15:
There are numerous ways to download a pull request, but the simplest is to download the raw patch and apply it:
```bash
$ wget https://patch-diff.githubusercontent.com/raw/raspberrypi/linux/pull/2267.patch
$ git am 2267.patch
```
Then build and install the kernel as normal. (Note: this will apply the patch to your current branch rather than the branch the PR was submitted against, but as long as you're on 4.9 the difference should be academic).

JamesH65, 2018-06-27:
@pelwell Looks like this was fixed?

closed 2018-06-27 by pelwell

EmanuelHalllgren, 2022-01-17:
> https://github.com/raspberrypi/linux/blob/dbe9b03bef2b949093b4393d5f4efd8ef5cdb772/drivers/net/can/spi/mcp251x.c#L939
> 
> I'm working with a Raspberry Pi 3 and a PiCAN Duo, and my can0/can1 interfaces will hang after a period of time ranging from 30 seconds to 15 minutes.
> 
> I hooked up a logic analyzer to the MCP2515 and saw that the interrupt line was getting stuck low (active) and the can interface would resume if the line was pulled high again. I suspect that the falling edge interrupt is being missed, though I don't understand why.
> 
> I changed line 939 to be
> 
> ```c
> unsigned long flags = IRQF_ONESHOT | IRQF_TRIGGER_LOW;
> ```
> 
> (level triggered low) and the issue has gone away. I haven't had a chance to test if this causes the CAN interface to miss messages or not, but it seems to be working as far as I can tell.
> 
> If the falling edge trigger is required for some reason, or I'm doing something foolish, please let me know.
> 
> Chris

Anyone how to do the same trick in PGPIO (https://abyz.me.uk/rpi/pigpio/cif.html) My pin gets randomly stuck low. Sometimes I can run my CAN-bus for 2 hours. Sometimes 1 minute.