raspberrypi/linux #2177: Could nf_tables be enabled in raspbian

github-user-1, 2017-08-27:
Hello all,
the module for nf_tables (and it's possible dependencies) is not loaded in the latest Raspbian Stretch. I thought it would be (based on the announcement that it will become available with Debian Stretch). Since netfiltering should be moved from iptables etc. over to nftables in the long run I thought this would be a good start now with Stretch being released.

Are there any plans to enable nft (nftables) by standard for Raspbian as well ?

popcornmix, 2017-08-27:
To be clear, are you requesting ``CONFIG_NF_TABLES=m`` be added to kernel config?

github-user-1, 2017-08-27:
Probably yes - however I consider myself rather an end-user of linux (gnu cli rather). So I don't know much about the in and outs of the linux kernel used for raspbian but I was simply wondering why nft wasn't working with Raspbian Stretch (as it was officially announced for Debian Stretch to become the replacement of iptables etc.). Looking for the reason why nft didn't work after I manually installed it from the repo, I can see e.g. that the modul is not loaded:
```
$ modinfo nf_tables
modinfo: ERROR: Module nf_tables not found.
```
whereas iptables is for example:
```
 $ modinfo ip_tables
filename:       /lib/modules/4.9.41-v7+/kernel/net/ipv4/netfilter/ip_tables.ko
description:    IPv4 packet filter
author:         Netfilter Core Team <coreteam@netfilter.org>
license:        GPL
srcversion:     073F6A28550B59E4A9A7F74
depends:        x_tables
intree:         Y
vermagic:       4.9.41-v7+ SMP mod_unload modversions ARMv7 p2v8
```

Ferroin, 2017-08-28:
For nft to actually be usable will require more than just CONFIG_NF_TABLES=m.  Ideally, most of the options under that in the config menu should be enabled as well, otherwise you can't really do much with nft.

github-user-1, 2017-08-30:
Just as a start and to provide some hints what might be required in the kernel config (copied from https://home.regit.org/netfilter-en/nftables-quick-howto/):

`$ make oldconfig

Netfilter Xtables support (required for ip_tables) (NETFILTER_XTABLES) [M/y/?] m
Netfilter nf_tables support (NF_TABLES) [N/m] (NEW) m
  Netfilter nf_tables payload module (NFT_PAYLOAD) [N/m] (NEW) m
  Netfilter nf_tables IPv6 exthdr module (NFT_EXTHDR) [N/m] (NEW) m
  Netfilter nf_tables meta module (NFT_META) [N/m] (NEW) m
  Netfilter nf_tables conntrack module (NFT_CT) [N/m] (NEW) m
  Netfilter nf_tables rbtree set module (NFT_RBTREE) [N/m] (NEW) m
  Netfilter nf_tables hash set module (NFT_HASH) [N/m] (NEW) m
  Netfilter nf_tables counter module (NFT_COUNTER) [N/m] (NEW) m
  Netfilter nf_tables log module (NFT_LOG) [N/m] (NEW) m
  Netfilter nf_tables limit module (NFT_LIMIT) [N/m] (NEW) m
  Netfilter nf_tables nat module (NFT_NAT) [N/m] (NEW) m
  Netfilter x_tables over nf_tables module (NFT_COMPAT) [N/m/?] (NEW) m

IPv4 nf_tables support (NF_TABLES_IPV4) [N/m] (NEW) m
  nf_tables IPv4 reject support (NFT_REJECT_IPV4) [N/m] (NEW) m
  IPv4 nf_tables route chain support (NFT_CHAIN_ROUTE_IPV4) [N/m] (NEW) m
  IPv4 nf_tables nat chain support (NFT_CHAIN_NAT_IPV4) [N/m] (NEW) m

IPv6 nf_tables support (NF_TABLES_IPV6) [M/n] m
  IPv6 nf_tables route chain support (NFT_CHAIN_ROUTE_IPV6) [M/n] m
  IPv6 nf_tables nat chain support (NFT_CHAIN_NAT_IPV6) [M/n] m

Ethernet Bridge nf_tables support (NF_TABLES_BRIDGE) [N/m/y] (NEW) m`

fabianfrz, 2017-09-01:
I experience the same issue. All calls fail. For example:
```
nft list ruleset
internal:0:0-0: Error: Could not receive tables from kernel: Invalid argument
```

netcaf, 2017-11-03:
I have the same issue when I try nftables.

ageru, 2017-11-14:
Hi,

I seem to have a similar issue with iptables/netfilter functionalities not working due to missing kernel flags. In addition to the above flags, my need would be mainly for _Netfilter Xtables support > "recent" match support_. It's useful to pick and drop brute-force attempts.

Many thanks.

miwagner1, 2017-11-25:
I hope this gets added in soon, I really despise using iptables and really love using nftables after using Cisco IOS for some time. The syntax is similar which makes it easer for me to alternate between them.

JamesH65, 2017-12-05:
We are wary of adding large amount of modules to the system, without knowing the impact. So has anyone  found out how much storage is taken up, how much the kernel bloats when it is added, and whether there are performance impacts?

fabianfrz, 2017-12-05:
```
find . -name "*.ko" -exec ls -l {} \; | grep nft 
-rw-r--r-- 1 myuser users 6640  5. Dez 18:32 ./net/ipv6/netfilter/nft_chain_nat_ipv6.ko
-rw-r--r-- 1 myuser users 16832  5. Dez 18:32 ./net/netfilter/nft_ct.ko
-rw-r--r-- 1 myuser users 9704  5. Dez 18:32 ./net/netfilter/nft_set_rbtree.ko
-rw-r--r-- 1 myuser users 6240  5. Dez 18:32 ./net/netfilter/nft_fwd_netdev.ko
-rw-r--r-- 1 myuser users 7896  5. Dez 18:32 ./net/netfilter/nft_rt.ko
-rw-r--r-- 1 myuser users 8264  5. Dez 18:32 ./net/netfilter/nft_set_bitmap.ko
-rw-r--r-- 1 myuser users 6640  5. Dez 18:32 ./net/netfilter/nft_masq.ko
-rw-r--r-- 1 myuser users 27376  5. Dez 18:32 ./net/netfilter/nft_set_hash.ko
-rw-r--r-- 1 myuser users 6240  5. Dez 18:32 ./net/netfilter/nft_dup_netdev.ko
-rw-r--r-- 1 myuser users 11632  5. Dez 18:32 ./net/netfilter/nft_exthdr.ko
-rw-r--r-- 1 myuser users 17456  5. Dez 18:32 ./net/netfilter/nft_meta.ko
-rw-r--r-- 1 myuser users 8648  5. Dez 18:32 ./net/netfilter/nft_nat.ko
-rw-r--r-- 1 myuser users 6504  5. Dez 18:32 ./net/ipv4/netfilter/nft_chain_nat_ipv4.ko
-rw-r--r-- 1 myuser users 6800  5. Dez 18:32 ./net/ipv4/netfilter/nft_masq_ipv4.ko
```

Total: around 143 kB

I probably have not compiled everything in but as you can see, it is not too large.

fabianfrz, 2017-12-05:
Ok, Forgot about the nf_tables modules:
```
-rw-r--r-- 1 myuser users 9704  5. Dez 18:32 ./net/ipv6/netfilter/nf_tables_ipv6.ko
-rw-r--r-- 1 myuser users 7824  5. Dez 18:32 ./net/netfilter/nf_tables_inet.ko
-rw-r--r-- 1 myuser users 10344  5. Dez 18:32 ./net/netfilter/nf_tables_netdev.ko
-rw-r--r-- 1 myuser users 132352  5. Dez 18:32 ./net/netfilter/nf_tables.ko
-rw-r--r-- 1 myuser users 9216  5. Dez 18:32 ./net/ipv4/netfilter/nf_tables_ipv4.ko
```

Total: 165 kB

So the total of both: 308 kB

sceext2, 2017-12-31:
How to compile and install these kernel modules to make nftables work ?

sceext2, 2018-01-01:
Follow this guide: <http://lostindetails.com/blog/post/Compiling-a-kernel-module-for-the-raspberry-pi-2>
and this: <https://www.raspberrypi.org/documentation/linux/kernel/building.md>

Download build tools: <https://github.com/raspberrypi/tools>
and kernel source: <https://github.com/raspberrypi/linux/releases/tag/raspberrypi-kernel_1.20171029-1>

<hr />

Then I cross-compile a kernel and install and run it.

I'm running `raspbian` on a `Raspberry pi 3B`. This is the kernel I built:
```
$ uname -a
Linux sceext-p3-201712 4.9.59-v7 #3 SMP Mon Jan 1 11:36:00 CST 2018 armv7l GNU/Linux
```

<hr />

Before build the kernel, use `make ARCH=arm CROSS_COMPILE=$CCPREFIX menuconfig`  to add some modules for `nftables`.

Then nftables works fine:
```
$  sudo nft list ruleset
table ip nat {
	chain nat-dnat {
		type nat hook prerouting priority 0; policy accept;
	}

	chain nat-snat {
		type nat hook postrouting priority 100; policy accept;
		ip saddr 10.150.0.0/16 oifname "eth1" masquerade
		ip saddr 10.151.0.0/16 oifname "eth1" masquerade
	}
}
table ip vadsll {
	chain out {
		type filter hook postrouting priority 150; policy accept;
		ip daddr 0.0.0.0 oifname "eth1" accept
		ip daddr 10.0.0.0/8 oifname "eth1" accept
		ip daddr 172.16.0.0/12 oifname "eth1" accept
		ip daddr 192.168.0.0/16 oifname "eth1" accept
		ip daddr 255.255.255.255 oifname "eth1" accept
		oifname "eth1" counter packets 112957 bytes 9520267 queue num 44001 bypass
	}
}
$  lsmod | grep nf
nfnetlink_queue        12450  1
nft_queue               3538  1
nft_counter             2633  1
nft_masq_ipv4           1393  2
nf_nat_masquerade_ipv4     3123  1 nft_masq_ipv4
nft_masq                2172  1 nft_masq_ipv4
nft_meta                6536  8
nft_chain_nat_ipv4      1722  2
nf_conntrack_ipv4       8890  1
nf_defrag_ipv4          1824  1 nf_conntrack_ipv4
nf_nat_ipv4             6071  1 nft_chain_nat_ipv4
nf_nat                 18854  2 nf_nat_masquerade_ipv4,nf_nat_ipv4
nf_conntrack          104634  4 nf_conntrack_ipv4,nf_nat_masquerade_ipv4,nf_nat_ipv4,nf_nat
nf_tables_ipv4          2363  3
nf_tables              70146  41 nft_chain_nat_ipv4,nf_tables_ipv4,nft_masq,nft_queue,nft_meta,nft_masq_ipv4,nft_counter
nfnetlink               7322  3 nfnetlink_queue,nf_tables
$ modinfo nf_tables
filename:       /lib/modules/4.9.59-v7/kernel/net/netfilter/nf_tables.ko
alias:          nfnetlink-subsys-10
author:         Patrick McHardy <kaber@trash.net>
license:        GPL
srcversion:     3A2EAEF39C36D06566983D6
depends:        nfnetlink
intree:         Y
vermagic:       4.9.59-v7 SMP mod_unload modversions ARMv7 p2v8 
```

<hr />

Here is my kernel config:
```
$ diff -u old_config .config
--- old_config	2018-01-01 18:45:22.190685726 +0800
+++ .config	2018-01-01 18:45:33.431083373 +0800
@@ -829,7 +829,30 @@
 CONFIG_NF_NAT_SIP=m
 CONFIG_NF_NAT_TFTP=m
 CONFIG_NF_NAT_REDIRECT=m
-# CONFIG_NF_TABLES is not set
+CONFIG_NF_TABLES=m
+CONFIG_NF_TABLES_INET=m
+CONFIG_NF_TABLES_NETDEV=m
+CONFIG_NFT_EXTHDR=m
+CONFIG_NFT_META=m
+CONFIG_NFT_NUMGEN=m
+CONFIG_NFT_CT=m
+CONFIG_NFT_SET_RBTREE=m
+CONFIG_NFT_SET_HASH=m
+CONFIG_NFT_COUNTER=m
+CONFIG_NFT_LOG=m
+CONFIG_NFT_LIMIT=m
+CONFIG_NFT_MASQ=m
+CONFIG_NFT_REDIR=m
+CONFIG_NFT_NAT=m
+CONFIG_NFT_QUEUE=m
+CONFIG_NFT_QUOTA=m
+CONFIG_NFT_REJECT=m
+CONFIG_NFT_REJECT_INET=m
+CONFIG_NFT_COMPAT=m
+CONFIG_NFT_HASH=m
+CONFIG_NF_DUP_NETDEV=m
+CONFIG_NFT_DUP_NETDEV=m
+CONFIG_NFT_FWD_NETDEV=m
 CONFIG_NETFILTER_XTABLES=m
 
 #
@@ -921,14 +944,14 @@
 CONFIG_IP_SET_BITMAP_IPMAC=m
 CONFIG_IP_SET_BITMAP_PORT=m
 CONFIG_IP_SET_HASH_IP=m
-# CONFIG_IP_SET_HASH_IPMARK is not set
+CONFIG_IP_SET_HASH_IPMARK=m
 CONFIG_IP_SET_HASH_IPPORT=m
 CONFIG_IP_SET_HASH_IPPORTIP=m
 CONFIG_IP_SET_HASH_IPPORTNET=m
-# CONFIG_IP_SET_HASH_MAC is not set
-# CONFIG_IP_SET_HASH_NETPORTNET is not set
+CONFIG_IP_SET_HASH_MAC=m
+CONFIG_IP_SET_HASH_NETPORTNET=m
 CONFIG_IP_SET_HASH_NET=m
-# CONFIG_IP_SET_HASH_NETNET is not set
+CONFIG_IP_SET_HASH_NETNET=m
 CONFIG_IP_SET_HASH_NETPORT=m
 CONFIG_IP_SET_HASH_NETIFACE=m
 CONFIG_IP_SET_LIST_SET=m
@@ -980,12 +1003,20 @@
 #
 CONFIG_NF_DEFRAG_IPV4=m
 CONFIG_NF_CONNTRACK_IPV4=m
+CONFIG_NF_TABLES_IPV4=m
+CONFIG_NFT_CHAIN_ROUTE_IPV4=m
+CONFIG_NFT_REJECT_IPV4=m
+CONFIG_NFT_DUP_IPV4=m
+CONFIG_NF_TABLES_ARP=m
 CONFIG_NF_DUP_IPV4=m
 # CONFIG_NF_LOG_ARP is not set
 CONFIG_NF_LOG_IPV4=m
 CONFIG_NF_REJECT_IPV4=m
 CONFIG_NF_NAT_IPV4=m
+CONFIG_NFT_CHAIN_NAT_IPV4=m
 CONFIG_NF_NAT_MASQUERADE_IPV4=m
+CONFIG_NFT_MASQ_IPV4=m
+CONFIG_NFT_REDIR_IPV4=m
 CONFIG_NF_NAT_SNMP_BASIC=m
 CONFIG_NF_NAT_PROTO_GRE=m
 CONFIG_NF_NAT_PPTP=m
@@ -1016,11 +1047,18 @@
 #
 CONFIG_NF_DEFRAG_IPV6=m
 CONFIG_NF_CONNTRACK_IPV6=m
+CONFIG_NF_TABLES_IPV6=m
+CONFIG_NFT_CHAIN_ROUTE_IPV6=m
+CONFIG_NFT_REJECT_IPV6=m
+CONFIG_NFT_DUP_IPV6=m
 CONFIG_NF_DUP_IPV6=m
 CONFIG_NF_REJECT_IPV6=m
 CONFIG_NF_LOG_IPV6=m
 CONFIG_NF_NAT_IPV6=m
+CONFIG_NFT_CHAIN_NAT_IPV6=m
 CONFIG_NF_NAT_MASQUERADE_IPV6=m
+CONFIG_NFT_MASQ_IPV6=m
+CONFIG_NFT_REDIR_IPV6=m
 CONFIG_IP6_NF_IPTABLES=m
 CONFIG_IP6_NF_MATCH_AH=m
 CONFIG_IP6_NF_MATCH_EUI64=m
@@ -1040,6 +1078,10 @@
 CONFIG_IP6_NF_NAT=m
 CONFIG_IP6_NF_TARGET_MASQUERADE=m
 CONFIG_IP6_NF_TARGET_NPT=m
+CONFIG_NF_TABLES_BRIDGE=m
+CONFIG_NFT_BRIDGE_META=m
+CONFIG_NFT_BRIDGE_REJECT=m
+CONFIG_NF_LOG_BRIDGE=m
 CONFIG_BRIDGE_NF_EBTABLES=m
 CONFIG_BRIDGE_EBT_BROUTE=m
 CONFIG_BRIDGE_EBT_T_FILTER=m
$
```

aueelis, 2019-02-22:
Is there anything we can do, to bring this issue forward?

skrueger8, 2019-02-27:
Hi,

Debian Buster will use nftables instead of iptables, see https://wiki.debian.org/nftables

> Starting with Debian Buster, nf_tables is the default backend when using iptables, by means of the iptables-nft layer (i.e, using iptables syntax with the nf_tables kernel subsystem).

Raspbian Kernel 4.19 doesn't have the nf_tables* modules for this right now.

```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.25-v7+ #1205 SMP Mon Feb 25 18:19:20 GMT 2019 armv7l GNU/Linux
pi@raspberrypi:~ $ find /lib/modules -name "*nf_tables*"
nada
pi@raspberrypi:~ $ sudo nft add table inet filter
Error: Could not process rule: Operation not supported
add table inet filter
^^^^^^^^^^^^^^^^^^^^^^
```
Debian Buster alpha5 has:

```
root@debian:~# find /lib/modules -name "*nf_tables*"
/lib/modules/4.19.0-2-amd64/kernel/net/netfilter/nf_tables.ko
/lib/modules/4.19.0-2-amd64/kernel/net/netfilter/nf_tables_set.ko
```
IMHO we might end up with having no working firewall solutions when Raspbian switches to Buster.

JamesH65, 2019-02-28:
@XECDesign Any comment on the firewall in buster comment?

XECDesign, 2019-02-28:
This seems to be purely a kernel issue. If, for whatever reason, we can't enable nftables in the kernel, then I'd have to figure out how to work around that and make sure iptables still works.

It is my understanding that nftables is the standard and iptables use is discouraged now. Diverging from upstream distros tends to create headaches, but I'm sure it's possible if required.

dirdi, 2019-02-28:
Actually, there are tools[0] to use the "new" (available since kernel 3.13!) nf_tables kernel backend with the iptables frontend, but not vice versa. Hence, moving from iptables to nftables should be smooth. Sticking to iptables will create troubles in near-/mid-future.

[0] https://wiki.nftables.org/wiki-nftables/index.php/Legacy_xtables_tools

XECDesign, 2019-02-28:
I don't think the plan would be to support nftables syntax through iptables, but to remove 
iptables-nft so that iptables works like it does in stretch. And then saying "we don't support that" nftables issues start popping up.

Anyway, this is up to the kernel guys. If the answer is no, I suspect we'll be able to find a solution before Buster is stable.

dirdi, 2019-02-28:
@XECDesign sorry, I may expressed myself mistakable: What I intent to propose is to support iptables syntax support through nftables.

XECDesign, 2019-02-28:
That seems to be the default on Buster.

skrueger8, 2019-02-28:
The problem is that with Buster we'll still have a /usr/sbin/iptables binary, but it'll use the nf_* kernel modules. There's still a xt_* kernel modules iptables, but it's called /usr/sbin/iptables-legacy though.

So option A seems to be: compile/provide nf_ kernel modules
option B is change the iptables package (and whatnot) and rename ip{,6}tables-legacy{,-restore,-save} to iptabes without -legacy...

popcornmix, 2019-06-03:
As buster is approaching, I've had a closer look at the cost of these options.
I've tried enabling just the options explicitly mentioned in this PR ("part") and all options under CONFIG_NF_TABLES ("all").

Basically little effect on kernel.img size or free memory from boot.
"part" adds 490K of extra modules.
"full" adds 640K of extra modules.

Suggest going for "full" option. This will be in next rpi-update kernel build for testing.

popcornmix, 2019-06-03:
Latest rpi-update should have NF_TABLES enabled.

keithwilliamsnp, 2019-06-17:
> Latest rpi-update should have NF_TABLES enabled.

Unfortunately it doesn't. So no working iptables and no working Nftables.

popcornmix, 2019-06-17:
> Unfortunately it doesn't. So no working iptables and no working Nftables.

Can you say exactly what is missing?
```
pi@domnfs:~ $ uname -a
Linux domnfs 4.19.50-v7+ #1234 SMP Thu Jun 13 11:06:37 BST 2019 armv7l GNU/Linux
pi@domnfs:~ $ sudo modprobe configs
pi@domnfs:~ $ zcat /proc/config.gz |grep NF_TABLE
CONFIG_NF_TABLES=m
CONFIG_NF_TABLES_SET=m
CONFIG_NF_TABLES_INET=y
CONFIG_NF_TABLES_NETDEV=y
CONFIG_NF_TABLES_IPV4=y
CONFIG_NF_TABLES_ARP=y
CONFIG_NF_TABLES_IPV6=y
CONFIG_NF_TABLES_BRIDGE=y
```

keithwilliamsnp, 2019-06-17:
Not sure what is missing. Could take a look on my desktop, I guess. That was upgraded from stretch to buster at the same time and no problems with that. What I have done is alias ipt=iptables-legacy and iptables seems to work but iptables-restore does not seem to. I dislike iptables, but need a firewall till nftables works. (I did the alias to save typing)

keithwilliamsnp, 2019-06-17:
modprobe: FATAL: Module nf_tables not found in directory /lib/modules/4.19.42-v7+

popcornmix, 2019-06-17:
@keithwilliamsnp did you reboot after running rpi-update?

keithwilliamsnp, 2019-06-17:
Yes, rebooted to ensure newest kernel being used

popcornmix, 2019-06-17:
And you ran rpi-update first? You don't have the kernel version with the fix in.
Report output of ``uname -a``

keithwilliamsnp, 2019-06-17:
Linux piebox 4.19.42-v7+ #1219 SMP Tue May 14 21:20:58 BST 2019 armv7l
GNU/Linux

popcornmix, 2019-06-17:
That is the stable kernel from apt. That doesn't contain the NF_TABLES support.
Have you run rpi-update?

keithwilliamsnp, 2019-06-17:
just noticed the date/time is horridly wildly out of date. Could that
affect things?
Will reset that

keithwilliamsnp, 2019-06-17:
yes. will rerun it

keithwilliamsnp, 2019-06-17:
*** Performing self-update
  % Total    % Received % Xferd  Average Speed   Time    Time     Time
 Current
                                 Dload  Upload   Total   Spent    Left
 Speed
  0     0    0     0    0     0      0      0 --:--:--  0:02:09 --:--:--
  0curl: (7) Failed to connect to raw.githubusercontent.com port 443:
Connection timed out
 !!! Failed to download update for rpi-update!
 !!! Make sure you have ca-certificates installed and that the time is set
correctly

Will need to look at ca-certs, then have another go in the morning. It is
midnight now

keithwilliamsnp, 2019-06-17:
~# apt-get install ca-certificates
Reading package lists... Done
Building dependency tree
Reading state information... Done
ca-certificates is already the newest version (20190110).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

popcornmix, 2019-06-17:
You mentioned the time was wrong - that can stop https connections from working.
Also check you have basic internet connectivity (e.g. from browser in raspbian).

keithwilliamsnp, 2019-06-18:
SOLVED
Thanks popcornmix.
OK here is what I did. Firstly, I have to apologise as yes I had tried
running rpi-update when I had upgraded earlier in the day, but as it had
not run properly, in fact it had hung at 2 minutes in, it had not done its
job and I like every bad workman blamed my tools.
1. Check that there were no firewall blocks on port 443 between Pi and the
outside world. There were none. As I run Pi headless through an SSH
connection I did this by using wget to download from a HTTPS site that I
run on a remote server
2. check time on system clock, it seemed to be more or less right but I had
got a mention of time as a possible error. so a) install npt b) synchronise
system clock with ntp servers
3. I had done a distribution upgrade over a previous install, so I did a
bit of cleaning up. I had some non standard programs installed - no idea
whether these could cause conflicts or not. Apt-get autoremove and a clean
did get rid of a lot of junk
4. Checked through ethernet connection once more
5. rpi-update and reboot
6. service nftables start - SUCCESS
7. nft list ruleset - output contents of /etc/nftables.conf
Thanks again popcornmix. The error was obviously mine

keithwilliamsnp, 2019-06-18:
A further problem has arisen which suggests that I may not have been
totally wrong.
I had to do a further reboot this morning due to physically moving the
hardware. I got exactly the same results as I was getting yesterday when I
tried to get the nftables service started. It would not start. Lots of
fiddling around afterwards showed that it was attempting to load a config
file that I had not put there, ipv6 NAT. This, of course was referencing
non-existent commands. Further investigation showed that
netfilter-persistent was attempting to load this.
With NFTables this is not only unnecessary but it interferes. So I did a
apt-get purge netfilter-persistent. Followed by a reboot. NFTables then was
able to start and automatically load /etc/nftables.conf which contains all
the firewall rules I had written

keithwilliamsnp, 2019-06-19:
OK  One more problem. NFTables seems to be adding a a lot of unnecessary and unwanted tables
The idea of NFTables is that on starting the service, it loads the ruleset from /etc/nftables.conf. Also each rule in its definition contains an instruction as to which family it applies to. ip = ipv4, inet = ipv4 and ipv6 etc.
When I boot up and the check nft list ruleset, it shows a lot of tables - all ipv6 related which are not contained in /etc/nftables.conf. I have checked nftables.service and that is OK it only loads the correct conf file. So some script, somewhere is adding the extra tables which are totally unnecessary.
Any clues as to what and where that script might be so I can disable or edit it, please

JamesH65, 2019-07-30:
Closing this issue as questions answered/issue resolved. NF_TABLES now enabled by default in the kernel.

closed 2019-07-30 by JamesH65

cybern0id, 2019-09-10:
When will the nftables fix and the kernel including them become available from the apt repos?

pelwell, 2019-09-10:
Have you tried a recent Raspbian? The last two images (2019-06-24 and 2019-07-10/12) both contain kernels with NF_TABLES enabled.

cybern0id, 2019-09-10:
I have a pi in the field (so can't deploy a new image), recently upgraded to raspbian buster successfully. Stil,l the current kernel is 4.14.98-v7+ and the one available from apt repo as follows:

`apt-cache show raspberrypi-kernel
Package: raspberrypi-kernel
Status: install ok installed
Priority: extra
Section: kernel
Installed-Size: 115654
Maintainer: Serge Schneider <serge@raspberrypi.org>
Architecture: armhf
Source: raspberrypi-firmware
Version: 1.20190215-1
Replaces: raspberrypi-bootloader (<< 1.20160324-1)
Provides: linux-image
Breaks: raspberrypi-bootloader (<< 1.20160324-1)
Conffiles:
 /etc/default/raspberrypi-kernel aa1f56ed7333c9749dba288036f96334
Description: Raspberry Pi bootloader
 This package contains the Raspberry Pi Linux kernel.
Description-md5: 7536ab227cf13dcddacdeb86a3d06624
Homepage: https://github.com/raspberrypi/firmware`

neither of which have nftables enabled.

pelwell, 2019-09-10:
That doesn't look very upgraded to me - is it perhaps the last Stretch kernel? Did you change "stretch" to "buster" in all the necessary places before the upgrade?

cybern0id, 2019-09-10:
My bad! It seems the rpi specific apt sources in /etc/apt/sources.list.d/raspi.list had been commented out (I can't remember if by me manually during upgrade from strech or automatically as part of the upgrade). Uncommenting these (and changing to buster) have provided a load of upgrades including the recent kernel.

Apologies for the noise.