raspberrypi/linux #2184: Is it possible to enable target_core_mod in Raspbian?

ovirtanen, 2017-08-30:
Hi all,

I was interested in configuring a iSCSI target but noticed that target_core_mod is missing when installing targetcli-fb

`$ modprobe target_core_mod`
`modprobe: FATAL: Module target_core_mod not found in directory /lib/modules/4.9.46-v7+`

Any plans to add this module?

pelwell, 2017-09-01:
That looks fairly specialist - explain to us why is should be part of a standard Raspberry Pi image, and reassure us that it won't bloat the kernel image or reduce the free memory unless the module is loaded.

JamesH65, 2017-12-05:
@ovirtanen Is this still an issue for you?

JamesH65, 2018-06-27:
Do you have any figures for impact on kernel size and performance? Given a limited resources device like the Pi we like to ensure that kernel changes for smaller use cases do not have a detrimental impact.

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

JamesH65, 2018-08-31:
Closing due to lack of activity. Please request to be reopened if you feel this issue is still relevant.

closed 2018-08-31 by JamesH65

BjoKaSH, 2019-12-30:
Old issue, hope it is OK to revive it:

@pelwell asked, why enabling LIO (iSCSI target kernel support) could be useful.

I use my RPis also as a kind of custom-made NAS, with currently 4x 3TB hard-drives attached via multi-bay USB enclosures and ZFS 0.7.12 (build from source, the Raspbian zfs-dkms packages don't compile).

The old iscsitarget-dkms packages in Raspbian do no longer compile (since some years, it seems, although I only run across them this week) and are unsupported since 2014 and now [removed from Debian](https://bugs.launchpad.net/ubuntu/+source/iscsitarget/+bug/1613758).  So the mainline kernel module is the only remaining option the serve block IO.

Yes, it is not very performant, due to only USB2 hardware and network and drives competing for the same bus, but for my use-case as backup storage it would be fine.

Therefore I would very much appreciate if "CONFIG_TARGET_CORE" could be set to "module" in the next (minor) kernel update.

I can't say how much this will increase the kernel size, (I haven't yet been successful compiling a custom kernel on my RPi 3), but I thought modules do only take RAM when loaded?

Niich, 2020-01-25:
I would also appreciate if iSCSI target support was included as a loadable module.

I am trying to use my RPi to share a disk drive as a block device to allow me to keep my main server remote but still have block-level read/write to a cdrom drive. 

I have tried to use nbd but it does not provide the passthrough I need for cdrom. 

@BjoKaSH I was successful compiling a custom kernel that included CONFIG_TARGET_CORE 

I used a combination of these guides:
https://www.raspberrypi.org/documentation/linux/kernel/configuring.md
https://www.raspberrypi.org/documentation/linux/kernel/building.md

The modifications I made in menuconfig:
<M> Generic Target Core Mod (TCM) and ConfigFS Infrastructure  ---->
            --- Generic Target Core Mod (TCM) and ConfigFS Infrastructure                          
                 <M>   TCM/IBLOCK Subsystem Plugin for Linux/BLOCK                               
                 <M>   TCM/FILEIO Subsystem Plugin for Linux/VFS                                       
                 <M>   TCM/pSCSI Subsystem Plugin for Linux/SCSI                                      
                 <M>   TCM/USER Subsystem Plugin for Linux                                             
                 <M>   TCM Virtual SAS target and Linux/SCSI LDD fabric loopback module  
                 <M>   Linux-iSCSI.org iSCSI Target Mode Stack

list of commands I used (cross-compile for pi3)
cd ~
git clone https://github.com/raspberrypi/tools ~/tools
cd tools/
echo PATH=\$PATH:~/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin >> ~/.bashrc
source ~/.bashrc
git clone --depth=1 https://github.com/raspberrypi/linux
cd linux
KERNEL=kernel7
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- bcm2709_defconfig
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- menuconfig
cat .config | grep -i iscsi # used to verify the items i need are properly marked in the config
cat .config | grep -i target # used to verify the items i need are properly marked in the config
make -j 23 ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- zImage modules dtbs

stevemyatt, 2020-01-31:
Thanks @BjoKaSH and @Niich for your excellent work on this.

@JamesH65 and @pelwell : I am much more a beginner in this area than those guys.

My particular use case: 

Trying to breathe a few more years of life into an ageing, 2G RAM (max) laptop; nice 15.4 inch screen, good sound etc - but Micro$oft Windows 7 declares end of support and recommends "Windows 10 on a new PC" ! Sorry guys, but to a seasoned Ubuntu advocate, keen to gain another convert to the cause, that's a red rag to a bull.

Said laptop seems unable to boot USB stick with Ubuntu 19.10 live CD, suspect antique BIOS, so next option is LAN boot.

The resident BT HomeHub (6) doesn't appear to support addressing a PXE boot server. I understand from https://en.wikipedia.org/wiki/Preboot_Execution_Environment that a proxydhcp server can serve the additional details required to support PXE boot.

I have a number of "Raspberry Pi Model A Plus Rev 1.1" spare so began with raspbian buster.
I encountered this issue while following "Diskless Boot for a Raspberry Pi over PXE and iSCSI" (at https://tech.xlab.si/posts/pxe-boot-raspberry-pi-iscsi/) - in particular, the early setup for provisioning the boot server, "Creating the iSCSI target". Paraphrasing, this amounts to

**root@pi-db356398:~# apt purge targetcli* python-rtslib* python-configshell*
...
root@pi-db356398:~# apt install targetcli-fb dnsmasq
...
root@pi-db356398:/home/pi# targetcli
b'modprobe: FATAL: Module target_core_mod not found in directory /lib/modules/4.19.93+\n'
root@pi-db356398:/home/pi#**

I would respectfully suggest that this feature would be a boon in a domestic/maker/hobby environment, and request its provision.

rur0, 2021-01-15:
Please review this issue again, recent addition of network boot enabled tftpboot via NFS however not all applications run as expected. Some applications like kubernetes (k3s) require block based storage devices, which can be achieved using ISCSI while being over the network. It is possible to compile your own kernel, but to update it you need to repeat that process and when you are running more than 1 RPI can be very annoying.

Is it possible to create seperate packages for ISCSI kernel modules so only when you need them they can be installed?

git-developer, 2021-12-05:
Please re-open this issue and enable the _target_ kernel modules as [described](https://github.com/raspberrypi/linux/issues/2184#issuecomment-578410400) by Niich. This would enable an RPi to be used as host for remote block devices (e.g. storage and disc drives).