raspberrypi/linux #2185: RTL8191S USB wireless adapter / r8712u kernel module not connecting in 4.9.41-v7+ kernel

drmullins, 2017-08-31:
Fresh install of Raspbian Stretch Lite with 4.9.41-v7+ kernel
CSL 300MBit/s USB wireless adapter in powered hub 
RTL8191S chipset r8712u kernel module
Configured /etc/wpa_supplicant/wpa_supplicant.conf correctly
ls -1 /sys/class/net gives predictable name for usb adapter
lsmod lists r8712u kernel module

ADAPTER DOES NOT CONNECT
iwconfig (4.9.41-v7+ kernel - stretch) gives :

wlx24050f6664f6  unassociated  Nickname:"rtl_wifi"
          Mode:Auto  Access Point: Not-Associated   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

wlan0     IEEE 802.11  ESSID:"D & A Visitors"  
          Mode:Managed  Frequency:2.432 GHz  Access Point: **:**:**:**:**:**   
          Bit Rate=72.2 Mb/s   Tx-Power=31 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on
          Link Quality=70/70  Signal level=-34 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

where true MAC address of the Access Point has been suppressed

With 4.9.35-v7+ kernel in Jessie the adapter works correctly (without powered hub)  
with both internal and external (also using r8712u kernel module) adapters connecting simultaneously

iwconfig (4.9.35-v7+ kernel - jessie) gives

wlan0     IEEE 802.11  ESSID:"D & A Visitors"  
          Mode:Managed  Frequency:2.432 GHz  Access Point: **:**:**:**:**:**   
          Bit Rate=24 Mb/s   Tx-Power=31 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on
          Link Quality=61/70  Signal level=-49 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

wlan1     IEEE 802.11bgn  ESSID:"D & A Visitors"  Nickname:"rtl_wifi"
          Mode:Managed  Frequency:2.432 GHz  Access Point: **:**:**:**:**:**   
          Bit Rate:150 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=97/100  Signal level=100/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

pelwell, 2017-08-31:
Have you modified your wpa_supplicant.conf to use the new unique interface id `wlx24050f6664f6`?

drmullins, 2017-08-31:
my wpa_supplicant.conf is plain as follows (but please give me a link to use of the new interface id) :

country=GB
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

network={
	ssid="D & A Visitors"
	psk="****************"
}

a ralink USB adapter works OK - I am only having the problem with the realtek

iwconfig with ralink and realtek both in hub is as follows (realtek first, ralink last) :

wlx24050f6664f6  unassociated  Nickname:"rtl_wifi"
          Mode:Auto  Access Point: Not-Associated   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

wlan0     IEEE 802.11  ESSID:"D & A Visitors"  :
          Mode:Managed  Frequency:2.432 GHz  Access Point: **:**:**:**:**:**   
          Bit Rate=65 Mb/s   Tx-Power=31 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on
          Link Quality=70/70  Signal level=-35 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
:
wlx0019864344f3  IEEE 802.11  ESSID:"D & A Visitors"  
          Mode:Managed  Frequency:2.432 GHz  Access Point: **:**:**:**:**:**   
          Bit Rate=65 Mb/s   Tx-Power=20 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=70/70  Signal level=-25 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:44   Missed beacon:0

pelwell, 2017-08-31:
> a ralink USB adapter works OK - I am only having the problem with the realtek

That's useful information.

Have you tried downgrading the Stretch kernel to

pelwell, 2017-08-31:
As I was saying, have you tried downgrading the Stretch kernel to 4.9.35-v7+ as used in Jessie?
```
sudo rpi-update a8ee902ac25b98b77478c252af5580d631eb75a9
```
should get the same kernel and modules.

drmullins, 2017-08-31:
I've downgraded kernel to Jessie version to no avail. I've got three of these adapters and tried two - neither will connect, while the ralink still connects - so perhaps not a kernel issue but a problem nevertheless. Thanks anyway

pelwell, 2017-09-01:
Recent kernels have up to three possible drivers for RealTek devices, and you have to use blacklisting to choose which driver to use.

Try these steps on a working Jessie and a non-working Stretch:

1. Run `lsmod | grep -E "(rtl|8192)"` to show which driver is being loaded on your Pi.
2. Run `grep -E "(rtl|8192)" /etc/modprobe.d/*` to find any relevant blacklists.

drmullins, 2017-09-01:
lsmod | grep -E "(rtl|8192)" shows nothing in either case - but in both cases r8712u is absent in lsmod listing without adapter plugged in and in both cases r8712u is present after inserting adapter. Also in both cases lshw -C network says driver=r8712u for realtek adapter.

grep -E "(rtl|8192)" /etc/modprobe.d/* says blacklist rtl8192cu and blacklist rtl8xxxu in both cases

Also  as wpa_cli -i wlx24050f6664f6 produces the error Could not connect to wpa_supplicant in non-working stretch I have also run
sudo strace wpa_cli -i wlx24050f6664f6 and the first line with an error says
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)

Running dmesg after removing and reinserting r8712u yields the following
1241.525152] r8712u: module is from the staging directory, the quality is unknown, you have been warned.
[ 1241.531202] r8712u: register rtl8712_netdev_ops to netdev_ops
[ 1241.531239] usb 1-1.4.1: r8712u: USB_SPEED_HIGH with 4 endpoints
[ 1241.532067] usb 1-1.4.1: r8712u: Boot from EFUSE: Autoload OK
[ 1242.044156] usb 1-1.4.1: r8712u: CustomerID = 0x000a
[ 1242.044172] usb 1-1.4.1: r8712u: MAC Address from efuse = 24:05:0f:66:64:f6
[ 1242.044183] usb 1-1.4.1: r8712u: Loading firmware from "rtlwifi/rtl8712u.bin"
[ 1242.046292] usbcore: registered new interface driver r8712u
[ 1242.062535] r8712u 1-1.4.1:1.0 wlx24050f6664f6: renamed from wlan1
[ 1242.914273] r8712u 1-1.4.1:1.0 wlx24050f6664f6: 1 RCR=0x153f00e
[ 1242.914902] r8712u 1-1.4.1:1.0 wlx24050f6664f6: 2 RCR=0x553f00e
[ 1245.034273] IPv6: ADDRCONF(NETDEV_UP): wlx24050f6664f6: link is not ready

pelwell, 2017-09-01:
The same driver is being used, so that rules out one potential difference. Does the dmesg output on insertion differ between the two OSs?

drmullins, 2017-09-01:
dmesg with working jessie shows it connects after a few seconds where non-working stretch doesn't
[   58.224249] r8712u: module is from the staging directory, the quality is unknown, you have been warned.
[   58.226652] r8712u: register rtl8712_netdev_ops to netdev_ops
[   58.226668] usb 1-1.2: r8712u: USB_SPEED_HIGH with 4 endpoints
[   58.236962] usb 1-1.2: r8712u: Boot from EFUSE: Autoload OK
[   58.659396] usb 1-1.2: r8712u: CustomerID = 0x000a
[   58.659407] usb 1-1.2: r8712u: MAC Address from efuse = 24:05:0f:66:64:f6
[   58.659412] usb 1-1.2: r8712u: Loading firmware from "rtlwifi/rtl8712u.bin"
[   58.660812] usbcore: registered new interface driver r8712u
[   59.517203] r8712u 1-1.2:1.0 wlan1: 1 RCR=0x153f00e
[   59.517739] r8712u 1-1.2:1.0 wlan1: 2 RCR=0x553f00e
[   61.637409] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[   64.017227] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[   74.531165] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready

pelwell, 2017-09-01:
Is it the same firmware in both cases?
```
md5sum /lib/firmware/rtlwifi/rtl8712u.bin
```

drmullins, 2017-09-01:
yes - same checksum !

pelwell, 2017-09-01:
I'm running out of suggestions, but you could try adding `net.ifnames=0` to `/boot/cmdline.txt` to see if that makes a difference.

drmullins, 2017-09-01:
I've done that previously The only thought I have is that r8712u is a staging driver and perhaps stretch is treating it differently from jessie and this may be related to the wpa_cli error in stretch. Thanks for all attempts so for - I'll keep investigating to see if can find any other clues

drmullins, 2017-09-02:
This was a problem in Fedora 25
https://lists.fedoraproject.org/archives/list/users@lists.fedoraproject.org/thread/5ZEL6I4WPKGX4K4NSCXLIU4B4QSQNRIQ/
I updated to r92su driver. I attach WITHOUT WARRANTY the outline of a script. The steps worked for me.

#!/bin/bash
# Install r92su kernel module replacing r8712u
# Initial update
sudo apt-get update
# Install kernel headers
sudo apt-get install raspberrypi-kernel-headers
# Install git 
sudo apt-get install git
# Clone the rtl8192su repository
cd ~
git clone https://github.com/chunkeey/rtl8192su.git
# Make r92su driver
cd rtl8192su
make -f Makefile.r92su
# Create kernel module updates directory and copy new driver to it
UPDATES=/lib/modules/$(uname -r)/updates
if [ ! -d $UPDATES ] ; then
    sudo mkdir $UPDATES
    echo creating $UPDATES
fi
sudo cp r92su/r92su.ko $UPDATES
# Blacklist r8712u if not previously black listed
if [ ! -e /etc/modprobe.d/blacklist-r8712u.conf ] ; then
    sudo echo "blacklist r8712u" >/etc/modprobe.d/blacklist-r8712u.conf
fi
# update dependencies
sudo depmod
# Reboot at this point to enable new driver to be used

drmullins, 2017-09-03:
Issue also reported in Raspbian forum
https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=191844

JamesH65, 2017-12-05:
There's a lot of churn on the Realtek drivers in the kernel at the moment. Not sure when fully working stuff will make it to staging or mainline but it is close. My preference for this particular issue is to close it, as the work is being done in mainline (eventually). OK to close?

drmullins, 2017-12-06:
Yes close. I've got a work round using r92su driver as given in Raspbian
forum in any case. It may be a Stretch problem as well as a kernel
problem but I hope it is not forgotten !Thanks

Links:

  1. https://github.com/raspberrypi/linux/issues/2185#issuecomment-349274179
  2. https://github.com/

JamesH65, 2017-12-06:
OK, will close since I believe mainline/staging is getting a lot of realtek work done on it which is pointless to replicate here.

closed 2017-12-06 by JamesH65