raspberrypi/linux #2186: dhcpcd: fails to reconfigure/restart NTP with DHCP-provided NTP servers

bwarden, 2017-08-31:
In ntp-4.2.8p10, /etc/init.d/ntp now looks for a DHCP-provided configuration file in /run/ntp.conf.dhcp. The hook in dhcpcd5-6.10.1-1, **/lib/dhcpcd/dhcpcd-hooks/50-ntp.conf**, still points to the older location, /var/lib/ntp/ntp.conf.dhcp.

Workaround: explicitly setting $NTP_DHCP_CONF to "/run/ntp.conf.dhcp"

With this workaround, /run/ntp.conf.dhcp is properly generated upon restarting the dhcpcd service, however, it does not restart the ntp service. Enabling debug on dhcpcd reveals that the hook is apparently looking for the ntpd service, not the ntp service:
Aug 31 09:38:09 rocky dhcpcd[2609]: Failed to get unit file state for ntpd.service: No such file or directory

Workaround: explicitly setting $ntp_service to "ntp"

pelwell, 2017-08-31:
This is a Raspbian issue rather than a kernel issue - please report it [here](https://github.com/RPi-Distro/repo/issues).

bwarden, 2017-08-31:
Whoops, sorry. Done.

closed 2017-08-31 by bwarden