raspberrypi/linux #2192: bcm2835-i2s.c - run Pi i2s clock continuously

Gadgetoid, 2017-09-06:
I'm attempting to identify a way in which I can run the i2s clock on the Raspberry Pi continuously, and prevent the i2s driver from starting/stopping the clock.

This is effectively like running the Pi as clock slave, but while still sourcing the clock from the Pi itself.

I am investigating issues we have with DACs and speaker pops when opening/closing the audio output device, and have found a solution in Pulse Audio which can hold the audio device open indefinitely and totally eliminate the problem- at a huge (~25-30% on a Pi Zero!) CPU cost.

Since the i2s clock is a hardware peripheral, I believe that effectively slaving the Pi to itself, and wresting control of the i2s clock from the audio driver so it can be run continuously will solve this problem with zero CPU cost or overhead.

The trouble is- documentation is scant, and my understanding is limited. I've Googled fruitlessly and delved through as many device tree files as I can find. Is it even possible with the current driver to have bcm2835-i2s leave the clock alone while still slaved to the Pi's hardware clock? If so, how might I go about accomplishing this?

Thank you.

Note, this issue replaces where I was clearly barking up the wrong tree: https://github.com/raspberrypi/linux/issues/2172

HiassofT, 2017-09-06:
I'm not convinced keeping the clock always on is the correct solution. Better look into DAPM sequencing of the card/DAC driver

https://github.com/raspberrypi/linux/blob/rpi-4.9.y/Documentation/sound/alsa/soc/pops_clicks.txt
https://github.com/raspberrypi/linux/blob/rpi-4.9.y/Documentation/sound/alsa/soc/dapm.txt

Gadgetoid, 2017-09-06:
The trouble with DAPM is that it, presumably, requires the DAC to support it, so it would be impossible to implement on any existing product. Still this is potentially useful to consider in future, thanks!

While I tend to agree that correct power management is the proper approach, it's not always possible, and sometimes not always strictly necessary if it's not a portable application.

In our case, with a handful of existing, shipped and widespread products any trick to keep the i2s clock running would eliminate the problem, even if not ideal. I'm certain an attempt to leverage DAPM would require hardware changes and may not be possible at all with the DAC ICs currently in use. Running the clock would certainly consume less power than using a CPU-intensive, and otherwise largely unnecessary, process like Pulse to accomplish the same thing.

JamesH65, 2017-12-05:
@popcornmix @pelwell Do we have any sort of answer to the question?

Gadgetoid, 2017-12-08:
This issue has received much wider discussion here: https://github.com/raspberrypi/linux/issues/2212

The TLDR of this is that it's not actually sufficient to simply run the i2s clock alone- the LRclock (or frame sync) must also be run for the specific DAC with which I'm having an issue. Running just the i2s clock causes more problems than it solves.

I recommend this issue simply be closed.

JamesH65, 2017-12-08:
Duplicate of #2212

closed 2017-12-08 by JamesH65