raspberrypi/linux #2195: Hardware random number generator interrupts in DTS

SylvainGarrigues, 2017-09-12:
FreeBSD was using its own DTB so far but now all our drivers are compatible with Linux DTBs (the versions from the RPI's GitHub). All drivers, except one: the RNG driver.

Our DTS include the RNG interrupt number (found in https://www.kernel.org/doc/Documentation/devicetree/bindings/interrupt-controller/brcm%2Cbcm2835-armctrl-ic.txt) and interrupt-parent, but yours do not.

Would it be possible to include 'interrupts = <2 49>' and 'interrupt-parent' lines in your DTB for the rng node?

lategoodbye, 2017-09-13:
@SylvainGarrigues I'm not familiar with FreeBSD, can you point us to the RNG driver / devicetree binding in the FreeBSD sources?

SylvainGarrigues, 2017-09-13:
Thanks @lategoodbye for considering our request!

The driver is here: https://github.com/freebsd/freebsd/blob/master/sys/arm/broadcom/bcm2835/bcm2835_rng.c

It uses the RNG interrupt to signal its FIFO is full. See commit details (https://github.com/freebsd/freebsd/commit/b1e796bd34620b7b91473db7255e13f185d7123e) for further explanations. It has been approved by FreeBSD security team a while ago.

The legacy FreeBSD DTS is here:
https://github.com/freebsd/freebsd/blob/master/sys/boot/fdt/dts/arm/bcm2835.dtsi#L399

But as of today all our drivers have added the Linux compatibility strings (or are in review internally) and some of us at the embedded team do run BSD with your stock DTBs from raspberrypi/firmware - i.e. we will soon no longer need custom DTBs with your help hopefully.  

The only things which bugs us at the moment is that we cannot use the IRQ mode of the RNG driver if the "interrupts =" (and interrupt-parent) is not specified in the DTS. 

We do understand Linux doesn't need any, but we believe FDT should be OS agnostic and since there is an interrupt for the RNG, it doesn't hurt to mention it.

I'm just a regular FreeBSD contributor dedicated to bringing a top-notch user experience on all the RPI products, if you need more details on the driver itself I can bring in Security members or Core team members of FreeBSD in the loop.

Again, thanks for even replying, it's great to see your reactivity even to non-Linux guys.

Cheers

lategoodbye, 2017-09-13:
I've looked at the Linux RNG driver and it seems to me that the driver uses an obsolete polling API, which cannot take advantage of the interrupt.

So these are the tasks:
- add interrupt to binding document and dts file
- update bcm2835-rng.c from obsolete data_read to new read API

lategoodbye, 2017-11-18:
@SylvainGarrigues I plan to send a patch for the DTS part upstream. But AFAIK the interrupt-parent property shouldn't be necessary in this case.

JamesH65, 2018-06-27:
@lategoodbye Did anything come of this? Not sure whether to close or not.

lategoodbye, 2018-06-27:
Florian applied my patch for upstream 4.18

JamesH65, 2018-06-27:
Would it be work backporting to 4.14, or is this low enough priority we just wait for the next kernel bump?

pelwell, 2018-06-27:
It's a simple cherry-pick - compile-test ongoing.

pelwell, 2018-06-27:
Pushed to rpi-4.14.y and rpi-4.17.y.

closed 2018-06-27 by pelwell