raspberrypi/linux #2196: FTBFS on Debian

mati75, 2017-09-12:
Hi,

On build kernel in the Debian way with sources from this repository. Build is failed on:

```
/<<PKGBUILDDIR>>/drivers/char/broadcom/vc_sm/vmcs_sm.c: In function 'vc_sm_global_state_show':
/<<PKGBUILDDIR>>/drivers/char/broadcom/vc_sm/vmcs_sm.c:576:48: error: format '%X' expects argument of type 'unsigned int', but argument 3 has type 'dma_addr_t {aka long long unsigned int}' [-Werror=format=]
    seq_printf(s, "           DMA_ADDR     0x%08X\n",
```
The same issue is on 4.9 and 4.12.

JamesH65, 2017-09-13:
What system are you building on, and which version of gcc?

6by9, 2017-09-13:
https://github.com/raspberrypi/linux/blob/rpi-4.9.y/include/linux/types.h#L142
```
#ifdef CONFIG_ARCH_DMA_ADDR_T_64BIT
typedef u64 dma_addr_t;
#else
typedef u32 dma_addr_t;
#endif
```
Why have you got CONFIG_ARCH_DMA_ADDR_T_64BIT defined, or are you trying to build a 64bit kernel?
We don't support 64 bit kernels, and I don't know how well this driver is going to work in a 64 bit world. 

I'll investigate whether it is possible when I get a chance, but you'll have to give a bundle of information on what "the Debian way" is.

mati75, 2017-09-13:
@JamesH65 raspbian version of gcc-6 from stretch repo and buster repo.

@6by9 Here is sources for packages: https://github.com/mati75/linux-rpi In the same way I was build month ago 4.12.6 and it build fine. After changes in vmcs_sm.c something is wrong.

6by9, 2017-09-13:
Lovely repo. A few instructions would help? Which config are you actually building?

What is "wrong" is that you're building with a different set of config options to the standard Pi builds, and your compiler is saying that ```printf("%u", u64)``` will result in truncation.
Without any extra info you'll get a cast ```printf("%u", (u32)u64)``` which will silence the warning but isn't a fix.

mati75, 2017-09-24:
I found the reason of this bug.
I build kernel for v6 and v7 in the same time. When CONFIG_BCM_VC_SM=y is enabled when build failt.
The same problem is in Arch arm: https://github.com/archlinuxarm/PKGBUILDs/commit/a34d7052b6b2f90d8bd2dcabccd1472d63967904#comments

JamesH65, 2017-12-05:
@mati75 So this can be closed?

closed 2018-06-27 by JamesH65