raspberrypi/linux #2198: Suspend to ram support: suspend is fine (after some tweaking), but can't resume from it

piso77, 2017-09-14:
Hi there,

i was trying to make the 'suspend to ram' work on the raspberrypi boards (both on rpi2 and rpi3, branch rpi-4.13.y, but the same applies to 4.9 apparently), and i got to the point where i can succesfully suspend, but i can't find a way to resume from it.

By default, suspend support is off in the raspberry kernel but after tweaking the .config and enabling the necessary option (CONFIG_SUSPEND&c), i was finally able to test it:

$ sudo sh -c 'echo mem > /sys/power/state'
sh: echo: I/O error

...
[  317.843806] PM: Syncing filesystems ... done.
[  317.852920] PM: Preparing system for sleep (freeze)
[  317.853143] PM: Adding info for No Bus:vcs63
[  317.853395] PM: Adding info for No Bus:vcsa63
[  317.857163] Freezing user space processes ... (elapsed 0.001 seconds) done.
[  317.858883] OOM killer disabled.
[  317.858885] Freezing remaining freezable tasks ... (elapsed 0.001
seconds) done.
[  317.860187] PM: Suspending system (freeze)
[  317.860191] Suspending console(s) (use no_console_suspend to debug)
[  317.880754] platform regulatory.0: bus suspend
[  317.880848] sdio mmc1:0001:2: bus suspend
[  317.880867] sdio mmc1:0001:1: bus suspend
[  317.880891] mmcblk mmc0:aaaa: bus suspend
[  317.880915] mmc mmc1:0001: bus suspend
[  317.882143] sdhost-bcm2835 3f202000.mmc: bus suspend
[  317.882193] mmc-bcm2835 3f300000.mmc: bus suspend
[  317.882207] bcm2835_thermal 3f212000.thermal: bus suspend
[  317.882230] i2c-bcm2835 3f804000.i2c: bus suspend
[  317.882260] spi-bcm2835 3f204000.spi: bus suspend
[  317.882354] bcm2835-aux-uart 3f215040.serial: bus suspend
[  317.882367] bcm2835-aux-clk 3f215000.aux: bus suspend
[  317.882373] usb 1-1.1: type suspend
[  317.882385] uart-pl011 3f201000.serial: bus suspend
[  317.882469] bcm2835-clk 3f101000.cprman: bus suspend
[  317.882719] leds led1: class suspend
[  317.882893] leds led0: class suspend
[  317.883007] platform Fixed MDIO bus.0: bus suspend
[  317.883701] smsc95xx 1-1.1:1.0 eth0: entering SUSPEND2 mode
[  317.884184] usb 1-1: type suspend
[  317.884233] usb usb1: type suspend
[  317.884407] dpm_run_callback(): usb_dev_suspend+0x0/0x20 returns -2
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[  317.884416] PM: Device usb1 failed to suspend async: error -2
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[  317.884449] PM: Some devices failed to suspend, or early wake event detected
[  317.884528] usb 1-1: type resume
[  317.884779] platform Fixed MDIO bus.0: bus resume
[  317.884803] leds led0: class resume
[  317.884827] leds led1: class resume
[  317.884848] bcm2835-clk 3f101000.cprman: bus resume
[  317.884861] uart-pl011 3f201000.serial: bus resume
[  317.884883] bcm2835-aux-clk 3f215000.aux: bus resume
[  317.884894] bcm2835-aux-uart 3f215040.serial: bus resume
[  317.884915] spi-bcm2835 3f204000.spi: bus resume
[  317.884933] i2c-bcm2835 3f804000.i2c: bus resume
[  317.884947] bcm2835_thermal 3f212000.thermal: bus resume
[  317.885009] mmc-bcm2835 3f300000.mmc: bus resume
[  317.885021] sdhost-bcm2835 3f202000.mmc: bus resume
[  317.885062] platform regulatory.0: bus resume
[  317.885247] usb 1-1.1: type resume
[  317.885257] mmcblk mmc0:aaaa: bus resume
[  317.885267] mmc mmc1:0001: bus resume
[  317.939038] mmc1: queuing unknown CIS tuple 0x80 (2 bytes)
[  317.940621] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[  317.942203] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[  317.945048] mmc1: queuing unknown CIS tuple 0x80 (7 bytes)
[  318.033873] sdio mmc1:0001:1: bus resume
[  318.033878] sdio mmc1:0001:2: bus resume
[  318.033896] PM: resume of devices complete after 149.435 msecs
[... 97 lines cut ...]

$ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- 57 -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         

$ sudo hwclock -r
Thu Sep 14 11:22:50 2017  .922813 seconds
$ sudo hwclock -s

and set the wakealarm in the future:

$ sudo sh -c "echo 0 > /sys/class/rtc/rtc0/wakealarm"
$ sudo sh -c "echo `date '+%s' -d '+ 3 minutes'` > /sys/class/rtc/rtc0/wakealarm"

suspended the board as usual, but it didn't wake up after the timeout.

So, i'm running out of ideas at the moment, and i was wondering if you happen to know if there's a way to resume the board after suspending it or shall i just mark the feature as 'not available'?
And tangentially from that, is there a reason why the raspberry kernel still uses the dwc_otg driver instead of the upstream dwc2 one?

popcornmix, 2017-09-14:
> or shall i just mark the feature as 'not available'?

That would be my suggestion. In general with software if something has never been tested it won't work. I'm pretty sure dwc_otg won't handle this.

More significantly the GPU firmware won't handle this. You may suspend when the GPU is driving HDMI in a different resolution from the default boot resolution, perhaps outputting audio through a currently running DMA. There may be 3D or ISP running with a huge amount of context (both RAM and in peripheral registers). A huge amount of work would need to be done to support this.

You are free to use the dwc2 driver, but you'll find its performance is far lower than dwc_otg (partly due to not supporting the FIQ as well as multiple years of bug fixes and performance improvements).

lategoodbye, 2017-09-14:
@piso77 Nice to see that someone working on this. How about implementing suspend to idle, which should be easier?

JamesH65, 2017-12-05:
Any progress to report on this? Should this particular issue be closed as a wont fix? (Since the GPU cannot really function in this mode without considerable work)

piso77, 2017-12-05:
Yeah, wont' fix makes sense.

closed 2017-12-05 by piso77