raspberrypi/linux #2199: CONFIG_GPIO_WATCHDOG=m

PetrJakes, 2017-09-14:
I am asking CONFIG_GPIO_WATCHDOG=m to be added to the standard configurations.
Reason:
We are building Raspberry pi HAT acting as an external watchdog.
To make this work we need the standard Linux watchdog "machinery" tickling to the GPIO output pin regularly.
I can confirm above mentioned solution works (see last two posts in: [https://www.raspberrypi.org/forums/viewtopic.php?f=107&t=193163&p=1210952#p1210952](url)

pelwell, 2017-09-14:
This works for me and only adds a 10kB module to the firmware. Any objections, @popcornmix?

popcornmix, 2017-09-14:
Sounds okay to me.

pelwell, 2017-09-14:
The module has been enabled, so it will appear in the next rpi-update release.

JamesH65, 2017-12-05:
Please update to the latest kernel which may contain a fix for this issue.

closed 2017-12-05 by JamesH65