raspberrypi/linux #2202: please consider supporting /proc/config.gz in bcmrpi_defconfig and bcm2709_defconfig by default

stefson, 2017-09-18:
It would be great if you could change the default value for accessing the kernel config of a running kernel via /proc/config.gz ; at the moment this is set to module and one has to enable it via menuconfig. 

I guess it is enough to change both related values to

IKCONFIG [=y] 
IKCONFIG_PROC [=y]

but I'm not an experienced coder, so please consider this a proposal for enhancing the accessibility

pelwell, 2017-09-18:
What's wrong with the current module setting?
```
pi@raspberrypi:~$ sudo modprobe configs
pi@raspberrypi:~$ ls -l /proc/config.gz
-r--r--r-- 1 root root 33980 Sep 18 15:00 /proc/config.gz
```

stefson, 2017-09-18:
If you have a reason to not to inherit it as build-in you can close this as wontfix. 

To answer your question, this way you need root access to load the module while it is possible for every user to read the config from /proc/config.gz - last time I compared distros a few years ago it used to be a normal thing to allow every user access to the kernel config by default. So why shouldn't it be inherited to bypass the need of root access just to read the config? 

Again, if you disagree mark as wontfix and close the issue.

pelwell, 2017-09-18:
It is enabled as a loadable module because, especially on early Pis, RAM is in short enough supply that we don't want to waste 40kB when all you need to do is add "configs" to /etc/modules.

popcornmix, 2017-09-18:
I don't have a pi in front of me, but can't you add ``configs`` to ``/etc/modules`` if you want the module to be preloaded?
No need for every user to lose RAM (and increase kernel load time) for something that can be achieved easily through other means.

pelwell, 2017-09-18:
Yes, that works as expected.

JamesH65, 2017-12-05:
Should this be closed?

closed 2017-12-05 by pelwell