raspberrypi/linux #2203: How I can get jessie system source code？

zewenhs, 2017-09-20:
Not the rpi3 kernel source code, but the whole jessie system source code. How can I get that and build it? Is there an tutorials? thanks!

pelwell, 2017-09-20:
You really don't want to do this - it's a HUGE amount of code - but @XECDesign can tell you first hand just how bad it is.

JamesH65, 2017-09-20:
Indeed  - I think there are 25 thousand packages at least in the Jessie
repo. Given a completely guessed and very approximate (and probably an
underestimate) build time of 5 minutes per package on average, that would
be about 90 days to build on  a single Pi3.

I think the Raspbian guys have a server farm to do the work, but you could
build a rack of Pi3's, a big disk, and appropriate software to get the
build time reduced.

XECDesign, 2017-09-20:
Here's an overview of how it works:
https://www.debian.org/devel/buildd/

JamesH65, 2017-09-22:
Closing, now being dealt with on forum.

closed 2017-09-22 by JamesH65