raspberrypi/linux #2209: dtoverlays README pin numbering scheme

neilneil2000, 2017-09-25:
Hi, as I understand it, the dtoverlays use the BCM numbering scheme for GPIO. Could a note to this effect be added to the README file to alleviate confusion?

pelwell, 2017-09-25:
The Linux kernel, the firmware and (as far as I know) everything published by Raspberry Pi all use the Broadcom GPIO numbering scheme; it's the only one that makes sense on all platforms. I think the onus should be on peddlers of other numbering schemes to apologise for confusing everybody.

Having said that, knock up a pull request with some suitable wording and I'll probably merge it.

popcornmix, 2017-09-25:
Agreed. I'm not sure we should be mentioning these other confusing numbering schemes.

I'm certain upstream linux would not consider documenting some numbering scheme used by a third party that has nothing to do with the linux kernel and is not recommended to be used. As such I don't think we should.

pelwell, 2017-09-25:
We could explicitly mention the One True Numbering Scheme (all hail its name) though.

JamesH65, 2017-12-05:
Progress? Anything to actually do? Close?

neilneil2000, 2017-12-05:
I raised a pull request and it was accepted so from my perspective this can be closed

closed 2017-12-05 by neilneil2000