raspberrypi/linux #2212: I2S audio output driver issues since removal of Pulse Audio

Gadgetoid, 2017-09-26:
Ahoy! This is a bit of a tricky one, since I'm not 100% sure where to point the finger.

Our DAC products use the I2S audio feed and start/stop in sync with the I2S clock.

## Without Pulse Audio

To reproduce: *without* Pulse Audio in the chain on Raspbian Stretch, use VLC to play audio to an I2S ALSA output device and then *pause* VLC.

Result: The I2S clock continues to run and data is low, however the Frame-Sync signal *does not continue* (I have verified this on a 4-input scope monitoring the I2S and audio signals). The Left/Right speaker terminals are clamped HIGH and LOW causing a potential difference across the speaker, unimpeded current flow, and overheating.

## With Pulse Audio

To reproduce: *with* Pulse Audio running on Raspbian Stretch, use VLC to play audio, and then *pause* VLC.

Note: I use this Pulse Audio configuration to keep the clock running intentionally to avoid pops: https://gist.github.com/Gadgetoid/3301cec3e47495e75b31d3120d8f17d9

Result: The I2S clock and Frame-Sync continue to run, the data signal is low, this clocks out constant midrail because twos-compliment 0s is simply 0.

The difference here seems to be that with ALSA only, the Frame Sync pin stops and this causes overheating. However with Pulse taking care of audio, it correctly plays silence and keeps the outputs running accordingly.

From our perspective the removal of Pulse Audio in Stretch is suddenly causing a lot of reports of speakers overheating when connected to DAC products and running with paused audio. I don't know if we're at fault and are suddenly running into expected behaviour that was masked before, or if it's a driver issue.

pelwell, 2017-09-27:
Without looking at the code I would guess that Pulse Audio is doing soft-mixing and always keeps the DAC fed with a string of zeroes, whereas without it ALSA detects the end of the input stream and stops the codec. I would expect the codec (DAC) and its driver to perform any necessary housekeeping when the stream is stopped. For example, the PCM512x devices have clock error and clock halt detection that powers down the outputs and puts the device into standby. The CS4265 doesn't appear to have an auto-mute, but the driver implements the digital_mute method which ALSA calls on stream startup and shutdown.

Which DAC are you using?

Gadgetoid, 2017-10-04:
We're using the MAX98357A.

If you look at the "Turn Off Response" on page 12 of the datasheet: https://datasheets.maximintegrated.com/en/ds/MAX98357A-MAX98357B.pdf it hints that the expectation is that BCLK stops, leaving LRCLK (Frame-Sync) running, and that would produce a safe transition to midrail.

Is this something I could solve with a new/modified codec? My attempts to chase a solution through the source have so far been unsuccessful, but I'm an eager student.

pelwell, 2017-10-04:
I think the explanation can be found on page 16 of the datasheet:

> Do not remove LRCLK while BCLK is present. Removing LRCLK while BCLK is present can cause  unexpected output behavior including a large DC output voltage.

So your problem seems to be that pausing the stream without PulseAudio doesn't stop the BCLK, just the LRCLK. If you were still in the design phase you could take another GPIO and use it to gate BCLK, or use a codec with a safer handling of clock stopage, but by the sounds of it you are hoping for a software solution.

The bcm2835-i2s driver appears to stop the clocks whenever the stream is stopped or paused or suspended, unless the SND_SOC_DAIFMT_CONT flag is set. As a first step I suggest you instrument the driver with some pr_err() calls to see which "triggers" are being delivered and whether or not the clock is being stopped. It could also be that VLC isn't actually pausing the stream - just stopping sample delivery - in which case there isn't much the I2S driver can do without adding watchdog timeouts etc.

HiassofT, 2017-10-04:
I dug a bit into the bcm2835 i2s driver: the PCM clock is enabled as long as an PCM device is opened. This is controlled via startup/shutdown which also enables the bcm2835 i2s block (EN bit in the CS_A register). This also seems to start driving the BCLK output.

LRCLK however seems only to be driven when a stream is active - this is controlled via trigger (which sets the RXON or TXON bits in the CS_A register).

Looking at the MAX98357A datasheet an easy solution seems to be to control /SD_MODE via trigger in the card driver - if it's pulled low MAX98357A will be put in shutdown mode.

See remarks in the wiring diagram in fig.19 / page 31 of the datasheet:
> *RESPONDS TO CHANNEL 0 WHEN GPIO IS HIGH.
> THE MAX98357A/B ARE SHUT DOWN WHEN GPIO IS LOW.

pelwell, 2017-10-04:
Thanks for your interest, HiassofT.

> I dug a bit into the bcm2835 i2s driver: the PCM clock is enabled as long as an PCM device is opened.

Do you see a problem with a driver modification to gate the BCLK when the stream is paused or stopped unless [SND_SOC_DAIFMT_CONT](http://elixir.free-electrons.com/linux/v4.2/source/include/sound/soc-dai.h#L41) is specified (other than the possibility that some users are assuming continuous clocks even though SND_SOC_DAIFMT_CONT is not set)? Unfortunately SND_SOC_DAIFMT_GATED has the value 0 so looks like it would be default. SND_SOC_DAIFMT_CONT is applied when the "continous-clock" property is present - something which doesn't occur anywhere in the raspberrypi/linux kernel trees.

> Looking at the MAX98357A datasheet an easy solution seems to be to control /SD_MODE via trigger in the card driver - if it's pulled low MAX98357A will be put in shutdown mode.

That sounds cleaner than my "gate BCLK" suggestion, but still falls into the category of a hardware modification, which may not be a viable option.

HiassofT, 2017-10-05:
I'm rather undecided about the BCLK and SND_SOC_DAIFMT_CONT/GATED changes.

Only very few drivers actually support / use DAIFMT_CONT and personally I never used that - I'm not sure the feature even works (i.e. if the bcm2835 drives any I2S clock outputs when the I2S block is in disabled state).

Disabling BCLK on stream pause should be fine, but that probably means we have to stop the clock and I'm a bit worried that could cause some unexpected behaviour inside the I2S block - plus of course issues with existing cards where codecs then switch off (any maybe "pop")  because no clock is present.

Implementing that feature shouldn't be too complicated (we'd have to replace the dai->active check with some code that checks for actually enabled RX/TX streams) but we'd need to do quite some testing.

As none of the other codecs seem to have a problem with BCLK running and LRCLK stopped and the issue with the MAX98357 should be easily solvable by manually putting it into shutdown mode I'm a bit inclined to keep the code as it is - the change could be quite disruptive.

pelwell, 2017-10-05:
I think the burden should be on the cards with codecs that "pop" to specify the DAIFMT_CONT flag - having _GATED as the default tells me that clocks are expected to be stopped - but rolling out that change would be disruptive and I don't see a pressing need for it now.

Gadgetoid, 2017-10-11:
Thank you both for the discussion around this. I'm still working to fully grok everything involved, so this is helpful. As a software developer, I find it painful that I can't afford the time to figure this out for myself- but as always; we stand on the shoulders of giants.

While I don't expect you to tweak software to fix what might ultimately be hardware issues, I'm keen to know if the software side of things is performing to reasonable expectations.

Obviously the onus is on us to make sure our hardware behaves correctly under any circumstances the software might present, but since it's impossible to change hardware already in the field I have to at least look into this.

Thanks!

Gadgetoid, 2017-11-22:
@gnbl I've since done some more investigation into this issue, attempting to resolve it with my own third-party patches to the kernel so I can distribute ready-to-go software for our radio kit.

I changed `bcm2835_i2s_stop` here: https://github.com/raspberrypi/linux/blob/b9f2f5bd67177e044ff859fc193f45b2938a1572/sound/soc/bcm/bcm2835-i2s.c#L641-L658

To: 

```c
static void bcm2835_i2s_stop(struct bcm2835_i2s_dev *dev,
                struct snd_pcm_substream *substream,
                struct snd_soc_dai *dai)
{
        uint32_t mask;

        printk(KERN_INFO "BCM2835 I2S DEBUG: bcm2835_i2s_stop\n");

        if (substream->stream == SNDRV_PCM_STREAM_CAPTURE)
                mask = BCM2835_I2S_RXON;
        else
                mask = BCM2835_I2S_TXON;

        regmap_update_bits(dev->i2s_regmap,
                        BCM2835_I2S_CS_A_REG, mask, 0);

        printk(KERN_INFO "BCM2835 I2S DEBUG: active: %d, format: %u \n", dai->active, dev->fmt);

        /* Stop also the clock when not SND_SOC_DAIFMT_CONT */
        if (!(dev->fmt & SND_SOC_DAIFMT_CONT))
                bcm2835_i2s_stop_clock(dev);

}
```

This skips the check for `dai->active`, which I believe is there for handling two channels of audio that might be playing/stopping independently. I had real trouble tracking this flag back through the code.

Interestingly, it was clear from my testing (stuffing printk into the kernel module) that VLC was correctly stopping the audio stream when paused (at least I believe that's the case) and, but this check for `dai->active` for some reason prevents the i2s clock from stopping.

gnbl, 2017-11-22:
@Gadgetoid I'd rather have this solved "properly" (unless the IC does not adhere to I2S standard), though your kernel modification probably saves quite a few CPU cycles compared to playing zeros all day..

Here are the oscilloscope captures (https://imgur.com/a/6PLTq):

With _i2s_test_ (see my [previous post](https://github.com/raspberrypi/linux/issues/2212#issuecomment-346136243)). Clock (yellow) starts two pulses before LRCLK (cyan). Audio out ramp-up visible on both channels (**no pop** or crackle):
[image]

Playing audio. Note glitch/ gap in clock (yellow, about 30 us clock followed by 150 us high). Output (pink/blue) seems undefined before **pop** occurs (at trigger) when LRCLK starts, about 30 ms before data starts (not shown):
[image]

Sometimes the pop isn't noticable, but the trace looks very similar.

pelwell, 2017-11-22:
I'm only a software engineer, but the clock in the "Playing audio" case does not look healthy - it's as if something is pulling the line high, fighting against the clock. Unless it's a sampling error in the scope.

gnbl, 2017-11-22:
Both captures are with the same hardware, the [Adafruit Speaker Bonnet for Raspberry Pi](https://www.adafruit.com/product/3346) (nothing else connected, see the [schematic](https://cdn-learn.adafruit.com/assets/assets/000/037/882/original/raspberry_pi_schem.png?1481135658)) so I think it's purely software, specifically I think that i2s_test may improperly configure the outputs (but I don't know). What is the voltage on your RPi?

Unfortunately I cannot tell whether it may be the software opening the device briefly, or if it's the driver causing the short clock burst. Have you noticed the audio player software opening/closing the device multiple times?
I'd like to feed the IC test data but I can't think of an easy way to do that (maybe modifying i2s_test).

pelwell, 2017-11-22:
`i2s_test` maps the PCM functions to two pin groups, which isn't very clever. Run `raspi-gpio` after boot and after running `i2s_test` to confirm.

gnbl, 2017-11-22:
Will do, in 12 hrs. I could remove the mapping to the other port in https://github.com/arisena-com/rpi_src/blob/master/apps/i2s_test/src/i2s_test.c#L1465 to test (not sure if that would be all).

pelwell, 2017-11-22:
I couldn't see a mention of your platform, but the second mapping (28-31) is likely to be the problem.

gnbl, 2017-11-22:
Sorry: Zero W

pelwell, 2017-11-22:
On the Zero W GPIOs 28 and 29 are used for I2C, so both lines have strong pullups. I don't know the electronics behind mapping functions to pins, i.e. whether a pull-up on SCLK on pin A could affect SCLK on pin B, but it is at least a coincidence.

HiassofT, 2017-11-22:
I dug a bit further and noticed there's already a driver for the max98357 included in the llinux kernel
https://github.com/raspberrypi/linux/blob/rpi-4.9.y/sound/soc/codecs/max98357a.c

This driver configures the sdmode on start/stop so the max98357 will automatically be put into shutdown mode when no audio is output.

So all you need to do is enable the driver in the kernel, write a simple-card DT overlay referencing the driver and fix the hardware so that sdmode is correctly wired to a GPIO and not +5V - see the reference schematics in the datasheet.

gnbl, 2017-11-23:
Pin states, after reboot:

```
pi@raspberrypi:~ $ raspi-gpio get
BANK0 (GPIO 0 to 27):
  GPIO 00: level=1 fsel=0 alt=  func=INPUT
  GPIO 01: level=1 fsel=0 alt=  func=INPUT
  GPIO 02: level=1 fsel=0 alt=  func=INPUT
  GPIO 03: level=1 fsel=0 alt=  func=INPUT
  GPIO 04: level=1 fsel=0 alt=  func=INPUT
  GPIO 05: level=1 fsel=0 alt=  func=INPUT
  GPIO 06: level=1 fsel=0 alt=  func=INPUT
  GPIO 07: level=1 fsel=1 alt=  func=OUTPUT
  GPIO 08: level=1 fsel=1 alt=  func=OUTPUT
  GPIO 09: level=0 fsel=4 alt=0 func=SPI0_MISO
  GPIO 10: level=0 fsel=4 alt=0 func=SPI0_MOSI
  GPIO 11: level=0 fsel=4 alt=0 func=SPI0_SCLK
  GPIO 12: level=1 fsel=0 alt=  func=INPUT
  GPIO 13: level=1 fsel=0 alt=  func=INPUT
  GPIO 14: level=1 fsel=2 alt=5 func=TXD1
  GPIO 15: level=1 fsel=2 alt=5 func=RXD1
  GPIO 16: level=1 fsel=0 alt=  func=INPUT
  GPIO 17: level=1 fsel=0 alt=  func=INPUT
  GPIO 18: level=0 fsel=4 alt=0 func=PCM_CLK
  GPIO 19: level=0 fsel=4 alt=0 func=PCM_FS
  GPIO 20: level=0 fsel=4 alt=0 func=PCM_DIN
  GPIO 21: level=0 fsel=4 alt=0 func=PCM_DOUT
  GPIO 22: level=0 fsel=0 alt=  func=INPUT
  GPIO 23: level=0 fsel=0 alt=  func=INPUT
  GPIO 24: level=0 fsel=0 alt=  func=INPUT
  GPIO 25: level=1 fsel=0 alt=  func=INPUT
  GPIO 26: level=1 fsel=0 alt=  func=INPUT
  GPIO 27: level=0 fsel=0 alt=  func=INPUT
BANK1 (GPIO 28 to 45):
  GPIO 28: level=1 fsel=0 alt=  func=INPUT
  GPIO 29: level=1 fsel=0 alt=  func=INPUT
  GPIO 30: level=0 fsel=7 alt=3 func=CTS0
  GPIO 31: level=0 fsel=7 alt=3 func=RTS0
  GPIO 32: level=1 fsel=7 alt=3 func=TXD0
  GPIO 33: level=1 fsel=7 alt=3 func=RXD0
  GPIO 34: level=0 fsel=7 alt=3 func=SD1_CLK
  GPIO 35: level=1 fsel=7 alt=3 func=SD1_CMD
  GPIO 36: level=1 fsel=7 alt=3 func=SD1_DAT0
  GPIO 37: level=1 fsel=7 alt=3 func=SD1_DAT1
  GPIO 38: level=1 fsel=7 alt=3 func=SD1_DAT2
  GPIO 39: level=1 fsel=7 alt=3 func=SD1_DAT3
  GPIO 40: level=0 fsel=1 alt=  func=OUTPUT
  GPIO 41: level=1 fsel=1 alt=  func=OUTPUT
  GPIO 42: level=0 fsel=0 alt=  func=INPUT
  GPIO 43: level=0 fsel=4 alt=0 func=GPCLK2
  GPIO 44: level=0 fsel=1 alt=  func=OUTPUT
  GPIO 45: level=1 fsel=1 alt=  func=OUTPUT
BANK2 (GPIO 46 to 53):
  GPIO 46: level=0 fsel=0 alt=  func=INPUT
  GPIO 47: level=0 fsel=1 alt=  func=OUTPUT
  GPIO 48: level=0 fsel=4 alt=0 func=SD0_CLK
  GPIO 49: level=1 fsel=4 alt=0 func=SD0_CMD
  GPIO 50: level=1 fsel=4 alt=0 func=SD0_DAT0
  GPIO 51: level=1 fsel=4 alt=0 func=SD0_DAT1
  GPIO 52: level=1 fsel=4 alt=0 func=SD0_DAT2
  GPIO 53: level=1 fsel=4 alt=0 func=SD0_DAT3
  

pi@raspberrypi:~ $ ./rpi_src/apps/i2s_test/src/i2s_test -t 2
clock source src        = 1 supported
number of mash stages   = 1
integer divisor DIVI    = 12
fractional divisor DIVF = 2048

pi@raspberrypi:~ $ raspi-gpio get
[... 186 lines cut ...]
  GPIO 37: level=1 fsel=7 alt=3 func=SD1_DAT1
  GPIO 38: level=1 fsel=7 alt=3 func=SD1_DAT2
  GPIO 39: level=1 fsel=7 alt=3 func=SD1_DAT3
  GPIO 40: level=0 fsel=1 alt=  func=OUTPUT
  GPIO 41: level=1 fsel=1 alt=  func=OUTPUT
  GPIO 42: level=0 fsel=0 alt=  func=INPUT
  GPIO 43: level=1 fsel=4 alt=0 func=GPCLK2
  GPIO 44: level=0 fsel=1 alt=  func=OUTPUT
  GPIO 45: level=1 fsel=1 alt=  func=OUTPUT
BANK2 (GPIO 46 to 53):
  GPIO 46: level=0 fsel=0 alt=  func=INPUT
  GPIO 47: level=0 fsel=1 alt=  func=OUTPUT
  GPIO 48: level=0 fsel=4 alt=0 func=SD0_CLK
  GPIO 49: level=1 fsel=4 alt=0 func=SD0_CMD
  GPIO 50: level=1 fsel=4 alt=0 func=SD0_DAT0
  GPIO 51: level=1 fsel=4 alt=0 func=SD0_DAT1
  GPIO 52: level=1 fsel=4 alt=0 func=SD0_DAT2
  GPIO 53: level=1 fsel=4 alt=0 func=SD0_DAT3
```

So, after playback PCM_CLK and PCM_FS are kept high.

I probably need to clarify my earlier statement of "nothing else connected" - I meant the I2S connections. Other pins on the RPi are being used: SPI (8,9,10,11) and some gpios: 4, 5, 6, 13, 16, 17, 24, 25, 26 as well as UART (14, 15).

I will verify with scope again in 10 hrs, but I'm pretty sure I've seen this before.

pelwell, 2017-11-23:
The pin mapping looks as I would expect it to. I'm not too bothered about the I2S clocks being left high provided it is the I2S interface driving it high. If you wanted to experiment you could use raspi-gpio to temporarily switch them to inputs to see if the line stays high.
```
$ raspi-gpio set 18 ip pd
$ raspi-gpio set 19 ip pd
$ raspi-gpio get 18 # perhaps a few times
$ raspi-gpio get 19 # perhaps a few times
$ raspi-gpio set 18 a0
$ raspi-gpio set 19 a0
```
[ raspi-gpio desperately needs a pin-range feature ]
18 and 19 should naturally pull low, but we're setting it explicitly just in case something else has changed the default state.

gnbl, 2017-11-24:
Confirmed (after reboot):
```
$ raspi-gpio get 18
GPIO 18: level=0 fsel=4 alt=0 func=PCM_CLK
$ raspi-gpio get 19
GPIO 19: level=0 fsel=4 alt=0 func=PCM_FS
$ espeak "thanks"
...
$ raspi-gpio get 18
GPIO 18: level=1 fsel=4 alt=0 func=PCM_CLK
$ raspi-gpio get 19
GPIO 19: level=1 fsel=4 alt=0 func=PCM_FS
$ raspi-gpio set 18 ip pd
$ raspi-gpio set 19 ip pd
$ raspi-gpio get 18
GPIO 18: level=0 fsel=0 func=INPUT
$ raspi-gpio get 19
GPIO 19: level=0 fsel=0 func=INPUT
$ raspi-gpio set 18 a0
$ raspi-gpio set 19 a0
$ raspi-gpio get 18
GPIO 18: level=1 fsel=4 alt=0 func=PCM_CLK
$ raspi-gpio get 19
GPIO 19: level=1 fsel=4 alt=0 func=PCM_FS
```

I will attempt to create different clock sequences to narrow down the cause.

pelwell, 2017-11-24:
Are either of the devices discussed making use of the /SD_MODE pin? If they are then the overlay should include the `sdmode-gpios` property to allow the driver to cleanly start and stop the codec. If they don't then I'm afraid it sounds like a poor design choice that can be masked by the used of PulseAudio.

pelwell, 2017-11-24:
P.S. A raspi-gpio update is in the pipeline, so you will soon be able to write:
```
$ raspi-gpio set 18,19 ip pd
$ raspi-gpio get 18-19 # perhaps a few times
$ raspi-gpio set 19-18 a0
```

gnbl, 2017-11-26:
The Adafruit "speaker bonnet" does not. The pin is also used to select the audio channel, i.e. one is hardwired to 5V (left), the other has a pull-up to 5V (right). Cutting traces to instead connect to 3.3 V GPIO and changing the resistor to 200...266k (-> 1.1...0.9V, though 210,2 k according to their equation for R_Small) should be doable (no hidden tracks under chip).

"Poor design", I don't know. It depends whether the chip is compliant to the specification. But it looks like the spec does not actually specify startup conditions.

https://en.wikipedia.org/wiki/I%C2%B2S links https://web.archive.org/web/20070102004400/http://www.nxp.com/acrobat_download/various/I2SBUS.pdf

Just by touching wires:
1. Playing audio and removing/adding BLCK - no problem at all, **no noise**
2. Playing audio without LRCLK or removing it while playing - **very loud and distorted output**
3. A few simulated conditions (bit-banged on microcontroller at around 512 kHz) that seem to show **the cause** 
  A. High level on BCLK and LRCLK, no gap -> NO pop! [image]
  B. High level on BCLK and LRCLK, with low-level gap -> NO pop! [image]
  C. High level on BCLK and LRCLK, with **high-level gap** -> POP! [image]

(images hosted at https://imgur.com/a/i22GB)

So, what can I do to help eliminate the BCLK gap (or change default level to low) that the RPI seems to generate? 

PS; the raspi-gpio pin set specifier looks cool :-)

gnbl, 2017-11-26:
An attempt to dig through the code, basically following the BCM2835_I2S_TXON bit to figure out if the driver could be stopping the BCLK (alternatively, the device probably is opened twice).

Resources
- https://www.kernel.org/doc/html/latest/sound/soc/codec.html
- Pperators http://www.alsa-project.org/~tiwai/writing-an-alsa-driver/ch05s06.html

Code

- [DAI ops](https://github.com/raspberrypi/linux/blob/rpi-4.9.y/sound/soc/bcm/bcm2835-i2s.c#L740)

- [hw_params](https://github.com/raspberrypi/linux/blob/rpi-4.9.y/sound/soc/bcm/bcm2835-i2s.c#L336) calls clear_fifos

- [startup](https://github.com/raspberrypi/linux/blob/rpi-4.9.y/sound/soc/bcm/bcm2835-i2s.c#L693) stops clock, enables peripheral, disables standby

- [prepare ](https://github.com/raspberrypi/linux/blob/rpi-4.9.y/sound/soc/bcm/bcm2835-i2s.c#L615) starts clock, optionally calls clear_fifos  
  > Note that this [hw_params] and prepare callbacks may be called multiple times per initialization.

- [trigger](https://github.com/raspberrypi/linux/blob/rpi-4.9.y/sound/soc/bcm/bcm2835-i2s.c#L660) starts clock, sets TXON

- [clear_fifos](https://github.com/raspberrypi/linux/blob/rpi-4.9.y/sound/soc/bcm/bcm2835-i2s.c#L160) stops and restarts TXON, if running, but it stops the module!

- [stop](https://github.com/raspberrypi/linux/blob/rpi-4.9.y/sound/soc/bcm/bcm2835-i2s.c#L641) clears TXON

So maybe it's clear_fifos [stopping the module](https://github.com/raspberrypi/linux/blob/rpi-4.9.y/sound/soc/bcm/bcm2835-i2s.c#L187) that's causing the "gap"? Unfortunately I'm not setup for cross-compiling to add some debug output there or whenever TXON is written...

gnbl, 2017-11-27:
From [1st post](https://github.com/raspberrypi/linux/issues/2212#issue-260728305):
> The I2S clock continues to run and data is low, however the Frame-Sync signal does not continue

Sorry, I seem to have hijacked this issue. Is there a way to split my comments into a new issue (other than by deleting here and reposting, which probably isn't a good way), e.g. "I2S driver bit clock restart potentially causes pops on some audio DACs"?
I've managed to cross-compile with Ubuntu on VirtualBox, so once I can confirm this working on the Pi, I might be able to "trace" the glitch by adding debug prints, maybe confirming above guess.

P33M, 2017-11-27:
>Sorry, I seem to have hijacked this issue. Is there a way to split my comments into a new issue (other than by deleting here and reposting, which probably isn't a good way), e.g. "I2S driver bit clock restart potentially causes pops on some audio DACs"?

No, the research is applicable to both cases. The I2S peripheral should behave in a configurable way to accommodate codecs that either 
a) dislike LRCLK being removed while BCLK is running or
b) dislike BCLK being removed while LRCLK is running.

You are correct that the I2S spec has no defined startup/shutdown condition - it relies on codecs either recognizing that a clock has disappeared/started or on external pins controlling state. Avoiding "bad" states by fiddling with the ordering of enables/disables in the I2S block would be a useful feature to have.

pelwell, 2017-11-27:
Just to be clear, I'm not trying to be awkward. If there is a simple change we can make to the bcm2835-i2s driver that improves the shutdown behaviour for you without breaking anything else then I'm prepared to take a patch. A bit more research into the dai->active flag (what affects it, why it is used as a predicate when stopping a stream) would help to build the case.

HiassofT, 2017-11-27:
dai->active is the open stream refcount. It's incremented at the end of soc_pcm_open and decremented at the beginning of soc_pcm_close and can be used as an indicator of "someone is using this interface".

IMHO the use of this flag in bcm2835-i2s looks quite reasonable. It ensures that the clock is only stopped when the last user of the device closes it.

gnbl, 2017-11-27:
I don't seem to have the problem of the DAC going crazy when sound stops (AFAIK, both BCLK and LRCLK stop).
Compiling now seems to have worked, but I'm afraid I'm in over my head so all I can do to help is test things...

These debug output lines

```
diff --git a/sound/soc/bcm/bcm2835-i2s.c b/sound/soc/bcm/bcm2835-i2s.c
index 2e449d7..df065f3 100644
--- a/sound/soc/bcm/bcm2835-i2s.c
+++ b/sound/soc/bcm/bcm2835-i2s.c
@@ -144,6 +144,7 @@ static void bcm2835_i2s_start_clock(struct bcm2835_i2s_dev *dev)
 	case SND_SOC_DAIFMT_CBS_CFM:
 		clk_prepare_enable(dev->clk);
 		dev->clk_prepared = true;
+		printk("I2S: START");
 		break;
 	default:
 		break;
@@ -152,6 +153,7 @@ static void bcm2835_i2s_start_clock(struct bcm2835_i2s_dev *dev)
 
 static void bcm2835_i2s_stop_clock(struct bcm2835_i2s_dev *dev)
 {
+	printk("I2S: STOP");
 	if (dev->clk_prepared)
 		clk_disable_unprepare(dev->clk);
 	dev->clk_prepared = false;
@@ -177,6 +179,7 @@ static void bcm2835_i2s_clear_fifos(struct bcm2835_i2s_dev *dev,
 	/* Backup the current state */
 	regmap_read(dev->i2s_regmap, BCM2835_I2S_CS_A_REG, &csreg);
 	i2s_active_state = csreg & (BCM2835_I2S_RXON | BCM2835_I2S_TXON);
+	if(csreg&BCM2835_I2S_TXON) printk("I2S: TX is ON"); else printk("I2S: TX is OFF");
 
 	/* Start clock if not running */
 	clk_was_prepared = dev->clk_prepared;
@@ -184,6 +187,7 @@ static void bcm2835_i2s_clear_fifos(struct bcm2835_i2s_dev *dev,
 		bcm2835_i2s_start_clock(dev);
 
 	/* Stop I2S module */
+	if(off&BCM2835_I2S_TXON) printk("I2S: TX OFF"); else printk("I2S: TX ON");
 	regmap_update_bits(dev->i2s_regmap, BCM2835_I2S_CS_A_REG, off, 0);
 
 	/*
@@ -219,6 +223,7 @@ static void bcm2835_i2s_clear_fifos(struct bcm2835_i2s_dev *dev,
 		bcm2835_i2s_stop_clock(dev);
 
 	/* Restore I2S state */
+	if(i2s_active_state&BCM2835_I2S_TXON) printk("I2S: TX ON"); else printk("I2S: TX OFF");
 	regmap_update_bits(dev->i2s_regmap, BCM2835_I2S_CS_A_REG,
 			BCM2835_I2S_RXON | BCM2835_I2S_TXON, i2s_active_state);
 }
@@ -649,6 +654,7 @@ static void bcm2835_i2s_stop(struct bcm2835_i2s_dev *dev,
 	else
 		mask = BCM2835_I2S_TXON;
 
+	if(mask&BCM2835_I2S_TXON) printk("I2S: TX OFF"); else printk("I2S: TX ON");
 	regmap_update_bits(dev->i2s_regmap,
 			BCM2835_I2S_CS_A_REG, mask, 0);
 
@@ -674,6 +680,7 @@ static int bcm2835_i2s_trigger(struct snd_pcm_substream *substream, int cmd,
 		else
 			mask = BCM2835_I2S_TXON;
 
+		if(mask&BCM2835_I2S_TXON) printk("I2S: TX ON"); else printk("I2S: TX OFF");
 		regmap_update_bits(dev->i2s_regmap,
 				BCM2835_I2S_CS_A_REG, mask, mask);
 		break;
@@ -702,6 +709,7 @@ static int bcm2835_i2s_startup(struct snd_pcm_substream *substream,
 	bcm2835_i2s_stop_clock(dev);
 
 	/* Enable PCM block */
[... 139 lines cut ...]
and it's repeatable

```
[ 3939.190700] I2S: STOP
[ 3939.190714] I2S: ENABLE
[ 3939.190732] I2S: STBY
[ 3939.191465] I2S: TX is OFF
[ 3939.191483] I2S: START
[ 3939.191488] I2S: TX OFF
[ 3939.191505] I2S: STOP
[ 3939.191516] I2S: TX OFF
[ 3939.191683] I2S: START
[ 3939.191978] I2S: TX ON

[ 3940.617207] I2S: TX OFF
[ 3940.617273] I2S: TX OFF
[ 3940.617283] I2S: STOP
[ 3940.617307] I2S: DISABLE
[ 3940.617315] I2S: STOP
```

Unfortunately, I have no clue what part of this sequence could be causing the "BLCK gap".

Off-topic:
How do you guys usually debug this low level peripheral hardware stuff? Step through the code (debugger?) and toggle pins to sync a logic analyser?

gnbl, 2020-06-06:
I'm revisiting this as I have obtained another I2S-DAC/amplifier board that is unfortunately also based on the MAX98357 (like [Adafruit 3006](https://learn.adafruit.com/adafruit-max98357-i2s-class-d-mono-amp)), resulting in pops and cracks on playback due to the I2S driver/hardware causing BCLK to be high briefly before LRCLK starts (see old oscilloscope screenshots [here ](#issuecomment-346409470)and [here](#issuecomment-347016922)).

Adafruit customers are "suffering" from the problem, too - [their workaround](https://learn.adafruit.com/adafruit-max98357-i2s-class-d-mono-amp?view=all#i2s-audio-faq):

> Even with dmixer enabled, I get a staticy-pop when the Pi first boots or when it first starts playing audio
> Yep, this is a known Raspbian Linux thing. Yay Linux! We don't have a fix for it. If it makes you feel better, my fancy Windows development computer does the same thing with my desktop speakers.
> You can reduce popping a lot with the '/dev/zero play' option in the i2s setup script. (We added it in October 2018) but you still might get that one initial pop

While I do not have a patch that configures the I2S properly (I am still hoping for an experienced develeoper to implement that), I have some bits of information about the current state and workarounds that might help others:

In contrast to the stereo "Speaker Bonnet" this mono board has the SD pin (active-low / voltage level on pin selects SD/L+R/R/L) brought out, but support for that is limited:

- there has been a driver that supports the SD since 2015: https://github.com/raspberrypi/linux/blob/rpi-4.19.y/sound/soc/codecs/max98357a.c
  - but it's **not available** yet to be configured (built) - [it will be in 5.4](https://github.com/raspberrypi/linux/commit/3d1e5203531fd1791762f018538e670bf10f722c) 
- there has been an DT overlay since 2019: https://github.com/raspberrypi/linux/blob/rpi-4.19.y/arch/arm/boot/dts/overlays/max98357a-overlay.dts (this requires the driver module to be present, otherwise: "no soundcards found". It defaults to using GPIO4 as SD line)

Activation would be as simple as modifying `/boot/config.txt` 

    dtoverlay=max98357a
    #dtparam=audio=on

and providing an `/etc/asound.conf` (see Adafruit tutorial).

But even if you do manage to get this driver module running by building your own linux/kernel, there will be problems with playback: due to this particular hardware, setting the SD pin high will override the external (1M) and internal (100k) resistors intended to configure the device's SD pin voltage to mix left and right channels, resulting in **playback of left channel only**.

I see two options to fix this:
1. modify hardware: remove 1M pull-up and add a resistor with appropriate value in series with SD pin
2. modify driver to be open-drain instead, i.e. do not drive pin high to enable the device, only low when disabling it.

This could be a simple change and it would be a nice option for the driver :-)

However, there are other pitfalls:
1. documention on libgpiod functions is hard to find in the open - here's a "trace":

   - https://www.raspberrypi.org/documentation/usage/gpio/
   - https://kernel.googlesource.com/pub/scm/libs/libgpiod/libgpiod/+/v0.2.x/README.md
   - https://duckduckgo.com/&q=libgpiod
   - https://github.com/brgl/libgpiod
   - https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/about/
   - https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/tree/include/gpiod.h
   
   But you might end up with something like
   ```
    //gpiod_set_value(sdmode, 1);
    gpiod_direction_input(sdmode);
    ....
    // gpiod_set_value(sdmode, 0);
    gpiod_direction_output(sdmode, 0);
   ```
   But ...

2. the pin is configured with [pull-up by default](https://www.raspberrypi.org/documentation/hardware/raspberrypi/gpio/README.md), so this still doesn't work (test disabling it manually with `raspi-gpio set 4 pn`).
3. libgpiod does not yet support pull-up/down resistors ([it apparently will from version 5.5](https://microhobby.com.br/blog/2020/02/02/new-linux-kernel-5-5-new-interfaces-in-gpiolib/))

Options: 
1. disable the pull-up in the driver using the old "API" (obsolete soon)
2. Disable it using DTO in `/boot/config.txt` https://www.raspberrypi.org/documentation/configuration/pin-configuration.md

        gpio=4=pn

With that, using the SD line eliminates pops/cracks as expected.

Lastly, reading the discussion again about the `dai->active` flag gave me the idea that all that is required to keep the clock running would be opening the device, but not necessarily playing anything, as the `/dev/zero` "hack" does, at the cost of a wasteful few percent CPU load. 
It seems it's not as easy as opening the respective device file for writing, but it [needs some configuration](https://unix.stackexchange.com/a/529915). So the next best thing to using a library such as [pyaudio](http://people.csail.mit.edu/hubert/pyaudio/docs/#example-blocking-mode-audio-i-o) is using a program that opens and configures the device, but then does not play anything. What program might do that? `aplay` in interactive mode:

    aplay -i

(needs to be automatically run on startup as background process, see https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/master/i2samp.sh#L444)

This still means you get the inital pop/crack on power-up, though.

pelwell, 2020-06-06:
This really deserves an issue of its own - it should be easy to add OPEN_DRAIN designation to the relevant GPIOs.

gnbl, 2020-06-06:
I'm not sure.
This is very specific to this hardware (IC and board design), and I suspect it's one of the reasons why Adafruit seem not to be using the pin they put in their design themselves - that and that there is no SD pin support in the `pcm5102a` driver they seem to use as default I2S DAC driver via `hifiberry-dac`.
So while it would be nice to have an open-drain option in the DTO and driver, the SD pin would not be required at all for this specific device if the I2S clock "glitch" could be avoided in the first place (which isn't really the original issue here, only related, and it also might be specific to the SOC's I2S peripheral).
I only went down the SD pin "rabbit hole" because I have no chance in figuring out the clock "glitch" issue, unfortunately - my "Speaker Bonnet" (2x MAX98357 without SD pins brought out) has been collecting dust for over two years now...

Options
1. Fix I2S - helps all MAX98357
2. Support open drain - a couple of lines of code would help with one MAX98357 design
3. keep sound device open - easiest: eliminates most pops/cracks, except for start/stop
4. use a different chip that copes with the I2S output (

arm000, 2020-10-03:
Hi I found this thread as I ran into the same issue with the MAX98357 DAC in the Adafruit product and I've been trying to fix it.  Here's what my digging has found so far.  

The popping occurs as a result of ioctl SNDRV_PCM_IOCTL_HW_PARAMS which eventually results in bcm2835_i2s_hw_params() in  snd_soc_bcm2835_i2s called. The pop occurs when the clock input to the i2s controller is initialized (PLLD 500Mhz with clock divider set to 1411200Hz [44.1kHz * 16bits * 2 stereo channels]).  I notice that when bcm2835_i2s_start_clock() is called, BCLK immediately starts toggling because CLK_DIS of the MODE_A register is set to 0 (PCM clock enabled).  As pointed out previously in this thread, this DAC does not like seeing BCLK toggle without LRCLK and makes a loud pop (I think to the DAC it looks like an enormous data frame is being sent).

I tried a hack to disable BCLK output using CLK_DIS until the stream starts, and that eliminates all popping, but I'm not sure if it's the correct solution.  In particular it breaks clear_fifos() which relies on BCLK toggling to get the SYNC notification that the operation is complete.

arm000, 2020-10-11:
Posted patch to LKML for review: https://lkml.org/lkml/2020/10/10/121

MayaPosch, 2021-11-15:
I'm glad I found this issue thread, but rather displeased that even after four years this issue has not been resolved in any meaningful way. I too ended up with this audio bonnet with the I2S chip that seems to be at the root of this problem.I can't believe that AdaFruit is still selling the product when it's practically unusable.

Anyway, am I correct in assuming that this I2S chip may work better with other SoCs, or is it a purely driver problem? Or, is this I2S IC basically a lost cause and one should move on to a better I2S solution?

Thanks, and my apologies for the frustration. I have spent far too long on debugging what seemed like a simple issue.

arm000, 2021-11-16:
IMHO it's a combination of the board not wiring up the SD_MODE GPIO according to the Max98357A data sheet (diagrams 4 and 5 https://datasheets.maximintegrated.com/en/ds/MAX98357A-MAX98357B.pdf)  and the driver enabling BCLK before clearing the FIFOs and enabling LRCLK.  The patch I posted to LKML was rejected by the rpi i2s driver owner because he didn't want to risk changing the clock init order and breaking other devices.  IIRC the driver has support for configuring the SD_MODE GPIO via device-tree option if you're comfortable reworking the board to fix the circuit and hooking it up to an unused GPIO.  Otherwise you can cherry-pick that patch and rebuild your kernel which is what I did for the project I was using.

MayaPosch, 2021-12-02:
I have fixed my issue by modifying the board itself. I removed the 390k resistor on the right channel codec and replaced it with a 220k one which is only on the !SD pad, with the other side connected via a bodge wire to GPIO4.This puts 1V on the right channel !SD input, for right channel selection.

On the left channel side I have cut the !SD trace to 5V Vcc and wired this !SD also to GPIO 4 with a bodge wire.

After switching to the max98357a driver in the configuration, playback is now without popping and crackling.