raspberrypi/linux #2216: WiFI HotSpot / Adhoc

symbios24, 2017-10-01:
Hello, is it possible to add the option to the Latest Raspbian Stretch to Create an WIFI Hotspot or Adhoc Access point using the internal wifi or a usb wifi adapter ?

Thank you

JamesH65, 2017-10-01:
This is the wrong place for such requests - this github is for linux kernel issues.

There are instructions available on the Raspberry Pi website to create an AP. They are about to be updated for Stretch, but the principle is mostly there. Takes about 10 minutes of editing config files.

closed 2017-10-01 by JamesH65

symbios24, 2017-10-17:
i`m aware of the instructions to create an AP, i want with the same raspberry to take internet with one wifi adapter  and with another to give-create AP, i dont want to create an AP that is taking internet from the ethernet port.
Thanks

JamesH65, 2017-10-17:
Again, this github is for reporting bugs in the linux kernel on Raspberry Pi. Your question is much better suited to being asked on the forum. https://www.raspberrypi.org/forums