raspberrypi/linux #2217: Kernel oops in vc4_overflow_mem_work

anyc, 2017-10-03:
Hi,

I am using a RPi2 with a Qt5+QML app to display a slideshow with large moving images on an HDMI tv. This works well for quite some time - I'd say at least 30 minutes - but I always get a kernel oops eventually. I compiled the kernel myself and I use an Ubuntu userspace. I managed to get two oops reports that (as far as I can tell) occured at the same place in the vc4_overflow_mem_work function. Here is one of them:

```
[ 2697.125567] Unable to handle kernel NULL pointer dereference at virtual address 00000088
[ 2697.133678] pgd = 80004000
[ 2697.136386] [00000088] *pgd=00000000
[ 2697.139991] Internal error: Oops: 17 [#1] SMP ARM
[ 2697.144701] Modules linked in: rtc_ds1307 hwmon panel_raspberrypi_touchscreen r8188eu(C) cfg80211 rfkill i2c_gpio i2c_algo_bit bcm2835_gpiomem i2c_bcm2835 uio_pdrv_genirq uiod
[ 2697.173852] CPU: 0 PID: 944 Comm: kworker/0:0 Tainted: G         C      4.9.52-rpi23 #1
[ 2697.181863] Hardware name: BCM2835
[ 2697.185369] Workqueue: events vc4_overflow_mem_work [vc4]
[ 2697.190783] task: 825cac40 task.stack: 995d8000
[ 2697.195355] PC is at vc4_overflow_mem_work+0x80/0x120 [vc4]
[ 2697.200933] LR is at _raw_spin_lock_irqsave+0x24/0x68
[ 2697.205984] pc : [<7f173650>]    lr : [<807f0f24>]    psr: 600f0093
[ 2697.205984] sp : 995d9ec0  ip : 995d9ea8  fp : 995d9edc
[ 2697.217459] r10: 00000000  r9 : 00000000  r8 : b775cd00
[ 2697.222681] r7 : b6b5e0e0  r6 : 00000001  r5 : 00000000  r4 : b6b5e118
[ 2697.229205] r3 : 00000002  r2 : 00000003  r1 : 200f0013  r0 : 00080000
[ 2697.235730] Flags: nZCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
[ 2697.242949] Control: 10c5387d  Table: 1dd3806a  DAC: 00000055
[ 2697.248693] Process kworker/0:0 (pid: 944, stack limit = 0x995d8210)
[ 2697.255043] Stack: (0x995d9ec0 to 0x995da000)
[ 2697.259402] 9ec0: b6b5e118 b5c4a200 b77594c0 b77594c0 995d9f14 995d9ee0 8013611c 7f1735dc
[ 2697.267581] 9ee0: 995d8018 b77594c0 b5c4a218 b5c4a200 b77594d8 995d8018 b77594c0 b5c4a218
[ 2697.275759] 9f00: 00000008 80c02d00 995d9f5c 995d9f18 80136480 80135fd8 995d9f3c 995d8000
[ 2697.283936] 9f20: 00000000 80c02d00 80136420 80c792a3 00000000 995c7a40 00000000 b5c4a200
[ 2697.292114] 9f40: 80136420 00000000 00000000 00000000 995d9fac 995d9f60 8013c280 8013642c
[ 2697.300292] 9f60: 825cac40 00000000 b788a588 b5c4a200 00000000 00000000 995d9f78 995d9f78
[ 2697.308470] 9f80: 00000000 00000000 995d9f88 995d9f88 995c7a40 8013c170 00000000 00000000
[ 2697.316647] 9fa0: 00000000 995d9fb0 80108128 8013c17c 00000000 00000000 00000000 00000000
[ 2697.324825] 9fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 2697.333003] 9fe0: 00000000 00000000 00000000 00000000 00000013 00000000 05000116 00000000
[ 2697.341246] [<7f173650>] (vc4_overflow_mem_work [vc4]) from [<8013611c>] (process_one_work+0x150/0x454)
[ 2697.350650] [<8013611c>] (process_one_work) from [<80136480>] (worker_thread+0x60/0x55c)
[ 2697.358745] [<80136480>] (worker_thread) from [<8013c280>] (kthread+0x110/0x128)
[ 2697.366147] [<8013c280>] (kthread) from [<80108128>] (ret_from_fork+0x14/0x2c)
[ 2697.373373] Code: e1a03613 e5043004 f57ff04e e514000c (e5952088) 
[ 2697.379469] ---[ end trace fd3dc7cb8570316e ]---
[ 2697.385086] Unable to handle kernel paging request at virtual address ffffffec
[ 2697.392331] pgd = 80004000
[ 2697.395039] [ffffffec] *pgd=37f7e861, *pte=00000000, *ppte=00000000
[ 2697.401328] Internal error: Oops: 37 [#2] SMP ARM
[ 2697.406029] Modules linked in: rtc_ds1307 hwmon panel_raspberrypi_touchscreen r8188eu(C) cfg80211 rfkill i2c_gpio i2c_algo_bit bcm2835_gpiomem i2c_bcm2835 uio_pdrv_genirq uiod
[ 2697.435110] CPU: 0 PID: 944 Comm: kworker/0:0 Tainted: G      D  C      4.9.52-rpi23 #1
[ 2697.443108] Hardware name: BCM2835
[ 2697.446515] task: 825cac40 task.stack: 995d8000
[ 2697.451052] PC is at kthread_data+0x18/0x20
[ 2697.455236] LR is at wq_worker_sleeping+0x18/0xd8
[ 2697.459938] pc : [<8013ce94>]    lr : [<801373c8>]    psr: 200f0193
[ 2697.459938] sp : 995d9be8  ip : 995d9bf8  fp : 995d9bf4
[ 2697.471413] r10: 80c0442c  r9 : 80b87940  r8 : 825cb060
[ 2697.476634] r7 : 825cac40  r6 : b7759940  r5 : 825cb068  r4 : 00000000
[ 2697.483157] r3 : 00000000  r2 : 00000020  r1 : 00000000  r0 : 825cac40
[ 2697.489682] Flags: nzCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
[ 2697.496901] Control: 10c5387d  Table: 1dd3806a  DAC: 00000055
[ 2697.502643] Process kworker/0:0 (pid: 944, stack limit = 0x995d8210)
[ 2697.508993] Stack: (0x995d9be8 to 0x995da000)
[ 2697.513351] 9be0:                   995d9c04 995d9bf8 801373c8 8013ce88 995d9c74 995d9c08
[ 2697.521529] 9c00: 807ec620 801373bc 8017d0d8 801e1050 825cac40 00000000 82471a40 82479600
[ 2697.529707] 9c20: 80c834d4 00000001 995d9c44 995d9c38 80148dac 36bd2000 80b87940 995d9c48
[ 2697.537885] 9c40: 8011f790 8017d328 80c02040 995d98d4 b70d8000 825cafe0 995d9c90 00000001
[ 2697.546063] 9c60: 00000001 825cac40 995d9c84 995d9c78 80148dac 807ec188 995d9cc4 995d9c88
[ 2697.554243] 9c80: 801213e8 80148d20 995d9cb4 00000001 995d9c90 995d9c90 0000000b 80c832c4
[ 2697.562421] 9ca0: 80c08990 600f0193 0000000b 00000001 7f173652 7f173654 995d9d4c 995d9cc8
[ 2697.570598] 9cc0: 8010c260 80120d20 995d8210 0000000b 995d9e70 00000008 7f000000 00000000
[ 2697.578776] 9ce0: 80c08990 809703e0 65170668 33306131 20333136 34303565 34303033 37356620
[ 2697.586954] 9d00: 34306666 35652065 30303431 28206330 35393565 38383032 80002029 80207dc4
[... 301 lines cut ...]
[ 2708.997992] [a18361a1] CMD  19 200
[ 2709.001391] [a18361a2] REQ> b680e034 0
[ 2709.005136] [a226168e] TIM< 0 0
[ 2709.008280] mmc0: sbc op 23 arg 0x8 flags 0x15 - resp 00000000 00000000 00000000 00000000, err 0
[ 2709.017064] mmc0: cmd op 25 arg 0xdf5758 flags 0xb5 - resp 00000900 00000000 00000000 00000000, err 0
[ 2709.026281] mmc0: data blocks 8 blksz 200 - err 0
[ 2709.030984] mmc0: stop op 12 arg 0x0 flags 0x49d - resp 00000000 00000000 00000000 00000000, err 0
[ 2709.039937] mmc0: =========== REGISTER DUMP ===========
[ 2709.045157] mmc0: SDCMD  0x00000099
[ 2709.048640] mmc0: SDARG  0x00df5758
[ 2709.052122] mmc0: SDTOUT 0x017d7840
[ 2709.055605] mmc0: SDCDIV 0x00000003
[ 2709.059088] mmc0: SDRSP0 0x00000900
[ 2709.062570] mmc0: SDRSP1 0x00001918
[ 2709.066054] mmc0: SDRSP2 0xffffffff
[ 2709.069537] mmc0: SDRSP3 0x0002400f
[ 2709.073020] mmc0: SDHSTS 0x00000000
[ 2709.076502] mmc0: SDVDD  0x00000001
[ 2709.079985] mmc0: SDEDM  0x00010801
[ 2709.083468] mmc0: SDHCFG 0x0000040e
[ 2709.086951] mmc0: SDHBCT 0x00000200
[ 2709.090433] mmc0: SDHBLC 0x00000001
[ 2709.093915] mmc0: ===========================================
```
From time to time, there are also `[drm] Resetting GPU` messages in the kernel log. In the config.txt, `gpu_mem` is set to 128.

pelwell, 2017-10-03:
Clusters of apparently unrelated errors like this are usually either due to collateral damage caused by the first crash, memory corruption or an inadequate power supply. 

Check that your power supply is good - `vcgencmd get_throttled`should return `0x0` under representative conditions - and if it is then please report this issue in Eric Anholt's dedicated VC4 repo [here](https://github.com/anholt/linux/issues).

anyc, 2017-10-04:
Thank you! Looks like someone else has this problem too: https://github.com/anholt/linux/issues/114

closed 2017-10-04 by anyc

pelwell, 2017-10-04:
Be advised - Eric isn't always very responsive, but he will get round to it eventually.

lategoodbye, 2017-10-04:
@anyc Did my patch fixes the issue for you?

anyc, 2017-10-04:
It is now running for over an hour without errors - only some "Resetting GPU" messages. Looks good already but I'll keep it running non-stop for an afternoon on the weekend to be sure. Thank you!

lategoodbye, 2017-10-08:
Does the issue occur without the Qt app?
How long does it take to reproduce the issue?
Is there anything else connected to the Pi (USB, I2C, ...)?

anyc, 2017-10-08:
I will keep it running without the app. It's running now for about 1.5 hours.

According to my current backlog, the longest time without issues was 2.5 hours. It varies randomly, I'd say.

Only a TV, WiFi usb stick and a uart2usb adapter to another RPi. The other RPi uses kernel 4.9.36 that I used on the faulty RPi before, too.

lategoodbye, 2017-10-08:
In case the issue doesn't occur without app, try to remove WiFi usb stick and uart2usb adapter but use the app again.

anyc, 2017-10-08:
It also stopped with an oops eventually. :/ The GUI app did run for a short time in the beginning though as it is started automatically. I switch the SD cards now and check if the issue occurs also on the other RPi.
What puzzles me is that it always segfaults in the same place in the vc4 module, especially as it creates exactly the same log as in the other issue report.

lategoodbye, 2017-10-08:
Since your issue doesn't occur during boot, i don't think it's the same issue. In your case we have some kind of corruption which overwrite the spin lock structure.

anyc, 2017-10-09:
I just noticed during the last two oops that ~50 ms before the initial "Unable to handle kernel NULL pointer dereference" message there is a "[drm] Resetting GPU." message.

anyc, 2017-10-12:
What would be the best way to debug this? Setting a watchpoint on the structure? Would this be possible on the RPi over UART? I have not much experience with kernel debugging except printk debug output.

anyc, 2017-10-13:
The same thing also happens with an unmodified raspbian. Shall I open an issue in Eric's repo? Maybe he has an idea?