raspberrypi/linux #2219: CM3 /dev/mmcblk0p1 changes to /dev/mmcblk1p1 between 4.9 and 4.13 kernels

chewitt, 2017-10-08:
I'm booting LibreELEC on a CM3 equipped "Slice" box which has kernel etc. the same as our RPi image. On a current LE release with 4.9 kernel the CM eMMC storage shows as /dev/mmcblk0p1 and /dev/mmcblk0p2 but on a development image with 4.13 kernel the nodes are /dev/mmcblk1p1 and /dev/mmcblk1p2 so first boot after OS update fails as cmdline.txt has 4.9 details. I'm guessing there's a change to device tree or kernel, but not being an expert in either I haven't struck gold searching in Google or GitHub. Any help or pointers on how to solve this are greatly appreciated!

Gadgetoid, 2017-10-13:
Wasn't cmdline.txt updated recently ish to use partUUID instead of explicit (and variable) device node names?

I imagine switching to partUUID would allow you to update without incident.

chewitt, 2017-10-15:
All current Slice installs have /dev/mmcblk0p1 in cmdline.txt and this cannot be changed until after the box boots into the new image (update scripts are from the existing image, not the new one). If we rewrite to use UUID or LABEL it only affects new installs. For update to work for existing users, we have to ensure /dev/mmcblk0p1 exists.

pelwell, 2017-10-15:
From around 4.11 the upstream kernel started using different rules for assigning mmc IDs. At that time we applied a downstream patch (that was rejected upstream) that uses DT aliases to choose the IDs and another to add those aliases. As of rpi-4.13.y that patch got squashed into a composite patch adding the bcm2835-mmc driver, but it still exists, as do the required DT aliases.

You need to look for `mmc0` and `mmc1` in the CM3 DTB, and drivers/mmc/core/host.c should contain the line:
```
id = of_alias_get_highest_id("mmc");
```

chewitt, 2017-11-18:
There are no references to mmc0/mmc1 in bcm2710-rpi-cm3.dtb but if I follow the chain of includes the bcm2708-rpi.dtsi file has a bunch of aliases that look interesting. Would the following patch on that file work?

```--- a/arch/arm/boot/dts/bcm2708-rpi.dtsi	2017-11-18 04:56:28.149405247 +0000
+++ b/arch/arm/boot/dts/bcm2708-rpi.dtsi	2017-11-18 05:35:40.265159491 +0000
@@ -21,7 +21,7 @@
 		gpio = &gpio;
 		uart0 = &uart0;
 		sdhost = &sdhost;
-		mmc0 = &sdhost;
+		mmc1 = &sdhost;
 		i2s = &i2s;
 		spi0 = &spi0;
 		i2c0 = &i2c0;
@@ -29,7 +29,7 @@
 		spi1 = &spi1;
 		spi2 = &spi2;
 		mmc = &mmc;
-		mmc1 = &mmc;
+		mmc0 = &mmc;
 		i2c1 = &i2c1;
 		i2c2 = &i2c2;
 		usb = &usb;```

or am I barking up the wrong tree .. literally?

chewitt, 2017-11-18:
Having tested this it works (solves the [re]boot issue) but is it possible to make this change in the slice .dts file instead of patching the kernel? .. we prefer to not patch things when possible.

pelwell, 2017-11-18:
Yes. Put an aliases node in the slice .dts, containing just the properties you want to change. Provided the inclusion order is correct (and it should be) this will overwrite the earlier definitions and have the desired effect.

chewitt, 2017-11-18:
https://github.com/LibreELEC/slice-firmware/blob/master/slice3-dt-blob.dts is the file I'm trying to modify, and this is the patch I have attempted to apply:

```
--- a/slice3-dt-blob.dts	2017-06-27 17:59:57.000000000 +0100
+++ b/slice3-dt-blob.dts	2017-11-18 07:29:34.300302472 +0000
@@ -2,6 +2,11 @@
 
 / {
 
+	aliases {
+		mmc0 = &mmc;
+		mmc1 = &sdhost;
+	}
+
 	videocore {
 
 		pins_cm3 {
```

but when compiling the dtb I get this error:

```
Error: slice3-dt-blob.dts:10.8-17 syntax error
FATAL ERROR: Unable to parse input tree
```

and I'm out of ideas :(

pelwell, 2017-11-18:
I think you are just missing a semicolon after the aliases block - this isn't C.

chewitt, 2017-11-18:
I added the missing ; but dtc still complains about missing &mmc and &sdhost nodes. It did hint that you can use -f to force output, which I tried, but the resulting LE image didn't boot and I had to flash a new image to the emmc to revert things back to a working state again.

pelwell, 2017-11-18:
That's the wrong dts file - dt-blob files aren't really Device Tree. You can either modify the slice overlay (https://github.com/FiveNinjas/linux/blob/slice-4.0.y/arch/arm/boot/dts/overlays/slice-overlay.dts) to swap the aliases, or create your own overlay. The former would look like this:
```
     };
   };
 
+  fragment@7 {
+    target-path = "/aliases";
+    __overlay__ {
+      mmc0 = &mmc;
+      mmc1 = &sdhost;
+    };
+  };

 };
```
and the latter:
```
/dts-v1/;
/plugin/;

/ {
  compatible = "brcm,bcm2708";

  fragment@0 {
    target-path = "/aliases";
    __overlay__ {
      mmc0 = &mmc;
      mmc1 = &sdhost;
    };
  };
};
```
You will need to include the `-@` parameter when invoking dtc so it knows to expect unresolved symbols.

JamesH65, 2017-12-05:
@chewitt Anything further on this one? Can it be closed?

chewitt, 2017-12-05:
I didn't quite figure out the 'proper' way to do this but our Slice userbase is small and so the current working hack is okay if I never quite get around to it. Thanks for the pointers and support!

closed 2017-12-05 by chewitt