raspberrypi/linux #2221: Compute module GPIO 47 used for ACT LED

willnewton, 2017-10-10:
I noticed that in ```bcm2708-rpi-cm.dtsi``` GPIO 47 is used for the ACT LED:

        act_led: act {
                label = "led0";
                linux,default-trigger = "mmc0";
                gpios = <&gpio 47 0>;
        };

This seems to contradict the docs which indicate that this GPIO should only be used for eMMC enable.

6by9, 2017-10-10:
It is used for both purposes.

Look at the right hand side of page 2 of https://www.raspberrypi.org/documentation/hardware/computemodule/RPI-CMIO-V1_2-SCHEMATIC.pdf. GPIO47_1V8 is connected to the gate of Q2. If enabled it will make D2 (the ACT led) illuminate.

As documented immediately above that bit of schematic, if J4 has been set to 1-2 to enable USB boot, GPIO47_1V8 will be pulled high by default and therefore making Q2 conduct, then EMMC_DISABLE_N will be low via Q1 and Q2. D2 will also be illuminated, but that is irrelevant.
Once the rpiboot code has been downloaded it will set GPIO47_1V8 high, Q2 stops conducting, EMMC_DISABLE_N goes high, and the EMMC is enabled to allow it to be programmed. Linux isn't running, so nothing will be changing GPIO47_1V8 trying to set the LED, so the EMMC remains enabled.

If J4 has been set to 2-3 for normal boot, then Q1 never conducts and we can do whatever we like with GPIO47_1V8/Q2/D2 without affecting EMMC_DISABLE_N.

Specifically which docs are you referring to that say it should _only_ be used for eMMC enable?

willnewton, 2017-10-10:
My concern is a practical one. If you usbboot a CM then the eMMC is not accessible unless this GPIO is driven low.

Page 17 of this document indicates this GPIO should only be used for the purpose of enabling the eMMC:

https://www.raspberrypi.org/documentation/hardware/computemodule/RPI-CM-DATASHEET-V1_0.pdf

6by9, 2017-10-10:
You can't boot from a USB MSD on a CM, only a CM3 which is slightly different.
Probably one for @ghollingworth to work through all the permutations of USB MSD booting a CM3 and then wanting access to the EMMC, but that probably entails setting the OTP bits to ignore the EMMC for boot purposes anyway.

Or are you talking about the ```rpiboot``` tool in the usbboot repo? In which case it knows how to handle those GPIOs. Linux is not loaded at that point, so having the config in device tree is irrelevant.

willnewton, 2017-10-10:
I am talking about booting a kernel with rpiboot (from the usbboot repo). That kernel will not be able to access the eMMC on the CM in my tests unless the act_led section is removed from the DTS and replaced with code to drive the pin low.

ghollingworth, 2017-10-10:
Yes you'll need to remove the act_led section otherwise the eMMC will get disabled as soon as the LED is enabled from Linux

ghollingworth, 2017-10-10:
Yes you'll need to remove the act_led section...  Otherwise when linux touches the led it'll disable the eMMC.

JamesH65, 2017-12-05:
@willnewton OK to close this?

willnewton, 2017-12-05:
OK by me, although I wonder if the docs should be updated.

JamesH65, 2017-12-05:
If you flag up the doc to change, and what the change needs to be, I can do that. Or you can submit a PR on the docs github with the changes.

willnewton, 2017-12-05:
The document linked above contains this paragraph:

```
Note that the HDMHPDN1V8 and EMMCENN1V8 pins (on CM1 these were called GPIO461V8
and GPIO471V8 respectively) are 1.8V IO and are used for special functions (HDMI hot plug de-
tect and boot control respectively).   Please do not use these pins for any other purpose,  as the
software for the Compute Module will always expect these pins to have these special functions. If
they are unused please leave them unconnected.
```

Which seems to be at odds with what is actually done in practice ("any other purpose" seems to cover driving an LED).

JamesH65, 2017-12-05:
@ghollingworth Gordon, have you any preferences for changes to the docs to fix this?

pelwell, 2017-12-05:
On a practical note, adding `dtparam=act_led_trigger=none` to config.txt will hold the pin low.

ghollingworth, 2017-12-05:
I'm relatively ambivalent, if we change documentation then it should be documentation for CMIO board (i.e. the schematic) only since otherwise the documentation is correct, you shouldn't use that pin unless you really know what you're doing...

As for the device tree, I'd suggest it shouldn't specify the LED since that's implementation specific (i.e. you can put your LED on any GPIO as you see fit)

Gordon

McLion-Himself, 2017-12-15:
Has this been solved?
I'm currently in the same boat and don't get that to run as expected.
I use aCM3L on the original IO-board (as a reference only) as well as on our own design of an adapted base board. 
The only way to have the SD visible in Raspbian after USB booting is as follows: To boot from USB do NOT set the boot mode using the Jumper. Instead pull the SD. **Immediately** after the delay/fallback to USB insert the SD for the kernel to see it upon boot.
I found no way to get access to the SD while having set the boot mode using the Jumper as described in the documentation because the BCM does not drive low EMMC EN N 1V8 soon enough to have the kernel see the SD while booting.
Either this timing need to be changed in the firmware or there must be a way to kick off device enumeration later to find/see the SD.
Any help would be highly appreciated.
Thanks, McL

McLion-Himself, 2017-12-15:
The solution posted [here ](https://www.raspberrypi.org/forums/viewtopic.php?f=98&t=199957) by @6by9 works.