raspberrypi/linux #2223: Trying to checkout a specific kernel source using sha1

az2022, 2017-10-11:
I am trying to build a specific version of kernel and the way I find the commit hash is to lookup the history of "Makefile" in the root directory directly from Github. For e.g. I need kernel version 4.9.47 and sha1 of Makefile I see from Github is 458ca52f1564938c158d271f45bce0bc6ede2b3f

Steps I follow:
> git clone https://github.com/raspberrypi/linux.git
> cd linux
> git checkout 458ca52f1564938c158d271f45bce0bc6ede2b3f

Once this is done I don't see presence of 'arch/arm/configs/bcm2709_defconfig' file. I don't understand why is this file missing at every commit of the Makefile? Should I be grabbing a different commit-hash to get the proper sources associated with 4.9.47?

pelwell, 2017-10-12:
That method doesn't work because we pull upstream commits into the kernel directly or via merges - we never rebase. This means that the ancestors to any upstream commits are only upstream commits.

Merge commits are like symbolic links - they live in one branch but point to commits in another. This allows you to import commits without rewriting them (as rebase does), so the commit hash doesn't change and the ancestry is preserved. If you have `gitk` available, try using it to view the history and you will see that the rpi branches and the main kernel branch exist side by side, with downstream commits interspersed between periodic merges of the upstream releases (although something odd happened with 4.9.53).

I suggest you scan the history for the merge commits, and use them as your checkpoints.

az2022, 2017-10-12:
Thanks for helping understand the process. Using gitk I was able to pick a commit done by raspberry-pi org after the upstream merge was done.

closed 2017-10-12 by az2022