raspberrypi/linux #2228: I2C SMBUS problem on Pi Zero W

efbick, 2017-10-13:
I am having a problem a problem with python SMBUS communicating with a HMC6343. I have code that I have written that works fine on a pi 3 but when I switch over to the pi Zero W it doesn't work. The device shows up on i2cdetect and when I request anything from the device, it only returns 0XFF. Is there something specific about the Zero that would be causing this problem? I modified the code to get it to work on Adafruits I2C library but that also sometimes fails to read the sensor. The pi zero is on kernel 4.9.35 and the pi 3 is on 4.4.21 v7. I didn't want to update the pi 3 for fear that would also stop working.

pelwell, 2017-10-13:
Swap the SD cards and see if the fault moves. Raspbian images should work on all Pi models unless you've done some board-specific customisation.

efbick, 2017-10-17:
So the Pi Zero SD card works on the Pi 3 but when putting the Pi 3 SD card in the Zero it never gets further than the rainbow screen. The Pi Zero SD in the Pi 3 did work with SMBUS. So is there something different about the Pi Zero's hardware with I2C?

pelwell, 2017-10-17:
I think it is more likely that the Pi3 card predates the Zero support, but you've got the most important result - the image that failed in the Zero worked in the 3, so it shouldn't be a matter of kernel versions.

The Zero should not be any different with respect to I2C, so I'm beginning to suspect a hardware fault. Can you check the wiring - dry joints on the header, etc.? And have you tried reading from the device using i2cget to rule out software funnies?

efbick, 2017-10-18:
So the wiring should be good. I used the same hat with the sensors on both the pi 3 and pi zero. I also tried to use it on a second pi zero and that one had the same problem. So i2cget does work and adafruits library does but sometimes crashes with an I/O error but that could be their library. Any other ideas?

pelwell, 2017-10-18:
Zero's don't come with headers so there was a possibility of bad soldering, but if you say it's OK I'll accept that.

Are you saying that i2cget returns valid data? That would point the finger at the library, but then why would it work on the 3B?

Take a look at [this guide to using ftrace to snoop I2C transactions](https://technolinchpin.wordpress.com/2017/08/07/ftracesnoop-i2c-bus-transactions/). The Pi kernel has the necessary config settings, so you just need to use the debugfs (/sys/kernel/debug/ interface.

You'll need to be root, so starting the session with a `sudo su` will save typing. Enable tracing, run the test commands, then copy the contents of the `trace` to a file. If you can run the same commands on a 3B and a Zero we might get some idea of what is going on.

efbick, 2017-10-18:
Turns out I may have read from my wrong class when reading the sensor last time. Now when I read from the sensor with the Pi Zero sd card in the Pi 3 with SMBUS I end up with the same problem. It returns 0xFF. So maybe it is actually a kernel problem. I traced it anyway and its shown below. The first set is from the adafruit library and the second is from SMBUS. The first few writes are initialization and then I request and read the heading 3 times using 50-01.

Adafruit

          python-1361  [000] ....  1744.416824: i2c_write: i2c-1 #0 a=019 f=0000 l=2 [75-01]
          python-1361  [000] ....  1744.417093: i2c_result: i2c-1 n=1 ret=1
          python-1361  [000] ....  1744.418680: i2c_write: i2c-1 #0 a=019 f=0000 l=2 [65-01]
          python-1361  [000] ....  1744.418918: i2c_result: i2c-1 n=1 ret=1
          python-1361  [000] ....  1744.420452: i2c_read: i2c-1 #0 a=019 f=0001 l=1
          python-1361  [000] ....  1744.420647: i2c_reply: i2c-1 #0 a=019 f=0001 l=1 [11]
          python-1361  [000] ....  1744.420653: i2c_result: i2c-1 n=1 ret=1
          python-1361  [000] ....  1744.423459: i2c_write: i2c-1 #0 a=019 f=0000 l=2 [73-01]
          python-1361  [000] ....  1744.423708: i2c_result: i2c-1 n=1 ret=1
          python-1361  [000] ....  1744.424551: i2c_write: i2c-1 #0 a=019 f=0000 l=2 [65-01]
          python-1361  [000] ....  1744.424785: i2c_result: i2c-1 n=1 ret=1
          python-1361  [000] ....  1744.426231: i2c_read: i2c-1 #0 a=019 f=0001 l=1
          python-1361  [000] ....  1744.426397: i2c_reply: i2c-1 #0 a=019 f=0001 l=1 [12]
          python-1361  [000] ....  1744.426405: i2c_result: i2c-1 n=1 ret=1
          python-1361  [000] ....  1768.649765: i2c_write: i2c-1 #0 a=019 f=0000 l=2 [50-01]
          python-1361  [000] ....  1768.650029: i2c_result: i2c-1 n=1 ret=1
          python-1361  [000] ....  1768.651847: i2c_read: i2c-1 #0 a=019 f=0001 l=1
          python-1361  [000] ....  1768.652033: i2c_reply: i2c-1 #0 a=019 f=0001 l=1 [08]
          python-1361  [000] ....  1768.652040: i2c_result: i2c-1 n=1 ret=1
          python-1361  [000] ....  1768.652379: i2c_read: i2c-1 #0 a=019 f=0001 l=1
          python-1361  [000] ....  1768.652578: i2c_reply: i2c-1 #0 a=019 f=0001 l=1 [e8]
          python-1361  [000] ....  1768.652587: i2c_result: i2c-1 n=1 ret=1
          python-1361  [000] ....  1768.652891: i2c_read: i2c-1 #0 a=019 f=0001 l=1
          python-1361  [000] ....  1768.653667: i2c_reply: i2c-1 #0 a=019 f=0001 l=1 [00]
          python-1361  [000] ....  1768.653677: i2c_result: i2c-1 n=1 ret=1
          python-1361  [000] ....  1768.654072: i2c_read: i2c-1 #0 a=019 f=0001 l=1
          python-1361  [000] ....  1768.654943: i2c_reply: i2c-1 #0 a=019 f=0001 l=1 [c3]
          python-1361  [000] ....  1768.654955: i2c_result: i2c-1 n=1 ret=1
          python-1361  [000] ....  1768.655345: i2c_read: i2c-1 #0 a=019 f=0001 l=1
          python-1361  [000] ....  1768.656449: i2c_reply: i2c-1 #0 a=019 f=0001 l=1 [03]
          python-1361  [000] ....  1768.656459: i2c_result: i2c-1 n=1 ret=1
          python-1361  [000] ....  1768.656872: i2c_read: i2c-1 #0 a=019 f=0001 l=1
          python-1361  [000] ....  1768.657827: i2c_reply: i2c-1 #0 a=019 f=0001 l=1 [18]
          python-1361  [000] ....  1768.657836: i2c_result: i2c-1 n=1 ret=1
          python-1361  [000] ....  1770.569743: i2c_write: i2c-1 #0 a=019 f=0000 l=2 [50-01]
          python-1361  [000] ....  1770.569997: i2c_result: i2c-1 n=1 ret=1
          python-1361  [000] ....  1770.571843: i2c_read: i2c-1 #0 a=019 f=0001 l=1
          python-1361  [000] ....  1770.572033: i2c_reply: i2c-1 #0 a=019 f=0001 l=1 [08]
          python-1361  [000] ....  1770.572042: i2c_result: i2c-1 n=1 ret=1
          python-1361  [000] ....  1770.572382: i2c_read: i2c-1 #0 a=019 f=0001 l=1
          python-1361  [000] ....  1770.572554: i2c_reply: i2c-1 #0 a=019 f=0001 l=1 [e6]
          python-1361  [000] ....  1770.572560: i2c_result: i2c-1 n=1 ret=1
          python-1361  [000] ....  1770.572870: i2c_read: i2c-1 #0 a=019 f=0001 l=1
          python-1361  [000] ....  1770.573690: i2c_reply: i2c-1 #0 a=019 f=0001 l=1 [00]
          python-1361  [000] ....  1770.573705: i2c_result: i2c-1 n=1 ret=1
          python-1361  [000] ....  1770.574102: i2c_read: i2c-1 #0 a=019 f=0001 l=1
          python-1361  [000] ....  1770.574650: i2c_reply: i2c-1 #0 a=019 f=0001 l=1 [c4]
          python-1361  [000] ....  1770.574662: i2c_result: i2c-1 n=1 ret=1
          python-1361  [000] ....  1770.575027: i2c_read: i2c-1 #0 a=019 f=0001 l=1
          python-1361  [000] ....  1770.575911: i2c_reply: i2c-1 #0 a=019 f=0001 l=1 [03]
          python-1361  [000] ....  1770.575923: i2c_result: i2c-1 n=1 ret=1
          python-1361  [000] ....  1770.576460: i2c_read: i2c-1 #0 a=019 f=0001 l=1
          python-1361  [000] ....  1770.577826: i2c_reply: i2c-1 #0 a=019 f=0001 l=1 [16]
          python-1361  [000] ....  1770.577838: i2c_result: i2c-1 n=1 ret=1
          python-1361  [000] ....  1771.016975: i2c_write: i2c-1 #0 a=019 f=0000 l=2 [50-01]
          python-1361  [000] ....  1771.017218: i2c_result: i2c-1 n=1 ret=1
          python-1361  [000] ....  1771.018839: i2c_read: i2c-1 #0 a=019 f=0001 l=1
          python-1361  [000] ....  1771.019029: i2c_reply: i2c-1 #0 a=019 f=0001 l=1 [08]
          python-1361  [000] ....  1771.019035: i2c_result: i2c-1 n=1 ret=1
          python-1361  [000] ....  1771.019336: i2c_read: i2c-1 #0 a=019 f=0001 l=1
[... cut ...]
          python-1423  [000] ....  1893.278571: i2c_read: i2c-1 #1 a=019 f=0001 l=6
          python-1423  [000] ....  1893.279163: i2c_reply: i2c-1 #1 a=019 f=0001 l=6 [ff-ff-ff-ff-ff-ff]
          python-1423  [000] ....  1893.279173: i2c_result: i2c-1 n=2 ret=2
          python-1423  [000] ....  1893.279184: smbus_reply: i2c-1 a=019 f=0000 c=0 I2C_BLOCK_DATA l=7 [06-ff-ff-ff-ff-ff-ff]
          python-1423  [000] ....  1893.279189: smbus_result: i2c-1 a=019 f=0000 c=0 I2C_BLOCK_DATA rd res=0

efbick, 2017-10-18:
This is the same SMBUS code running on the Pi 3 with its SD card.

          python-1098  [001] ....   264.597801: smbus_write: i2c-1 a=019 f=0000 c=75 BYTE_DATA l=1 [01]
          python-1098  [001] ....   264.597823: i2c_write: i2c-1 #0 a=019 f=0000 l=2 [75-01]
          python-1098  [001] ....   264.598039: i2c_result: i2c-1 n=1 ret=1
          python-1098  [001] ....   264.598045: smbus_result: i2c-1 a=019 f=0000 c=75 BYTE_DATA wr res=0
          python-1098  [001] ....   264.618529: smbus_write: i2c-1 a=019 f=0000 c=65 BYTE_DATA l=1 [01]
          python-1098  [001] ....   264.618547: i2c_write: i2c-1 #0 a=019 f=0000 l=2 [65-01]
          python-1098  [001] ....   264.618775: i2c_result: i2c-1 n=1 ret=1
          python-1098  [001] ....   264.618783: smbus_result: i2c-1 a=019 f=0000 c=65 BYTE_DATA wr res=0
          python-1098  [001] ....   264.628950: smbus_read: i2c-1 a=019 f=0000 c=0 I2C_BLOCK_DATA
          python-1098  [001] ....   264.628965: i2c_write: i2c-1 #0 a=019 f=0000 l=1 [00]
          python-1098  [001] ....   264.628969: i2c_read: i2c-1 #1 a=019 f=0001 l=1
          python-1098  [001] ....   264.629224: i2c_reply: i2c-1 #1 a=019 f=0001 l=1 [11]
          python-1098  [001] ....   264.629228: i2c_result: i2c-1 n=2 ret=2
          python-1098  [001] ....   264.629234: smbus_reply: i2c-1 a=019 f=0000 c=0 I2C_BLOCK_DATA l=2 [01-11]
          python-1098  [001] ....   264.629238: smbus_result: i2c-1 a=019 f=0000 c=0 I2C_BLOCK_DATA rd res=0
          python-1098  [001] ....   264.639519: smbus_write: i2c-1 a=019 f=0000 c=73 BYTE_DATA l=1 [01]
          python-1098  [001] ....   264.639529: i2c_write: i2c-1 #0 a=019 f=0000 l=2 [73-01]
          python-1098  [001] ....   264.639719: i2c_result: i2c-1 n=1 ret=1
          python-1098  [001] ....   264.639723: smbus_result: i2c-1 a=019 f=0000 c=73 BYTE_DATA wr res=0
          python-1098  [001] ....   264.659946: smbus_write: i2c-1 a=019 f=0000 c=65 BYTE_DATA l=1 [01]
          python-1098  [001] ....   264.659951: i2c_write: i2c-1 #0 a=019 f=0000 l=2 [65-01]
          python-1098  [001] ....   264.660135: i2c_result: i2c-1 n=1 ret=1
          python-1098  [001] ....   264.660138: smbus_result: i2c-1 a=019 f=0000 c=65 BYTE_DATA wr res=0
          python-1098  [001] ....   264.670236: smbus_read: i2c-1 a=019 f=0000 c=0 I2C_BLOCK_DATA
          python-1098  [001] ....   264.670240: i2c_write: i2c-1 #0 a=019 f=0000 l=1 [00]
          python-1098  [001] ....   264.670243: i2c_read: i2c-1 #1 a=019 f=0001 l=1
          python-1098  [001] ....   264.670474: i2c_reply: i2c-1 #1 a=019 f=0001 l=1 [12]
          python-1098  [001] ....   264.670477: i2c_result: i2c-1 n=2 ret=2
          python-1098  [001] ....   264.670480: smbus_reply: i2c-1 a=019 f=0000 c=0 I2C_BLOCK_DATA l=2 [01-12]
          python-1098  [001] ....   264.670482: smbus_result: i2c-1 a=019 f=0000 c=0 I2C_BLOCK_DATA rd res=0
          python-1098  [001] ....   269.558048: smbus_write: i2c-1 a=019 f=0000 c=50 BYTE_DATA l=1 [01]
          python-1098  [001] ....   269.558066: i2c_write: i2c-1 #0 a=019 f=0000 l=2 [50-01]
          python-1098  [001] ....   269.558275: i2c_result: i2c-1 n=1 ret=1
          python-1098  [001] ....   269.558283: smbus_result: i2c-1 a=019 f=0000 c=50 BYTE_DATA wr res=0
          python-1098  [001] ....   269.568427: smbus_read: i2c-1 a=019 f=0000 c=0 I2C_BLOCK_DATA
          python-1098  [001] ....   269.568440: i2c_write: i2c-1 #0 a=019 f=0000 l=1 [00]
          python-1098  [001] ....   269.568444: i2c_read: i2c-1 #1 a=019 f=0001 l=6
          python-1098  [001] ....   269.569006: i2c_reply: i2c-1 #1 a=019 f=0001 l=6 [04-ed-00-90-fc-99]
          python-1098  [001] ....   269.569011: i2c_result: i2c-1 n=2 ret=2
          python-1098  [001] ....   269.569016: smbus_reply: i2c-1 a=019 f=0000 c=0 I2C_BLOCK_DATA l=7 [06-04-ed-00-90-fc-99]
          python-1098  [001] ....   269.569021: smbus_result: i2c-1 a=019 f=0000 c=0 I2C_BLOCK_DATA rd res=0
          python-1098  [001] ....   270.646133: smbus_write: i2c-1 a=019 f=0000 c=50 BYTE_DATA l=1 [01]
          python-1098  [001] ....   270.646155: i2c_write: i2c-1 #0 a=019 f=0000 l=2 [50-01]
          python-1098  [001] ....   270.646366: i2c_result: i2c-1 n=1 ret=1
          python-1098  [001] ....   270.646372: smbus_result: i2c-1 a=019 f=0000 c=50 BYTE_DATA wr res=0
          python-1098  [001] ....   270.656511: smbus_read: i2c-1 a=019 f=0000 c=0 I2C_BLOCK_DATA
          python-1098  [001] ....   270.656525: i2c_write: i2c-1 #0 a=019 f=0000 l=1 [00]
          python-1098  [001] ....   270.656528: i2c_read: i2c-1 #1 a=019 f=0001 l=6
          python-1098  [001] ....   270.657090: i2c_reply: i2c-1 #1 a=019 f=0001 l=6 [04-ea-00-91-fc-9c]
          python-1098  [001] ....   270.657094: i2c_result: i2c-1 n=2 ret=2
          python-1098  [001] ....   270.657099: smbus_reply: i2c-1 a=019 f=0000 c=0 I2C_BLOCK_DATA l=7 [06-04-ea-00-91-fc-9c]
          python-1098  [001] ....   270.657105: smbus_result: i2c-1 a=019 f=0000 c=0 I2C_BLOCK_DATA rd res=0
          python-1098  [001] ....   271.430508: smbus_write: i2c-1 a=019 f=0000 c=50 BYTE_DATA l=1 [01]
          python-1098  [001] ....   271.430532: i2c_write: i2c-1 #0 a=019 f=0000 l=2 [50-01]
          python-1098  [001] ....   271.430742: i2c_result: i2c-1 n=1 ret=1
          python-1098  [001] ....   271.430748: smbus_result: i2c-1 a=019 f=0000 c=50 BYTE_DATA wr res=0
[... cut ...]
          python-1098  [001] ....   271.440907: i2c_read: i2c-1 #1 a=019 f=0001 l=6
          python-1098  [001] ....   271.441470: i2c_reply: i2c-1 #1 a=019 f=0001 l=6 [04-e6-00-92-fc-99]
          python-1098  [001] ....   271.441474: i2c_result: i2c-1 n=2 ret=2
          python-1098  [001] ....   271.441481: smbus_reply: i2c-1 a=019 f=0000 c=0 I2C_BLOCK_DATA l=7 [06-04-e6-00-92-fc-99]
          python-1098  [001] ....   271.441484: smbus_result: i2c-1 a=019 f=0000 c=0 I2C_BLOCK_DATA rd res=0

pelwell, 2017-10-19:
One of the changes in the newer kernels is a switch to the upstream I2C driver. You can revert to the old, downstream driver by adding the following to config.txt:
```
dtoverlay=i2c-bcm2708
```
Try that on the Pi Zero and see if it makes a difference.

Also, please run `raspi-gpio get` and post the output.

efbick, 2017-10-19:
So I tried dtoverlay before. SMBUS crashes with a IO error below.

h=HMC6343()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "HMC6343.py", line 97, in __init__
    self.selectMode(self.ENTER_RUN)
  File "HMC6343.py", line 342, in selectMode
    OPMode1 = self.readOPMode1()
  File "HMC6343.py", line 276, in readOPMode1
    readValues = bus.read_i2c_block_data(self.I2C_ADDR,0,self.BLEN_EEPROM_REG)
IOError: [Errno 5] Input/output error

          python-1054  [000] ....   342.771400: smbus_write: i2c-1 a=019 f=0000 c=75 BYTE_DATA l=1 [01]
          python-1054  [000] ....   342.771429: i2c_write: i2c-1 #0 a=019 f=0000 l=2 [75-01]
          python-1054  [000] ....   342.772162: i2c_result: i2c-1 n=1 ret=1
          python-1054  [000] ....   342.772228: smbus_result: i2c-1 a=019 f=0000 c=75 BYTE_DATA wr res=0
          python-1054  [000] ....   342.793185: smbus_write: i2c-1 a=019 f=0000 c=65 BYTE_DATA l=1 [01]
          python-1054  [000] ....   342.793208: i2c_write: i2c-1 #0 a=019 f=0000 l=2 [65-01]
          python-1054  [000] ....   342.793949: i2c_result: i2c-1 n=1 ret=1
          python-1054  [000] ....   342.793964: smbus_result: i2c-1 a=019 f=0000 c=65 BYTE_DATA wr res=0
          python-1054  [000] ....   342.804368: smbus_read: i2c-1 a=019 f=0000 c=0 I2C_BLOCK_DATA
          python-1054  [000] ....   342.804390: i2c_write: i2c-1 #0 a=019 f=0000 l=1 [00]
          python-1054  [000] ....   342.804398: i2c_read: i2c-1 #1 a=019 f=0001 l=1
          python-1054  [000] ....   342.805192: i2c_result: i2c-1 n=0 ret=-5
          python-1054  [000] ....   342.805209: smbus_reply: i2c-1 a=019 f=0000 c=0 I2C_BLOCK_DATA l=2 [01-83]
          python-1054  [000] ....   342.805214: smbus_result: i2c-1 a=019 f=0000 c=0 I2C_BLOCK_DATA rd res=-5

This is the output of raspi-gpio get

BANK0 (GPIO 0 to 27):
  GPIO 00: level=1 fsel=0 alt=  func=INPUT
  GPIO 01: level=1 fsel=0 alt=  func=INPUT
  GPIO 02: level=1 fsel=4 alt=0 func=SDA1
  GPIO 03: level=1 fsel=4 alt=0 func=SCL1
  GPIO 04: level=0 fsel=0 alt=  func=INPUT
  GPIO 05: level=1 fsel=0 alt=  func=INPUT
  GPIO 06: level=1 fsel=0 alt=  func=INPUT
  GPIO 07: level=1 fsel=1 alt=  func=OUTPUT
  GPIO 08: level=1 fsel=1 alt=  func=OUTPUT
  GPIO 09: level=0 fsel=4 alt=0 func=SPI0_MISO
  GPIO 10: level=0 fsel=4 alt=0 func=SPI0_MOSI
  GPIO 11: level=0 fsel=4 alt=0 func=SPI0_SCLK
  GPIO 12: level=0 fsel=0 alt=  func=INPUT
  GPIO 13: level=0 fsel=0 alt=  func=INPUT
  GPIO 14: level=1 fsel=2 alt=5 func=TXD1
  GPIO 15: level=1 fsel=2 alt=5 func=RXD1
  GPIO 16: level=0 fsel=0 alt=  func=INPUT
  GPIO 17: level=0 fsel=0 alt=  func=INPUT
  GPIO 18: level=0 fsel=0 alt=  func=INPUT
  GPIO 19: level=0 fsel=0 alt=  func=INPUT
  GPIO 20: level=0 fsel=0 alt=  func=INPUT
  GPIO 21: level=0 fsel=0 alt=  func=INPUT
  GPIO 22: level=0 fsel=0 alt=  func=INPUT
  GPIO 23: level=0 fsel=0 alt=  func=INPUT
  GPIO 24: level=1 fsel=0 alt=  func=INPUT
  GPIO 25: level=1 fsel=0 alt=  func=INPUT
  GPIO 26: level=0 fsel=0 alt=  func=INPUT
  GPIO 27: level=0 fsel=0 alt=  func=INPUT
BANK1 (GPIO 28 to 45):
  GPIO 28: level=1 fsel=4 alt=0 func=SDA0
  GPIO 29: level=1 fsel=4 alt=0 func=SCL0
  GPIO 30: level=0 fsel=7 alt=3 func=CTS0
  GPIO 31: level=0 fsel=7 alt=3 func=RTS0
  GPIO 32: level=1 fsel=7 alt=3 func=TXD0
  GPIO 33: level=1 fsel=7 alt=3 func=RXD0
  GPIO 34: level=0 fsel=7 alt=3 func=SD1_CLK
  GPIO 35: level=1 fsel=7 alt=3 func=SD1_CMD
  GPIO 36: level=1 fsel=7 alt=3 func=SD1_DAT0
  GPIO 37: level=1 fsel=7 alt=3 func=SD1_DAT1
  GPIO 38: level=1 fsel=7 alt=3 func=SD1_DAT2
  GPIO 39: level=1 fsel=7 alt=3 func=SD1_DAT3
  GPIO 40: level=0 fsel=1 alt=  func=OUTPUT
  GPIO 41: level=1 fsel=1 alt=  func=OUTPUT
  GPIO 42: level=0 fsel=0 alt=  func=INPUT
  GPIO 43: level=1 fsel=4 alt=0 func=GPCLK2
  GPIO 44: level=0 fsel=1 alt=  func=OUTPUT
  GPIO 45: level=1 fsel=1 alt=  func=OUTPUT
BANK2 (GPIO 46 to 53):
  GPIO 46: level=0 fsel=0 alt=  func=INPUT
  GPIO 47: level=0 fsel=1 alt=  func=OUTPUT
  GPIO 48: level=0 fsel=4 alt=0 func=SD0_CLK
  GPIO 49: level=1 fsel=4 alt=0 func=SD0_CMD
  GPIO 50: level=1 fsel=4 alt=0 func=SD0_DAT0
  GPIO 51: level=1 fsel=4 alt=0 func=SD0_DAT1
  GPIO 52: level=1 fsel=4 alt=0 func=SD0_DAT2
  GPIO 53: level=1 fsel=4 alt=0 func=SD0_DAT3

pelwell, 2017-10-19:
The GPIO state all looks reasonable.

The i2c_bcm2708 driver has a module parameter that enables combined transactions, making it more like the new driver. You can check the current value and/or change it like this:
```
$ cat /sys/module/i2c_bcm2708/parameters/combined
$ sudo sh -c '/bin/echo Y > /sys/module/i2c_bcm2708/parameters/combined'
$ sudo sh -c '/bin/echo N > /sys/module/i2c_bcm2708/parameters/combined'
```
Changes take effect immediately.

By default, `combined` should be disabled. If you find that enabling it doesn't prevent the -5 errors, or if the driver then fails in the same way as the new one on the Zero, then I've run out of things to suggest unless you can attach a logic analyser to the signals. If you don't have a logic analyser (or oscilloscope), you may be able to make use of [piscope](http://abyz.me.uk/rpi/pigpio/piscope.html), particularly if you can slow the I2C bus down.

In fact, slowing the bus down (or speeding it up) might not be a bad idea anyway just to confirm that it isn't a timing problem:
```
dtparam=i2c_arm_baudrate=20000
```
The default value is 100000 (100KHz).

efbick, 2017-10-19:
So I have tried changing the baud rate. That didn't seem to help. So there isn't an i2c_bcm2708 inside module. There is only a i2c_2835 and an i2c_dev. When looking inside i2c_2835/parameters combined does not exist.

pelwell, 2017-10-19:
You need to apply the i2c-bcm2708 overlay to switch the drivers.

efbick, 2017-10-26:
Yeah, I must have changed it back. So after overlaying i2c-bcm2708 the default combined is Y. So by switching it to N the library I have with SMBUS now works. Thanks. So what exactly is the problem and how would I go about adding this to boot config? It resets to Y after every reboot.

pelwell, 2017-10-26:
The problem is clearly triggered by the kernel taking a write followed by a read and combining them into a single transaction, allowing it to squash a stop followed by start into a restart.

There are two known bugs in the I2C controller. Both are discussed in this old issue: https://github.com/raspberrypi/linux/issues/254
I'm not going to attempt to summarise the conclusions or suggest a more permanent way forward until I've thoroughly grokked the thread myself.

You can permanently disable the combined trasnsactions by creating a .conf file in /etc/modprobe.d:
```
$ sudo sh -c "echo options i2c_bcm2708 combined=N > /etc/modprobe.d/i2c_bcm2708.conf"
```
However, since the driver default is for them to be disabled you must already have a conf file enabling them:
```
$ grep combined /etc/modprobe.d/*
```
If so, remove that line or delete the entire file.

efbick, 2017-10-26:
Alright thank you. I appreciate the help on this.

closed 2017-10-26 by efbick