raspberrypi/linux #2230: USB/8192CU Wifi with stretch

Chris56, 2017-10-16:
Bus 001 Device 009: ID 0bda:8178 Realtek Semiconductor Corp. RTL8192CU 802.11n WLAN Adapter
The hardware is ok, as with raspian wheezy on the same box everything works flawlessly.

With a fresh vanilla stretch raspian install (4.9.41+, Model B Rev 2) kernel shows:
[ 21.917607] ------------[ cut here ]------------
[ 21.918736] WARNING: CPU: 0 PID: 18 at net/wireless/sme.c:948 __cfg80211_disconnected+0x32c/0x378 [cfg80211]
[ 21.918748] Modules linked in: 8192cu hid_logitech_dj cfg80211 rfkill snd_bcm2835 snd_pcm snd_timer snd bcm2835_gpiomem uio_pdrv_genirq uio fixed ip_tables x_tables ipv6
[ 21.918827] CPU: 0 PID: 18 Comm: kworker/0:1 Not tainted 4.9.41+ #1023
[ 21.918833] Hardware name: BCM2835
[ 21.918869] Workqueue: usb_hub_wq hub_event
[ 21.918939] [] (unwind_backtrace) from [] (show_stack+0x20/0x24)
[ 21.918971] [] (show_stack) from [] (dump_stack+0x20/0x28)
[ 21.919001] [] (dump_stack) from [] (__warn+0xe4/0x10c)
[ 21.919024] [] (__warn) from [] (warn_slowpath_null+0x30/0x38)
[ 21.919982] [] (warn_slowpath_null) from [] (__cfg80211_disconnected+0x32c/0x378 [cfg80211])
[ 21.921660] [] (__cfg80211_disconnected [cfg80211]) from [] (cfg80211_process_wdev_events+0x1a4/0x210 [cfg80211])
[ 21.923650] [] (cfg80211_process_wdev_events [cfg80211]) from [] (cfg80211_netdev_notifier_call+0x328/0x574 [cfg80211])
[ 21.924774] [] (cfg80211_netdev_notifier_call [cfg80211]) from [] (notifier_call_chain+0x54/0x94)
[ 21.924815] [] (notifier_call_chain) from [] (raw_notifier_call_chain+0x28/0x30)
[ 21.924844] [] (raw_notifier_call_chain) from [] (call_netdevice_notifiers_info+0x40/0x6c)
[ 21.924875] [] (call_netdevice_notifiers_info) from [] (rollback_registered_many+0x204/0x320)
[ 21.924897] [] (rollback_registered_many) from [] (rollback_registered+0x38/0x50)
[ 21.924922] [] (rollback_registered) from [] (unregister_netdevice_queue+0x64/0xc0)
[ 21.924943] [] (unregister_netdevice_queue) from [] (unregister_netdev+0x28/0x30)
[ 21.925718] [] (unregister_netdev) from [] (rtw_unregister_netdevs+0x54/0x74 [8192cu])
[ 21.926672] [] (rtw_unregister_netdevs [8192cu]) from [] (rtw_dev_remove+0x2c/0x7c [8192cu])
[ 21.927166] [] (rtw_dev_remove [8192cu]) from [] (usb_unbind_interface+0x80/0x288)
[ 21.927198] [] (usb_unbind_interface) from [] (__device_release_driver+0x94/0x118)
[ 21.927220] [] (__device_release_driver) from [] (device_release_driver+0x2c/0x38)
[ 21.927241] [] (device_release_driver) from [] (bus_remove_device+0xe4/0x114)
[ 21.927278] [] (bus_remove_device) from [] (device_del+0x118/0x22c)
[ 21.927301] [] (device_del) from [] (usb_disable_device+0xb0/0x1f4)
[ 21.927334] [] (usb_disable_device) from [] (usb_disconnect+0x7c/0x1fc)
[ 21.927355] [] (usb_disconnect) from [] (usb_disconnect+0x1b4/0x1fc)
[ 21.927378] [] (usb_disconnect) from [] (hub_event+0x594/0x10f4)
[ 21.927403] [] (hub_event) from [] (process_one_work+0x11c/0x398)
[ 21.927424] [] (process_one_work) from [] (worker_thread+0x3c/0x508)
[ 21.927453] [] (worker_thread) from [] (kthread+0xf8/0x114)
[ 21.927486] [] (kthread) from [] (ret_from_fork+0x14/0x2c)
[ 21.927501] ---[ end trace db32b3af32e9c85f ]---
Same with current: 4.9.56+

Once you try to actually use the wifi adapter (e.g. starting wpa_supplicant), it is difficult to use the USB keyboard anymore as most keypresses are being lost. Therefore, the usb bus gets stuck I guess.

Let me know in case additional information will be needed!
BR
Chris

JamesH65, 2017-10-16:
@P33M Any ideas?

P33M, 2017-10-17:
My initial response would be power-related - Model B rev 2 is much more sensitive than subsequent models of Pi to voltage variations, partly due to the static drop introduced by the polyfuse and partly due to lack of current limiter on the USB ports.

If Wheezy is performing correctly, can you try updating the kernel in isolation via `rpi-update`?

Chris56, 2017-10-17:
I did rpi-update to 4.9.56+.
Works flawlessly! As before with wheezy). The only kernel message I noticed with 4.9.56+:
TCP: wlan0: Driver has suspect GRO implementation, TCP performance may be compromised.

I don't think it's power related.
a) It has been running on exactly this hardware (including the (good) power supply) for a long time and still works with wheezy.
b) The problems also show with just only one load on usb.
c) A ralink chipset usb wifi stick works fine.
d) A different (vendor ASUS) realtek 8192CU usb wifi stick also doen't work with stretch but does work with wheezy.
e) I've seen elsewhere (can't remember where) that someone mentioned running jessie + 8192cu without problems, but a fresh stretch install did not work.

JamesH65, 2017-12-05:
@Chris56 Would it be possible to try the latest kernel (4.14) since this has quite a few changes for RealTek devices?

JamesH65, 2017-12-05:
You can get the latest kernel using rpi-update

sudo BRANCH=next rpi-update

Note this is bleeding edge stuff - make sure of backups.

Chris56, 2017-12-06:
I installed 4.14:
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.4+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1060 Tue Dec 5 15:34:09 GMT 2017
...
[   37.108724] ------------[ cut here ]------------
[   37.110031] WARNING: CPU: 0 PID: 20 at net/wireless/sme.c:981 __cfg80211_disconnected+0x330/0x380 [cfg80211]
[   37.110046] Modules linked in: evdev joydev hid_logitech_hidpp 8192cu hid_logitech_dj cfg80211 rfkill snd_bcm2835(C) snd_pcm snd_timer snd uio_pdrv_genirq fixed uio ip_tables x_tables ipv6
[   37.110153] CPU: 0 PID: 20 Comm: kworker/u2:1 Tainted: G         C      4.14.4+ #1060
[   37.110162] Hardware name: BCM2835
[   37.111172] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[   37.111377] [<c0016554>] (unwind_backtrace) from [<c0013ea4>] (show_stack+0x20/0x24)
[   37.111488] [<c0013ea4>] (show_stack) from [<c0618188>] (dump_stack+0x20/0x28)
[   37.111530] [<c0618188>] (dump_stack) from [<c0021fd4>] (__warn+0xe4/0x10c)
[   37.111556] [<c0021fd4>] (__warn) from [<c00220c8>] (warn_slowpath_null+0x30/0x38)
[   37.112660] [<c00220c8>] (warn_slowpath_null) from [<bf140394>] (__cfg80211_disconnected+0x330/0x380 [cfg80211])
[   37.115013] [<bf140394>] (__cfg80211_disconnected [cfg80211]) from [<bf117af8>] (cfg80211_process_wdev_events+0x160/0x1b4 [cfg80211])
[   37.117103] [<bf117af8>] (cfg80211_process_wdev_events [cfg80211]) from [<bf117b88>] (cfg80211_process_rdev_events+0x3c/0x74 [cfg80211])
[   37.119136] [<bf117b88>] (cfg80211_process_rdev_events [cfg80211]) from [<bf112208>] (cfg80211_event_work+0x24/0x2c [cfg80211])
[   37.120310] [<bf112208>] (cfg80211_event_work [cfg80211]) from [<c0039218>] (process_one_work+0x11c/0x39c)
[   37.120440] [<c0039218>] (process_one_work) from [<c00394d4>] (worker_thread+0x3c/0x548)
[   37.120481] [<c00394d4>] (worker_thread) from [<c003f3d4>] (kthread+0x120/0x15c)
[   37.120522] [<c003f3d4>] (kthread) from [<c000fea0>] (ret_from_fork+0x14/0x34)
[   37.120537] ---[ end trace 34a6a847718dc7e8 ]---
[   37.233656] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   39.701195] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   39.750277] ------------[ cut here ]------------
[   39.751531] WARNING: CPU: 0 PID: 20 at net/wireless/sme.c:981 __cfg80211_disconnected+0x330/0x380 [cfg80211]
[   39.751667] Modules linked in: evdev joydev hid_logitech_hidpp 8192cu hid_logitech_dj cfg80211 rfkill snd_bcm2835(C) snd_pcm snd_timer snd uio_pdrv_genirq fixed uio ip_tables x_tables ipv6
[   39.751825] CPU: 0 PID: 20 Comm: kworker/u2:1 Tainted: G        WC      4.14.4+ #1060
[   39.751835] Hardware name: BCM2835
[   39.752922] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[   39.753273] [<c0016554>] (unwind_backtrace) from [<c0013ea4>] (show_stack+0x20/0x24)
[   39.753380] [<c0013ea4>] (show_stack) from [<c0618188>] (dump_stack+0x20/0x28)
[   39.753422] [<c0618188>] (dump_stack) from [<c0021fd4>] (__warn+0xe4/0x10c)
[   39.753447] [<c0021fd4>] (__warn) from [<c00220c8>] (warn_slowpath_null+0x30/0x38)
[   39.754612] [<c00220c8>] (warn_slowpath_null) from [<bf140394>] (__cfg80211_disconnected+0x330/0x380 [cfg80211])
[   39.756773] [<bf140394>] (__cfg80211_disconnected [cfg80211]) from [<bf117af8>] (cfg80211_process_wdev_events+0x160/0x1b4 [cfg80211])
[   39.758732] [<bf117af8>] (cfg80211_process_wdev_events [cfg80211]) from [<bf117b88>] (cfg80211_process_rdev_events+0x3c/0x74 [cfg80211])
[   39.760738] [<bf117b88>] (cfg80211_process_rdev_events [cfg80211]) from [<bf112208>] (cfg80211_event_work+0x24/0x2c [cfg80211])
[   39.761898] [<bf112208>] (cfg80211_event_work [cfg80211]) from [<c0039218>] (process_one_work+0x11c/0x39c)
[   39.762022] [<c0039218>] (process_one_work) from [<c00394d4>] (worker_thread+0x3c/0x548)
[   39.762060] [<c00394d4>] (worker_thread) from [<c003f3d4>] (kthread+0x120/0x15c)
[   39.762104] [<c003f3d4>] (kthread) from [<c000fea0>] (ret_from_fork+0x14/0x34)
[   39.762120] ---[ end trace 34a6a847718dc7e9 ]---
[   39.892182] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready

Thus, still not working!

By default, the rtl8192cu module is blacklisted.
When I use rtl8192cu instead, it does work. However, with it's (known) very bad performance.

JamesH65, 2018-06-27:
Please update to the latest kernel which may contain a fix for this issue and report back, otherwise this issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

closed 2018-08-31 by JamesH65