raspberrypi/linux #2234: Remoteproc support

trnila, 2017-10-19:
Hello, I would like to implement remoteproc so that I can load firmware for the last core on Raspberry Pi 2.

Is reseting last core just undocumented or not implemented in BCM2836?

Thank you

JamesH65, 2017-12-05:
Can you be a bit more specific about what you want, and what you are trying to do?

trnila, 2017-12-12:
I am interested in asymmetric multiprocessing, where one core is dedicated to a bare metal application that does a real-time peripheral processing (unfortunately, all peripheral interrupts can go only to a single core).
A bare metal application then could communicate with Linux via shared memory.

I am able to run a bare metal application on a dedicated core, but I can't find out how to stop that core, load new instructions and start executing it like it is possible with remoteproc on udoo neo for example.

When I want to load new instructions, I can park core in loop waiting for a new address to be written to the mailbox register, but this method may not work in all cases.

JamesH65, 2017-12-12:
@pelwell This ones a bit of of my field, any ideas?

pelwell, 2017-12-12:
As far as I know there is no way to reset one of the ARM cores without resetting the whole device using the watchdog. You can prevent one or more cores from starting up - there are a few ways to do that:
* `maxcpus=n` on the kernel command line (cmdline.txt).
* `force_cores=0xij` in config.txt, where `i` is a 4-bit mask of cores to enable and `j` is the number of the primary core.
* Write your own startup stub (https://github.com/raspberrypi/tools/tree/master/armstubs) - you can then choose your own way to start up the other cores. Just don't change the four words at 0xf0 - magic, version, ATAGS/DT and kernel start address.

But that's as far as we can go with support - you would be better off asking in the [Bare Metal forum](https://www.raspberrypi.org/forums/viewforum.php?f=72).

trnila, 2017-12-13:
I already asked in the Bare Metal forum.
Thank you

closed 2017-12-13 by trnila

SwagathGadde, 2018-08-17:
i saw this feature here below please have a look at , i think u need to have that respective api in bare metal also when you are running bare metal applications on the other core
http://www.wiki.xilinx.com/OpenAMP