raspberrypi/linux #2235: RPi3: kernel temperature reading ~5°C too low in 4.12-4.14

HiassofT, 2017-10-20:
I've noticed that the SoC temperature reported by the kernel is about 5°C lower than what vcgencmd shows. It looks like this only happens on RPi3 in kernel versions 4.12-4.14. 4.9 and 4.11 seem to be fine on RPi3.

Test on RPi3 with latest 4.14 kernel on Raspbian (compiled with bcm2709_defconfig):
```
# dmesg | grep Raspberry ; vcgencmd version ; uname -a ; cat /sys/class/thermal/thermal_zone0/temp ; vcgencmd measure_temp
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
Oct 11 2017 23:23:59
Copyright (c) 2012 Broadcom
version e5ad4a8e9f183413eb4f7fc4a7687886c715f739 (clean) (release)
Linux rpi3 4.14.0-rc5-v7+ #1 SMP Fri Oct 20 10:58:59 CEST 2017 armv7l GNU/Linux
34704
temp=39.7'C
```

Same setup on RPi2:
```
# dmesg | grep Raspberry ; vcgencmd version ; uname -a ; cat /sys/class/thermal/thermal_zone0/temp ; vcgencmd measure_temp
[    0.000000] OF: fdt: Machine model: Raspberry Pi 2 Model B Rev 1.1
Oct 11 2017 23:23:59
Copyright (c) 2012 Broadcom
version e5ad4a8e9f183413eb4f7fc4a7687886c715f739 (clean) (release)
Linux rpi3 4.14.0-rc5-v7+ #1 SMP Fri Oct 20 10:58:59 CEST 2017 armv7l GNU/Linux
36856
temp=36.9'C
```

A test with a milhouse LE build and kernel 4.14-rc5 on RPi B is fine, too:
```
# dmesg | grep Raspberry ; vcgencmd version ; uname -a ; cat /sys/cl
ass/thermal/thermal_zone0/temp ; vcgencmd measure_temp
[    0.000000] OF: fdt: Machine model: Raspberry Pi Model B Rev 2
Oct 11 2017 23:27:47
Copyright (c) 2012 Broadcom
version e5ad4a8e9f183413eb4f7fc4a7687886c715f739 (clean) (release)
Linux LibreELEC 4.14.0-rc5 #1 Mon Oct 16 22:19:01 BST 2017 armv6l GNU/Linux
43312
temp=43.3'C
```

I tested the latest 4.9 rpi-update kernel and various older kernels I had sitting on my HDD - all on RPi3:
```
# vcgencmd version ; uname -a ; cat /sys/class/thermal/thermal_zone0/temp ; vcgencmd measure_temp
Oct 11 2017 23:23:59
Copyright (c) 2012 Broadcom
version e5ad4a8e9f183413eb4f7fc4a7687886c715f739 (clean) (release)
Linux rpi3 4.9.56-v7+ #1044 SMP Fri Oct 13 15:23:13 BST 2017 armv7l GNU/Linux
46160
temp=46.2'C
```
```
Linux rpi3 4.11.0-rc7-v7+ #1 SMP Tue Apr 18 15:52:42 CEST 2017 armv7l GNU/Linux
46698
temp=46.7'C
```
```
Linux rpi3 4.12.0-v7+ #1 SMP Wed Jul 5 19:37:47 CEST 2017 armv7l GNU/Linux
42774
temp=47.8'C
```
```
Linux rpi3 4.13.0-v7+ #1 SMP Mon Sep 11 12:36:01 CEST 2017 armv7l GNU/Linux
42774
temp=47.2'C
```

@pelwell any idea what's missing there? ISTR there where some changes regarding thermal coefficients in DT, maybe that's somehow related?

I haven't tested all kernel/RPi combinations, so not sure if RPi0/0W or B+ are OK or not - didn't test with these at all.

popcornmix, 2017-10-20:
I have tested with raspbian and latest 4.14.y kernel and I get:
```
pi@domnfs:~ $ dmesg | grep Raspberry ; vcgencmd version ; uname -a ; cat /sys/class/thermal/thermal_zone0/temp ; vcgencmd measure_temp
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
Oct 20 2017 11:48:06 
Copyright (c) 2012 Broadcom
version 3b206771210e4725d250c571e2f577692d1d0613 (clean) (release)
Linux domnfs 4.14.0-rc5-v7+ #131 SMP Fri Oct 20 14:52:34 BST 2017 armv7l GNU/Linux
45084
temp=45.1'C
```
which all looks correct. This is where the parameters come from:
https://github.com/raspberrypi/linux/blob/rpi-4.14.y/arch/arm/boot/dts/bcm2837.dtsi#L80
which looks correct for 4.14.

Note: for Pi 2 the offset is different (by 5'C)
https://github.com/raspberrypi/linux/blob/rpi-4.14.y/arch/arm/boot/dts/bcm2836.dtsi#L82

so a mixup of bcm2837-rpi-3-b.dtb / bcm2836-rpi-2-b.dtb could perhaps cause the offset.

HiassofT, 2017-10-20:
@popcornmix which DT are you using? Here on RPi3 bcm2710-rpi-3-b.dtb is used:
```
# vcdbg log msg 2>&1 | grep Loading
001619.900: Loading 'kernel7.img' to 0x8000 size 0x497894
001627.967: Loading 'bcm2710-rpi-3-b.dtb' to 0x49f894 size 0x5d5c
```

It looks like the bcm2837 dtb has correct coefficients here, but the bcm2710 dtb has the same coefficients as RPi0-2:
```
$ dtc -q -I dtb arch/arm/boot/dts/bcm2710-rpi-3-b.dtb  | grep coefficients
                        coefficients = <0xfffffde6 0x635d8>;
$ dtc -q -I dtb arch/arm/boot/dts/bcm2837-rpi-3-b.dtb  | grep coefficients
                        coefficients = <0xfffffde6 0x64960>;
```

bcm2710-rpi-3-b.dts only seems to include bcm2710.dtsi, not bcm2835.dtsi and the coefficients seem to come from bcm2708-rpi.dtsi
```
$ grep dtsi arch/arm/boot/dts/.bcm2710-rpi-3-b.dtb.dts.tmp
# 1 "arch/arm/boot/dts/bcm2710.dtsi" 1
# 1 "arch/arm/boot/dts/bcm283x.dtsi" 1
# 2 "arch/arm/boot/dts/bcm283x.dtsi" 2
# 3 "arch/arm/boot/dts/bcm283x.dtsi" 2
# 4 "arch/arm/boot/dts/bcm283x.dtsi" 2
# 5 "arch/arm/boot/dts/bcm283x.dtsi" 2
# 156 "arch/arm/boot/dts/bcm283x.dtsi"
# 172 "arch/arm/boot/dts/bcm283x.dtsi"
# 2 "arch/arm/boot/dts/bcm2710.dtsi" 2
# 1 "arch/arm/boot/dts/bcm270x.dtsi" 1
# 3 "arch/arm/boot/dts/bcm270x.dtsi" 2
# 3 "arch/arm/boot/dts/bcm2710.dtsi" 2
# 1 "arch/arm/boot/dts/bcm2708-rpi.dtsi" 1
# 4 "arch/arm/boot/dts/bcm2710.dtsi" 2
# 100 "arch/arm/boot/dts/bcm2710.dtsi"
# 1 "arch/arm/boot/dts/bcm283x-rpi-smsc9514.dtsi" 1
```

snippet from arch/arm/boot/dts/.bcm2710-rpi-3-b.dtb.dts.tmp:
```
# 1 "arch/arm/boot/dts/bcm2708-rpi.dtsi" 1
...
&cpu_thermal {
 coefficients = <(-538) 407000>;

 /delete-node/ trips;
};
# 4 "arch/arm/boot/dts/bcm2710.dtsi" 2
```

popcornmix, 2017-10-20:
```
$ hd /sys/firmware/devicetree/base/thermal-zones/cpu-thermal/coefficients
00000000  ff ff fd e6 00 06 49 60                           |......I`|
00000008
$ printf "%d\n" 0x064960
412000
```
do you get the same?

pelwell, 2017-10-20:
`od --endian=big -t d4 <file>` will get you a decimal result in one step.

HiassofT, 2017-10-20:
On RPi3, with bcm2710-rpi-3-b loaded I get 407000
```
# od --endian=big -t d4 /sys/firmware/devicetree/base/thermal-zones/cpu-thermal/coefficients
0000000        -538      407000
0000010
```

HiassofT, 2017-10-20:
It looks to me that the upstream bcm2837 dts file, including bcm2837.dtsi is OK but the downstream bcm2710 dts files, including bcm2710.dtsi, don't set the thermal coefficients correctly. So I guess CM3 would also show a wrong temperature.

```
$ grep coefficients arch/arm/boot/dts/.bcm*.tmp
arch/arm/boot/dts/.bcm2708-rpi-0-w.dtb.dts.tmp: coefficients = <(-538) 407000>;
arch/arm/boot/dts/.bcm2708-rpi-0-w.dtb.dts.tmp: coefficients = <(-538) 407000>;
arch/arm/boot/dts/.bcm2708-rpi-b.dtb.dts.tmp: coefficients = <(-538) 407000>;
arch/arm/boot/dts/.bcm2708-rpi-b.dtb.dts.tmp: coefficients = <(-538) 407000>;
arch/arm/boot/dts/.bcm2708-rpi-b-plus.dtb.dts.tmp: coefficients = <(-538) 407000>;
arch/arm/boot/dts/.bcm2708-rpi-b-plus.dtb.dts.tmp: coefficients = <(-538) 407000>;
arch/arm/boot/dts/.bcm2708-rpi-cm.dtb.dts.tmp: coefficients = <(-538) 407000>;
arch/arm/boot/dts/.bcm2708-rpi-cm.dtb.dts.tmp: coefficients = <(-538) 407000>;
arch/arm/boot/dts/.bcm2709-rpi-2-b.dtb.dts.tmp: coefficients = <(-538) 407000>;
arch/arm/boot/dts/.bcm2709-rpi-2-b.dtb.dts.tmp: coefficients = <(-538) 407000>;
arch/arm/boot/dts/.bcm2710-rpi-3-b.dtb.dts.tmp: coefficients = <(-538) 407000>;
arch/arm/boot/dts/.bcm2710-rpi-cm3.dtb.dts.tmp: coefficients = <(-538) 407000>;
arch/arm/boot/dts/.bcm2835-rpi-a.dtb.dts.tmp: coefficients = <(-538) 407000>;
arch/arm/boot/dts/.bcm2835-rpi-a-plus.dtb.dts.tmp: coefficients = <(-538) 407000>;
arch/arm/boot/dts/.bcm2835-rpi-b.dtb.dts.tmp: coefficients = <(-538) 407000>;
arch/arm/boot/dts/.bcm2835-rpi-b-plus.dtb.dts.tmp: coefficients = <(-538) 407000>;
arch/arm/boot/dts/.bcm2835-rpi-b-rev2.dtb.dts.tmp: coefficients = <(-538) 407000>;
arch/arm/boot/dts/.bcm2835-rpi-zero.dtb.dts.tmp: coefficients = <(-538) 407000>;
arch/arm/boot/dts/.bcm2835-rpi-zero-w.dtb.dts.tmp: coefficients = <(-538) 407000>;
arch/arm/boot/dts/.bcm2836-rpi-2-b.dtb.dts.tmp: coefficients = <(-538) 407000>;
arch/arm/boot/dts/.bcm2837-rpi-3-b.dtb.dts.tmp: coefficients = <(-538) 412000>;
```

pelwell, 2017-10-20:
Thanks, Hias - we have unconfused ourselves and get the same results as you. The fundamental cause of this discrepancy is a failure to provide BCM2710-specific coefficients. Not doing so has no effect on thermal limiting since that is provided by the firmware, but it is confusing.

4.9 and 4.11 work as expected because they read the temperatures via the firmware. It's the later kernels that are performing the readings directly that need the coefficients in the DT.

We won't be releasing 4.12. or 4.13 kernels, so I propose to refactor bcm2710.dtsi in 4.14 to include the new bcm2837.dtsi and only patch it where necessary.

HiassofT, 2017-10-20:
Thanks for looking into it, Phil, fixing it on 4.14 only is fine for me.

popcornmix, 2017-10-20:
Just FYI
```
diff --git a/arch/arm/boot/dts/bcm2710.dtsi b/arch/arm/boot/dts/bcm2710.dtsi
index 3e134a120861..dbe3cf23a8a8 100644
--- a/arch/arm/boot/dts/bcm2710.dtsi
+++ b/arch/arm/boot/dts/bcm2710.dtsi
@@ -139,6 +139,10 @@
 #endif
 };
 
+&cpu_thermal {
+       coefficients = <(-538)  412000>;
+};
+
 &intc {
        compatible = "brcm,bcm2836-armctrl-ic";
        reg = <0x7e00b200 0x200>;
```
was what made it work for me. That is an ugly fix as it takes the correct upstream coeffs, modifies then in downstream overlay and then fixes them back up.

But Phil will work on tidying this up by using more of the upstream dtsi directly.

pelwell, 2017-10-20:
You can see my first (untested) attempt here: https://gist.github.com/pelwell/1054f2d6d15189861ad205185fddddce

Feedback welcome.

pelwell, 2017-10-21:
Now a PR: https://github.com/raspberrypi/linux/pull/2236

MilhouseVH, 2017-11-02:
LibreELEC with 4.14-rc7 + c7114a20f9bc77312297fbd5b1ba01ea7e1bb491 is looking good - temperatures from `sysfs` and `vcgencmd` now match.

JamesH65, 2017-12-05:
Since this has been merged, can the issue be closed?

HiassofT, 2017-12-05:
thanks, closing issue

closed 2017-12-05 by HiassofT