raspberrypi/linux #2239: 4.9.58 completely broken

mkreisl, 2017-10-23:
Pi 2 and Pi 3 (Pi 1 not tested) does not boot when running kernel 4.9.58

Reverting this commit https://github.com/raspberrypi/linux/commit/b017e4f94b17ac322340159f63c546e8d115b104 makes kernel working again

Kernel was build with working 4.9.56 .config file

pelwell, 2017-10-23:
What's in your config.txt?

mkreisl, 2017-10-23:
> What's in your config.txt?

```
grep -v "^#" /boot/config.txt | grep -v "^$"
initramfs initramfs.gz  0x1500000
gpu_mem_1024=256
gpu_mem_512=144
gpu_mem_256=100
initial_turbo=3
hdmi_force_hotplug=0
hdmi_ignore_hotplug=0
hdmi_ignore_cec_init=1
hdmi_ignore_cec=0
disable_overscan=1
disable_splash=1
max_usb_current=1
decode_MPG2=0x00000000
decode_WVC1=0x00000000
dtparam=audio=on
dtoverlay=lirc-rpi,debug
hdmi_force_hotplug=1
```

pelwell, 2017-10-23:
Hmmm, your config looks OK.

@popcornmix Perhaps we should revert it before making any new releases, or at least hold off until I've had chance to investigate.

popcornmix, 2017-10-23:
@pelwell it is in current rpi-update fimware (pushed Friday) so it's obviously not breaking things for the majority. ``initramfs`` related perhaps?

pelwell, 2017-10-23:
In that case there must be some other reason why I failed to get it to bit before running out of time.

pbui, 2017-10-23:
It is also broken for me (only get the rainbow screen at startup).  I am not using an initramfs, so I don't think that is a problem.  My config.txt only has the line `dtparam=audio=on`.

I've tested this for armv6l and armv7l (a RPI1 and a RPI2).

pelwell, 2017-10-24:
Interesting. On a relatively clean image I can boot top-of-tree rpi-4.9.y, so it isn't a simple matter of bad commit. If you have a spare SD card, can you install a fresh Raspbian image of it then rpi-update (or apply your own kernel with https://github.com/raspberrypi/linux/commit/b017e4f94b17ac322340159f63c546e8d115b104) to see if that works for you.

6by9, 2017-10-24:
The latest rpi-update firmware also removes the VPU side of vc-cma.
Is it that the cmdline params that the old firmware is inserting (```vc-cma-mem=<size>@<base>```) is upsetting the kernel? Those are only populated if either ```cma_lwm``` or ```cma_hwm``` are defined though in config.txt (or otherwise), and they aren't by default.

pelwell, 2017-10-24:
The kernel completely ignores unrecognised command line parameters, so that shouldn't be an issue. I just added `vc-cma-mem=128M@128M` and it booted as normal.

pelwell, 2017-10-24:
Also `vc-cma-mem=0/0x11800000@0xd400000`.

6by9, 2017-10-24:
Very strange behaviour.
My builds with a Pi2 are now showing that without vc-cma the boot stops waiting for the root filing system that never shows up. Normally I'm NFS mounting, and the SMSC9514 never enumerates, so no root filing system. Even with SD card booting it sits there waiting for the filing system. The last message on the serial console in each case is 
```
of_cfs_init
of_cfs_init: OK
```
and in the case of SD card booting
``` Waiting for root device /dev/mmcblk0p2```
vc-cma has absolutely nothing to do with the root filing system, so it does imply a nasty side effect rather than anything intrinsic in it. Doubly bizarre that reverting back to b017e4f (4.9.56) or 2ddd04 (4.9.57) are also showing the same behaviour. @popcornmix 's rpi-update build from 2ddd04 is fine though, so it may be something different between our build environments.

Next thing to try is leaving vc-cma in the build but stubbing out all the functionality.

mkreisl, 2017-10-24:
> vc-cma has absolutely nothing to do with the root filing system, so it does imply a nasty side effect

Agree, I'm using initramfs and one of the first things there is printing a message to the screen, long before any mounts will be done. And that message never appears

pelwell, 2017-10-24:
A thought: the sdhost driver grabs a small amount of dma_coherent (uncached) memory for logging purposes - is that allocation failing?

pelwell, 2017-10-24:
Never mind - that should fail gracefully.

6by9, 2017-10-24:
Comparing logs I appear to also be missing 
```
BCM2708FB: allocated DMA memory f35000000
BCM2708FB: allocated DMA channel 0 @ bb80f000
```
which in the working case is happening BEFORE vc-cma is started.
Nobbling ```vc_cma_init``` so that it bails at line 1074
```
	if (!check_cma_config())
		goto out_release;
```
also fails in the same way as removing the entire driver. 
The call ```vchiq_add_connected_callback(vc_cma_connected_init);``` to actually do something useful over VCHIQ is NOT required.

Now removing more blocks of the init code to narrow it down.

pelwell, 2017-10-24:
You could try adding an explicit `cma=16m` (or similar to the command line.

6by9, 2017-10-24:
> You could try adding an explicit cma=16m (or similar to the command line.

Already got ```cam=128M``` due to onging Unicam and vc-sm work.

I am wondering if it is something over different caching options for different heaps, but vc-cma is now 
doing nothing active in my build (just the alloc_chrdev_region call in the last run, next really nothing which I'm expecting to fail again, then inserting a sleep).

pbui, 2017-10-24:
@pelwell I built the kernels from clean checkouts of the rpi-4.9.y tree (I'm actually building packages for @voidlinux), so the kernel should in sync with upstream.

That said, I will give raspbian and rpi-update a try later this afternoon.

pelwell, 2017-10-24:
I'd hold off for a while - now that @6by9 can switch between working and non-working cases, having you do the same won't tell us anything new.

6by9, 2017-10-24:
```msleep(1000)``` also fails.
It looks like something has stalled/failed earlier.
bcm2708_fb_probe hasn't been called at all when normally it is before vc-cma

Something I get inserted in the failure case only is
```
[0.103813] 3f201000.serial: ttyAMA0 at MMIO 0x3f201000 (irq = 87, base_baud = 0) is a PL011 rev2
[0.580578] console [ttyAMA0] enabled
```
Both have logged the ```Serial: AMBA PL011 UART driver``` line.
(I do have ```console=ttyAMA0``` in the command line, but that doesn't change with the failure case)

pelwell, 2017-10-24:
I would expect to see both of those in the success case as well, but perhaps a bit later - around the 1s and 2s marks on a Pi3 with `dtoverlay=pi3-disable-bt`.

6by9, 2017-10-24:
Hang on, the command line IS changing.
Working:
```
[    0.000000] Kernel command line: bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  cma=128M dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait
```
It looks highly suspicious that there is a double space between ```vc_mem.mem_size``` and ```cma=```
Failing:
```
[    0.000000] Kernel command line: earlyprintk console=ttyAMA0 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2708_fb.fbswap=1 dma.dmachans=0x7f35 bcm2709.boardrev=0x2a01041 bcm2709.serial=0x12345678 bcm2709.uart_clock=48000000 bcm2709.disk_led_gpio=47 bcm2709.disk_led_active_low=0 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  cma=128M dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait
```
So we've gained ```earlyprintk console=ttyAMA0```, and ```dma.dmachans=0x7f35 bcm2709.boardrev=0x2a01041 bcm2709.serial=0x12345678 bcm2709.uart_clock=48000000 bcm2709.disk_led_gpio=47 bcm2709.disk_led_active_low=0```

I'll have a look at the firmware to see what criteria it uses for adding that lot

6by9, 2017-10-24:
> I'll have a look at the firmware to see what criteria it uses for adding that lot

It's if the firmware thinks it needs a legacy_cmdline due to no soc dma_ranges, or it's been told there is no DT at all. Firing up the debugger....

pelwell, 2017-10-24:
That doesn't explain everything though - it should never add earlyprintk, but that is in the upstream bcm283x DTBs. Why is the firmware loading the wrong class of DTB?

6by9, 2017-10-24:
Clean build of 4.9.58 and NOT using ```mkknlimg``` is working fine.

mkreisl, 2017-10-24:
Does that mean that `mkknlimg` is not needed any more?

6by9, 2017-10-24:
Issue found with mkknlimg I think. https://github.com/raspberrypi/linux/blob/rpi-4.9.y/scripts/mkknlimg#L81 is looking for the ```vc_cma_init``` symbol to switch between 270x and 2835.

6by9, 2017-10-24:
> Does that mean that mkknlimg is not needed any more?

Not needed, and not recommended. Docs updated back in March. https://github.com/raspberrypi/documentation/commit/4f5b2dba737ca244f9789c34862aaa76c311cd38#diff-7c9a371d1e95ab0957dd174df8b6e046

mkreisl, 2017-10-24:
> Issue found with mkknlimg I think. https://github.com/raspberrypi/linux/blob/rpi-4.9.y/scripts/mkknlimg#L81 is looking for the vc_cma_init symbol to switch between 270x and 2835.

OMG, what a stupid script

So this is culprit, output of wrong kernel built was

```
Version: Linux version 4.9.58+ (manfred@kmnote5.kmhome.com) (gcc version 4.9.3 20141031 (prerelease) (Linaro GCC 2014.11) ) #1 SMP PREEMPT Mon Oct 23 17:08:06 CEST 2017
DT: y
DDT: y
270x: n
283x: y
```

Will build kernel again an see what happens

mkreisl, 2017-10-24:
Ok, that solves the issue., but:

Can anybody tell me why a mkknlimgd kernel 4.13.x is working fine ??? This kernel does not have that vc-cma stuff included

6by9, 2017-10-24:
It's not clear to me why it is complaining without vc-cma anyway.

Adding extra debug into mkknlimg I appear to be seeing all my builds identifying as 270x due to "bcm2708_fb" and "vc-mem" being present. That includes the 4.9.58 kernel that I'd built earlier and was failing. "vc_cma_init" should have just been an extra symbol that also set the 270X flag.

pelwell, 2017-10-24:
Yes, and yet I have seen it fail on a downstream Pi 2/3 kernel. Perhaps something in the byte stream is tripping up `strings`?

pbui, 2017-10-24:
I can confirm not using mkknlimgd fixes broken kernel builds.  Thanks.

closed 2017-10-25 by mkreisl

6by9, 2017-10-27:
It looks like we'd crossed this bridge before with 4.11, but none of us had remembered.
https://github.com/raspberrypi/linux/commit/b24e557c487169d149e1cf4550b586131b682375#diff-fbf412837baae5ead31c76ef50753323

The strings found with the 4.9.58 kernel were
```
Checking string brcm,bcm2835-gpio - adding flags 03
Checking string brcm,bcm2835-sdhost - adding flags 01
Checking string brcm,bcm2835-mmc - adding flags 01
Checking string of_cfs_init - adding flags 06
Checking string brcm,bcm2836 - adding flags 13
Checking string bcm2708_fb - adding flags 01
```
So it was only bcm2708_fb that could trigger the 270X flag.

zertyz, 2019-07-26:
I'm facing the same issue on my raspberry Pi 2 with archlinuxarm & kernel 4.19.59, after upgrading from before rPi4 was taken into consideration on the arch packages. I'm booting from an raid array of 4 USB sticks + sd card.

In my case, I am able to boot (without those freaking USB errors) when I use 256m of GPU ram (or more). Decreasing that a little (192, 160, 128...) sometimes issues a failure into initializing a random CPU (only 3 raspberries shows on my debug monitor and the boot process hangs for more than I can wait trying to wake the cpu up).

When downgrading my system to the versions before the update, everything works fine.