raspberrypi/linux #2241: sc16is7xx uart i2c overlay problems

IchKG, 2017-10-23:
My test:
I open 2 terminals on the pi and start the python intepreter on both then I connect one to ttySC0 and the other to ttySC1 the tx-line is in direct loopback to rx for both channels
If I send a string a receive it immediately back as it should be.
Now I have this small script that just sends the string 'test' and checks if it was received back correctly for 1000 times (takes a few moments):
Code: Select all

`for i in range(1000):
	serial1.write('test')
	if serial1.read(4) == 'test':
		print 'pass'
	else:
		print 'fail'
		break`

If I run this script on one channel only it executes fine and I also see the data on my logic analyser but if I start the script on ttySC0 and then send something on ttySC1 something crashes 
the loop on ttySC0 directly prints fail and after that fail the receive of both ttySC0 AND ttySC1 are numb till I reload the kernelmodule.
With the logic analyser I see that tx is still working on both channels but non of them is receiving anything they are just returning an empty string after serial timeout like I never send anything
The channels are not electrically connected in any form the setup runs on 3,3V and the IRQ pin is correctly connected to the pi and has a 1k pullup to 3,3V.

As the setup works for both channels independent I think there is something wrong with the kernelmodule

PS:
I have this issue on a project that now uses about ~30 pis and the issue persists on all of them

pelwell, 2017-10-23:
Is this the dual-UART SC16IS752 or two single-UART SC16IS750s? Which HAT/hardware add-on are you using?

IchKG, 2017-10-23:
it's the SC16IS752 on a self designed pcb with the i2c wires kept short
frequencysource is a SER3809-ND

pelwell, 2017-10-24:
SC16IS752 breakout boards are impossible to come by now, so reproducing this is going to be difficult. I've found a forum thread from last year discussing the same problem (https://www.raspberrypi.org/forums/viewtopic.php?t=146908&start=25#p1082043) , but we didn't get anywhere for the same reasons - lack of hardware and lack of time.

pelwell, 2017-10-24:
It's interesting that the you are seeing the problem on I2C, while others have seen it using SPI. This makes me think that it is a problem in the SC16IS7XX driver itself, rather than either of the subsystems.

What speeds are you running the I2C bus and the UARTs at? Having you tried running I2C faster and/or the UARTs slower? `dtparam=i2c_arm_baudrate=200000` sets the I2C clock to 200KHz (default 100KHz).

pelwell, 2017-10-24:
According to the datasheet the maximum I2C clock speed is 400KHz, so your UART baudrates probably shouldn't exceed 115200 (and that may be optimistic once you factor in register access overheads).

IchKG, 2017-10-24:
my serials normally run at 19200 baud and i2c speed wasn't set so it should be the default value
I've tried 100k 200k and 400kHz i2c speed and also lowering baud to 9800 
in my last test with 400kHz i2c and 9800 baud it seems that it needs some more chars on SC1 to crash the loop of SC0 but the problem still persists... 

in my boot/config.txt I set speed like this is that correct or do I need to edit the devicetree overlay or something?
`dtparam=i2c_arm_baudrate=400000`
`dtoverlay=sc16is752-i2c`

pelwell, 2017-10-24:
Thanks for trying - yes, that all looks correct. I see you have a custom overlay, but since it works when the channels are accessed serially (no pun intended) I don't believe it is at fault.

Let me read the driver a bit - I had a thought while out jogging.

pelwell, 2017-10-24:
A few things about the driver bother me:
1. It's using a workqueue instead of a threaded interrupt handler, and it's returning IRQ_HANDLED before the interrupt is really handled. This may be harmless, but it may not.
2. It will process all interrupts for port 0 until there are no more, then do the same for port 1. What happens if an interrupt for port 0 is raised after we've looked at port 0 but before we return from handling port 1? Again, this may be harmless but I'm not certain.

If you are feeling brave:
i. Use rpi-update to get a 4.9.57 kernel (use `sudo rpi-update 2be9861b19131981c2da6163e8309f1b7f014c54` to be sure to get the right version), and reboot.
ii. Download [this replacement sc16is7xx.ko](https://drive.google.com/open?id=0B_P-i4u-SLBXYTJUbnBmYkFaUHM). N.B. This assumes you have a Pi 2 or Pi 3. If you are actually using  Zeroes or B+s etc, then I'll need to get you a v6 build of the module.
iii.
```
sudo mv /lib/modules/`uname -r`/kernel/drivers/tty/serial/sc16is7xx.ko{,.orig}
sudo cp sc16is7xx.ko /lib/modules/`uname -r`/kernel/drivers/tty/serial/
```
(watch the back-ticks)
iv. Reboot and try again.

pelwell, 2017-10-24:
Doh - think I've got it. The problem is likely to be this line in the overlay:
```
				interrupts = <24 2>; /* IRQ_TYPE_EDGE_FALLING */
```
Edge sensitive interrupts are usually a bad idea. Treating level-type interrupts sources as if they were edge-type is a disaster, since if you ever fail to clear any interrupt-generating state in the source you will never get an interrupt again. In this situation, my concerns above can lead to disaster.

Don't bother downloading my hacked module - instead, change the above line in the overlay to:
```
				interrupts = <24 8>; /* IRQ_TYPE_LEVEL_LOW */
```
Then recompile and install as usual:
```
$ dtc -@ -I dts -O dtb -o sc16is752-i2c.dtbo sc16is752-i2c-overlay.dts
$ sudo cp sc16is752-i2c.dtbo /boot/overlays
```
If I'm right, any interrupt state missed in a call to the interrupt handler will result in another call because the interrupt should still be asserted.

IchKG, 2017-10-24:
ok thanks for your detailed explanation 
I will try that in a few hours on the pi
It sound reasonable that the edge type irq is the problem but if that's true I'm wondering why all the overlays I can find on the internet are with that type of trigger and why is almost nobody complaining about the same problems with this overlay...
As I was looking at some overlays found on the internet I saw that they are either using gpio17 or gpio24 and they are both pull-down pins on my board the irq pin is gpio7 and that's a pull-up that's shouldn't be a problem as the irq signal is already pulled up by the resistor, but maybe there is something other wrong with the pin I've choosen (e.g. it's also the SPI0_CE1_N pin but I'm not using that interface)
Maybe I will try to route to one of the pins 17/24 because then my setup is almost the same which other people are using on the internet

pelwell, 2017-10-25:
I don't think there is anything wrong with using GPIO7 - apart from the default pull direction (as you've mentioned) all GPIOs are equal once in GPIO mode. I think GPIO24 is popular for SPI work because it sits next to the SPI pins on the 40-pin header and has no other (generally) useful functions available, and the I2C variant of the sc16is75x overlay is essentially a clone of the SPI version.

IchKG, 2017-10-25:
Ok that did the trick 
my script is now executing even when I'm running the send loop on both lines, but I'm now facing another big problem:
My pi sometimes freezes completely so that I have to unplug and repower it again. Looks like it sometimes get stuck in an endless loop maybe because the interrupt isn't cleared completely

pelwell, 2017-10-25:
Which Pi do you have? If it's a Pi 2 or a Pi 3 then there should be 3 cores available while one of them is stuck in an interrupt loop (if indeed that is what is happening). Also, stuck interrupts should eventually get disabled.

Can you try creating a background task that outputs a message somewhere you can see it (or strobing a GPIO pin) periodically, to see if the Pi really is dead? Also, are you sure you aren't suffering from poor power supply? You could display the output of "vcgencmd get_throttled" in your background task loop, assuming you have Pi that supports that feature (A+, B+, 2, 3).

IchKG, 2017-10-25:
it's a Pi3
normally I'm controlling it over ssh but it's also connected to a monitor
if it freezes the output on the monitor freezes as well even the status led on the pi freezes
Later I will try your suggestion with the background task but I don't think that anything is working on pi after it crashes

pelwell, 2017-10-26:
I'm wondering if the reason for the use edge-triggered interrupts is the deferral of the interrupt handling to a workqueue. Workqueues are handled by a special process and are probably used because workqueue jobs are allowed to sleep, which is necessary for I2C and SPI transactions. Since this process does not run in interrupt context, if a level-sensitive interrupt is used it will continue to fire until the process succeeds in tending to the UART's needs. Edge-sensitive interrupts fire once then stop, and must be re-armed by allowing the interrupt signal to be deasserted.

As I understand it, Linux's threaded IRQ support is roughly similar to a workqueue, the main difference being that the IRQ framework automatically disables the specific interrupt until the threaded IRQ handler has run. This gives the same benefits as the ad-hoc workqueue solution (the handler can sleep) without the continuous interrupts pitfall, so it would be safe to use a level-sensitive interrupt and avoid the nasty race conditions.

I think the sc16is7xx driver should be reworked to use a threaded IRQ. This probably isn't a large amount of work, but it would need to be well tested and eventually upstreamed. If you could spare some hardware for a while and your test app that might speed things up, otherwise I can take a guess and send you something for testing.

JamesH65, 2017-12-05:
Are we likely to see any progress on this, or should we close?

IchKG, 2017-12-05:
i'm currently working on some other stuff but I was planning to send my testhardware to pelwell in near future...

JamesH65, 2017-12-05:
OK, thanks, I'll leave it as is.

pelwell, 2017-12-05:
@IchKG Send an email to phil@raspberrypi.org and I'll reply with the address.

6by9, 2017-12-05:
Oddly https://github.com/raspberrypi/linux/commit/9e6f4ca3e567d5d9e4133d2da8bae8f476f85873#diff-8fbae94d4df6fa1150b189e754472502 switched from using a threaded irq to using a the work queue.
Probably worth finding the thread history on that one.

hdeveer, 2018-02-06:
It seems I'm running into similar problems that at some time the driver gets stuck, but a in complete other hardware setup. Nothing todo with the rpi but pure this driver related.
What I saw is that the sc16is7xx_port_irq() function does not handle the case SC16IS7XX_IIR_INPIN_SRC (should read at least the SC16IS7XX_MSR_REG). Even if not using the pins I saw IIR code 0x30 passing by sometimes. This is at least one of the causes of my driver lockup. Still it solves not all issues yet.

pelwell, 2018-02-06:
Do you see:
```
ttySCx: Unexpected interrupt: xx
```
in the kernel message log (dmesg)?

hdeveer, 2018-02-06:
In the original driver probably yes (but not sure). I added the extra cases with a default break, before I saw later (after some other fixes) that the status was stuck.
Anyhow I would certainly add something like:
```
case SC16IS7XX_IIR_INPIN_SRC:
    sc16is7xx_port_read(port, SC16IS7XX_MSR_REG);
    break;
```

pelwell, 2018-02-06:
Are you up for creating a Pull Request? If you write a reasonable commit message and include a Signed-off-by: line we can try to upstream it (I wouldn't anticipate much resistance).

hdeveer, 2018-02-06:
That is possible, but currently I'm debugging more issues in my hardware combination. It could be that I find more issues. So I will postpone that for later.
I just wanted to mention it upfront in case somebody ends up here in this thread.

pelwell, 2018-02-06:
Yes, and thank you for that. I'm still surprised by how often when I search for something the top hit is an RPi Forum or GitHub thread.

hdeveer, 2018-02-06:
It is even worse: I must add a:
`    sc16is7xx_port_read(port, SC16IS7XX_IOSTATE_REG);`
before the MSR_REG read otherwise the IRQ stays stuck at zero as I found out now.

Theoi-Meteoroi, 2018-02-06:
I have tried using the SC16S751 device (commonly available) over I2C.  Recieve doesn't work. Transmit does work, however, any loop-back testing fails.  After wasting much time on it, I settled with using USB adapters for serial devices.

hdeveer, 2018-02-07:
The loopback works with my setup (over SPI) just connect RX/TX. But I'm going to switch over to I2C later. I use it via a USB bridge even much worse. A very serious problem in this bridge case  is that the INT handling is doomed to fail because it's edge triggered and you have a race condition between reading the INT from the actual device and the automatic clearing of it when reading the status register. I probably have to make a hardware fix for that.

pelwell, 2018-02-21:
I've just pushed [a patch](https://github.com/raspberrypi/linux/commit/dbca2148f4c21454da461f1208b05d3a5887d133) that fixes (or works around) the originally reported problem by altering the way the interrupt handling works across multiple channels. The lengthy commit message explains the problem and the solution.

hdeveer, 2018-03-16:
Finally I had time to check this out on an I2C bus. With also the changes from 'pelwell' it works now on both channels no IRQ stuck lines or hangups. (In a simplified form and to make it conforming to the kernel coding standards.
I did to more changes to the driver (and the i2c-core.c kernel file), but that has to do with hotplugging and device attachment.

pelwell, 2018-03-16:
@IchKG Thank you for the test hardware - without it this investigation would still be ongoing.
@hdeveer Thank you for testing and feeding back your findings.

I'll close this issue now, but either re-open it or create a new issue you find there is still a problem.

closed 2018-03-16 by pelwell

krazykyran, 2018-05-14:
pelwell, instead of the extra time required to read the status register for pending interrupts, is there a way to read the status of the interrupt pin (gpio read) and return if high? Just an idea.

pelwell, 2018-05-14:
That would require that the sc16is7xx driver knows that it is using a GPIO as an IRQ, and which one, which it can't do in any reasonable way because of the abstraction layers Linux uses.

IchKG, 2018-06-26:
today I updated from 4.14.20 to 4.14.50 and now the kernelmodule doesn't work any more
I see the two ports and my py serial let me open them
If i send something with the write command I see some communication on the i2c lines but there is no data coming out of the tx pin

IchKG, 2018-06-28:
when I'm sending data the py serial write function returns the right number of bytes transmitted but the uart ic is outputting nothing
I've connected my logicanalyser that I use for debugging my uartprotocoll to the i2c lines. I've attached the files for my old kernel vs. 4.14.49 as pdfs. The first command is the serial initialization and the second is the teststring "test". Seems like the version with the newer kernel just asking for write and read commands but does not send any data. 
Flow control doesn't make any difference.

[i2c_com_notworking.pdf](https://github.com/raspberrypi/linux/files/2145704/i2c_com_notworking.pdf)
[i2c_com_working.pdf](https://github.com/raspberrypi/linux/files/2145705/i2c_com_working.pdf)

pelwell, 2018-06-28:
Can you receive any bytes?

pelwell, 2018-06-28:
And please try the two rpi-update releases I mentioned above.

pelwell, 2018-06-28:
The trace from the old, working kernel uses address 0x4D, while the trace from the new, failing system tries to use address 0x48. The new "official" sc16is752-i2c overlay has a default address of 0x48, so you will need to override the default using the `addr` parameter:
```
dtoverlay=sc16is752-i2c,addr=0x4d
```