raspberrypi/linux #2243: ASoC: snd-soc-dummy codec in sound/soc/soc-utils.c too restrictive.

ast, 2017-10-24:
Background: 
The `snd-soc-dummy` codec is a dummy codec used for testing and sometimes with i2s codecs that doesn't have their own driver. 

Example of use: 
https://github.com/skakri/asoc-i2s-loader/blob/master/asoc-i2s-loader.c

It restricts sample rates to common rates between 8000-192000ksps:
```
#define STUB_RATES	SNDRV_PCM_RATE_8000_192000
```

However, the bcm2835 i2s driver supports many more rates:
```
.rates =	SNDRV_PCM_RATE_CONTINUOUS,
.rate_min =	8000,
.rate_max =	384000,
```

I feel that the restrictions in the dummy codec is unnecessary and could at least be as permissive as the i2s driver. I can't see that this could break anything.

popcornmix, 2017-10-24:
If you are referring to https://github.com/raspberrypi/linux/blob/rpi-4.9.y/sound/soc/soc-utils.c#L94 then this is an upstream file, untouched by us.
If you think a different definition of supported rates would be generally useful, you should request it upstream. Assuming the change is accepted we will pick it up here.
See: https://www.kernel.org/doc/html/v4.12/process/submitting-patches.html

ast, 2017-10-24:
Ok. I will do that. I'm closing this issue then.

closed 2017-10-24 by ast