raspberrypi/linux #2244: RT patch for stable releases

az2022, 2017-10-26:
With the app I work on, one of the requirement is to use Raspbian kernel with real-time patch applied. It has become an issue of finding a match to what kernel version is released by raspberry-pi org and which kernel version patch is released by kernel.org (https://www.kernel.org/pub/linux/kernel/projects/rt/). Lot of times they don't match, for e.g. I got the latest Raspbian (Stretch) which hosts kernel v4.9.41 while the real time patch is available for 4.9.40 and 4.9.47. I checked-out Raspbian source commit available right after 4.9.47 was committed to the source tree, applied RT patch and built. Everything went well and Pi3 boots fine, but sometimes hangs when apps (for e.g. gitk) are started or midway of a  gcc build. Understandably Raspbian was not released with 4.9.47 but I had to try! Bottom line, it would be nice to have a RT patch available for the released/stable Rapbian images, unless there's a way around this? :)

pelwell, 2017-10-26:
Although the current Raspbian Stretch release comes with 4.9.41 there should not be any hard dependencies on a particular kernel version. The rpi-update repo hosts hundreds of kernel versions (full list available [here](https://github.com/Hexxeh/rpi-firmware/commits/master) - notice the hexadecimal numbers identifying each release), a large number of which should work with Stretch.

I suggest you use rpi-update to choose a 4.9.47 kernel:
```
sudo rpi-update 2663a143525df3c8305253c6f93f1707156e6a9d
```
Check that the kernel works (except without the RT support), then locate the corresponding point in the source tree - commit 807c5d97f188a5766a931f944e3c4dcc21af4fce ought to be OK - and apply the 4.9.47 patch.

az2022, 2017-10-27:
I did try your suggestion couple weeks ago and ran into issues of random sluggishness. Also when building my project it used to hang up. Today I started again from scratch with a fresh copy of Raspbian. After upgrading to v4.9.47 kernel using rpi-update, I did not see sluggishness and was able to compile my project (basically just compiling AVS device sdk from: https://github.com/alexa/avs-device-sdk/wiki/Raspberry-Pi-Quick-Start-Guide). After applying the RT patch, things still appear to work but when compiling my project again (using stock gcc) it randomly hangs in the middle, just as I saw before. After a reboot I found that some of the object files and a library were 0 bytes in size. After I removed these, I was able to complete building. If I recompile from scratch I am pretty sure it will hang again. Something is not right, which gets magnified when applying the RT patch.

pelwell, 2017-10-27:
Is the Pi perhaps running out of memory and spending all its time swapping? Check the SDK Makefile and looks for `-j<n>` flags that set the degree of parallelism, reducing it if it is greater than 1. You could also try running different tasks while stressing (`sudo apt-get install stress`) the system in different ways.

I'm happy to leave this issue open for others to see and comment on, but I'm afraid we can't support the RT patch series - we have more than enough to do with mainline Linux.

lategoodbye, 2017-10-29:
@az2022 Just for curiosity, what prevents you from using a mainline kernel + RT patches?

az2022, 2017-10-31:
@lategoodbye My Pi3 randomly hangs up.

vanfanel, 2017-11-08:
@az2022, @pelwell 
The problem with RT patches on the Pi seems to be related to the FIQ exception handler that the driver of the Synopsis DWC host USB controller:

https://www.osadl.org/Single-View.111+M5c03315dc57.0.html

https://autostatic.com/2017/06/27/rpi-3-and-the-real-time-kernel/

This could be easily fixed if you guys incorporate the fix to the Pi kernel sources.

@az2022 : try these kernel parameters
```

dwc_otg.fiq_enable=0
dwc_otg.fiq_fsm_enable=0
```

and your Pi shouldn't hang when using the RT patches. It's a "workaround", not a solution. We need the kernel guys to incorporate the fix or manually apply that patch...

pelwell, 2017-11-08:
I presume you mean [this patch](https://www.osadl.org/monitoring/patches/rbs3s/usb-dwc_otg-fix-system-lockup-when-interrupts-are-threaded.patch.html)?

Any comments, @P33M ?

vanfanel, 2017-11-08:
@pelwell 
There's an updated version of the patch (not totally up-to-date) here:

https://github.com/fedberry/kernel/blob/master/usb-dwc_otg-fix-system-lockup-when-interrupts-are-threaded.patch

You could easily have an RT branch... there's a lot of demand for RT kernels on the Raspberry, and finding matching RT patches is not easy.

P33M, 2017-11-09:
Currently the FIQ lock is held only within the IRQ handler, which implies no other IRQs can preempt the dwc_otg interrupt and the driver-specific spinlock is held. I suppose that adding a "real" IRQ save/restore would be harmless in the non-RT case.

The latency graph on the autostatic.com site is interesting. If the upper bound for latency is 75uS then this approaches the required hard-realtime horizon for some of the USB usecases - high-speed Isochronous transfers in particular. We know that even with the FIQ helping, it's possible to miss scheduling isochronous transfers at the boundary of every URB as this requires manipulating software queues. If RT makes this problem go away, then it'd be an interesting avenue to explore.

vanfanel, 2017-11-09:
@P33M: Would that improve USB HID input latency?

P33M, 2017-11-09:
No. The dwc_otg driver doesn't know about scheduling constraints higher up in the USB stack.

By default we restrict HID mouse polling to 60Hz to reduce the Xserver load on BCM2835-based devices, so setting usbhid.mousepoll=0 in cmdline.txt will restore the default polling interval.

vanfanel, 2017-11-09:
@P33M 
Do you only restrict polling rate for mouse, or for usb joysticks too? What is the default polling rate for usb joysticks in Raspbian kernel on the current versions? Isn't it 120Hz as in mainline kernels?

Also, a basic question I have been asking myself these days, and maybe you can answer because I can't find the information: usb joysticks and mice are accessed through the modern evdev interface in all low-latency APIs available on the Pi, like SDL2 and LibRetro/RetroArch. They all use evded. (X11 is not important for this matter).
Well, evdev is an event-driven interface: it's main objetive is precissely to avoid the old polling techniques. **How is the kernel polling rate (usbhid.jspoll, usbhid.mousepoll) related to evdev then?**
I don't get it, and you can be sure I have searched for an answer low and high...

az2022, 2017-11-09:
@vanfanel Spot on. For a quick check I applied the command line arguments and I have not been able to reproduce the problem for 2+ hours now, which is very encouraging!  Will leave it for over night test. 
BTW, since these are OTG related arguments, I imagine folks who don't have USB devices connected to their Pi3 may not see this issue?

nettings, 2017-11-11:
I'd be interested in "RT-ready" RPi kernels, too. Running https://autostatic.com/2017/06/27/rpi-3-and-the-real-time-kernel/ for a few projects now, and currently trying to update to the latest 4.9-rt. It would be a huge help if the lockup workaround was in RPi-linux mainline if it doesn't harm the common usecases (or is even beneficial). Willing to test...

az2022, 2017-11-15:
Setting kernel paramaters, dwc_otg.fiq_enable=0, dwc_otg.fiq_fsm_enable=0 works well with both keyboard and mice but after some testing noticed it made my USB mic unusable. Applied the usb-fiq patch and rebuilt, USB mic now works.

nettings, 2017-11-15:
I'm just building a kernel from 4.9.61 (current HEAD) with https://www.kernel.org/pub/linux/kernel/projects/rt/4.9/patch-4.9.47-rt37.patch.gz and https://github.com/fedberry/kernel/blob/master/usb-dwc_otg-fix-system-lockup-when-interrupts-are-threaded.patch. All NEW options with default answer, full preemption, HZ=1000, one implicit declaration error (removed -Werror-implicit-function-declaration for now). Seems to compile, will test tomorrow.

JamesH65, 2017-12-05:
What is the current status of this issue? Should it be closed?

nettings, 2017-12-05:
Maybe we should open another issue requesting that just the FIQ patch is merged into RPi mainline? So that the RT patch can be applied out-of-the-box....

az2022, 2018-01-13:
Sorry about the delayed response. Perhaps the issue could be well published/documented where users could immediately notice and choose whether to apply the USB patch or not. For a long term solution a RT ready image with the USB patch would be nice, but I understand the resource shortage as well.

markusr, 2018-01-25:
I got a recent RT kernel running on a Raspi 3. 

```
pi@raspberrypi:~/rt/rt-tests $ uname -a
Linux raspberrypi 4.14.14-rt10-v7 #1 SMP PREEMPT RT Thu Jan 25 14:08:05 UTC 2018 armv7l GNU/Linux
```

I adapted the instructions from here: http://ffall007.16mb.com/post/applying-preempt-rt-kernel-patch-on-raspberry-pi/

1. Download kernel source and extract.
```
wget https://github.com/raspberrypi/linux/archive/23816e63fcc5efa8400f248304db13d5df69792f.zip
```

I could not compile a later version, because of this patch:
https://github.com/raspberrypi/linux/commit/2f77f727fbcfae6c6078b84bd878f7cb9a36d6be
which is not compatible with the ```patch-4.14.12-rt10.patch```

2. Download RT patch
```
wget https://www.kernel.org/pub/linux/kernel/projects/rt/4.14/patch-4.14.12-rt10.patch.gz
```

3. Apply RT patch
```
zcat ../patch-4.14.12-rt10.patch.gz | patch -p1
```

Then follow the "normal" compilation and installation instructions for a RT kernel.

TiejunChina, 2018-02-23:
I'd like to maintain -rt branch,

https://github.com/TiejunChina/linux/tree/rpi-4.14.y-rt

This is based on rpi-4.14.y but with the latest v4.14.20-rt17.x.  This can boot on pi3,

# uname -a
Linux linux.suse 4.14.20-rt17-v8+ #1 SMP PREEMPT RT Fri Feb 23 13:19:32 CST 2018 aarch64 aarch64 aarch64 GNU/Linux

I also did cyclictest as well.

@pelwell , You know, we also had a little bit of discussion on this previously.  Please let me know how to put this into official raspberrypi/linux.

pelwell, 2018-03-01:
The first Raspberry Pi PREEMPT_RT kernel branch went live this morning:  https://github.com/raspberrypi/linux/tree/rpi-4.14.y-rt

Thank you for your efforts and continuing support, @TiejunChina.

nettings, 2018-03-01:
Amazing! Thanks to all contributors! Can't wait to try this one out.

paul-1, 2018-03-01:
Thanks for maintaining this, will save me some patching time. 

Out of curiosity, have you been able to run this on an armv6 device, with FIQ enabled?  It will run with FIQ disabled, but suffers from all of the pre-fiq usb problems.

TiejunChina, 2018-03-02:
@paul-1 Sorry, I have no armv6 device in my hand. I will try to get access to that later. But now, what is that problem? Is that a lockup issue?

paul-1, 2018-03-02:
Just hangs during boot.   There are no kernel panics/oops.

TiejunChina, 2018-03-02:
@paul-1 Would like to try this branch?

T-vK, 2018-03-05:
@TiejunChina I've been trying to build a working kernel out of your branch for over a week now. But I just can't get it to boot properly. I have posted more details in this forum thread, it would be nice if you could take a look at it: https://www.raspberrypi.org/forums/viewtopic.php?f=71&t=205604&p=1281396&sid=add236ca46e8f3ca05b9614a56f374d1#p1281396

pelwell, 2018-03-05:
It's on my list to look at - probably this afternoon.

pelwell, 2018-03-05:
Try adding `dtoverlay=dwc2` to config.txt. FIQ support is compromised in some way on the -rt branch.

soundchekk, 2018-03-12:
I just compiled the latest git rt-kernel branch. (I've been maintaining rt-kernels for Moode-Audio for a while)

dwc_otg related cmdline.txt options are: 
dwc_otg.fiq_fsm_enable=0 dwc_otg.fiq_enable=0 dwc_otg.nak_holdoff=0

cyclictest looks OK:
T: 0 (  334) P:80 I:500 C: 100000 Min:      8 Act:   12 Avg:   11 Max:      28

However. The old and known dwc_otg load up2 as seen below 11.2%  issue 
during idle operation still exists:
                                                        
 149 root     -51   0       0      0      0 R  11.2  0.0   1:05.54 irq/39-dwc2_hso                                                          
 144 root     -51   0       0      0      0 S   9.9  0.0   0:58.50 irq/39-3f980000                                                          
 148 root     -51   0       0      0      0 S   7.3  0.0   0:46.11 irq/39-3f980000                                                          

As you can see I also tried  the dwc2 overlay as suggested by pelwell.

Any ideas how to approach the subject?

soundchekk, 2018-03-12:
One more:

I forgot. I'm talking about:   
kernel 4.14.24-rt19
and I run a ArchLinux Arm system.

Below the loads with the dwc2 overlay removed again.

   80 root     -51   0       0      0      0 S  11.8  0.0   0:23.99 irq/39-dwc_otg_                                                                                                          
   78 root     -51   0       0      0      0 S  10.5  0.0   0:20.69 irq/39-dwc_otg                                                                                                           
   79 root     -51   0       0      0      0 S   7.6  0.0   0:15.63 irq/39-dwc_otg_                                                                                                          
  
I also tried to remove dwc_otg options from cmdline.txt. - now with the new kernel in place. 
After that the PI wouldn't boot properly anymore.

TiejunChina, 2018-03-13:
@soundchekk I think you should check if you rpi is always receiving irq/39-dwc_otg, like `cat /proc/interrupts` without enabling -rt Kconfig. If this is sort of irq/39-dwc_otg storm, that associated irq to irq/39-dwc_otg would spike cpu as a certain percentage in the case of -rt.

soundchekk, 2018-03-13:
I did some more testing.
Standard kernel: 4.14.24-1-ARCH

TC1:
cmdline.txt
"dwc_otg.fiq_fsm_enable=0 dwc_otg.fiq_enable=0 dwc_otg.nak_holdoff=0"
 
Average dwc_otg interrupts: 8164 /s
CPU load:  "0"

TC2:
cmdline.txt
"default dwc_otg - all options from TK1 taken out"

Average dwc_otg interrupts: 274 /s
CPU load:  "0"

########
rt-kernel: 4.14.24-rt19-sc1 - default config
#########
TC3:
cmdline.txt
"dwc_otg.fiq_fsm_enable=0 dwc_otg.fiq_enable=0 dwc_otg.nak_holdoff=0"
 
Average dwc_otg interrupts: 8244 /s
CPU load:  "12%"

TC4:
cmdline.txt
"default dwc_otg - all options from TK3 taken out"

Average dwc_otg interrupts: X /s
CPU load:  "X" 

Case failed because USB init seem to lock up at boot!

##################

Bottom line. 
The number of interrupts "normal vs. rt-kernel" are pretty much identical.
The FIQ... enable=0 params causing a hell of an IRQ load. 30 times higher. On both kernels.
The CPU impact of these high number of interrupts is quite a different one. 
At least on my setup the rt-kernel caused CPU load isn't looking OK.

Still. Don't know yet why my setup locks up (around USB dev init) when booting TC4.

TiejunChina, 2018-03-19:
@soundchekk Sorry for this delay response.

Could you try the latest rpi-4.14.y-rt? I validated this on my side with pi3 mode B where I preinstalled ubuntu mate 16.04.2. I think your TC4 should work now.

soundchekk, 2018-03-19:
with 4.14.27-rt19-sc1

TC5:
* kernel boots with FIQ on
* dwc_otg load: 0.3-0.7%
* dwc_otg interrupts:  278/s (still seems quite high) 

Perhaps you should have merged latest 4.14.27-rt21 patch.

Thx for looking into it.

TiejunChina, 2018-03-19:
@soundchekk Yes, I'm working on merging -rt21. I'm always curious why we have so many interrupts even in the case of none-rt.  You know, that means threaded irq always spikes on CPU for a while, as you obverse.

andreasBihlmaier, 2018-03-28:
After many years of handcrafting RT kernels for Pi for "normal" use cases, I might finally be able to let go of this occupation :)
Big thank you to @TiejunChina!

TiejunChina, 2018-04-01:
@andreasBihlmaier Thanks.  We are still working on some known issues but any contributions are welcome.

TiejunChina, 2018-04-01:
@jarodsmith Before we take a close look at your issue, please use the latest -rt branch version. Your -rt version is 4.14.22-rt17 is sort of outdated.

jarodsmith, 2018-04-01:
Thank you for the answer. I have cloned and built the latest branch 4.14.31-rt25+

I get exactly the same behaviour... Please see boot log below.

I would really appreciate it if you could help with this.

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.31-rt25+ (ja@ubuntu) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03)) #1
 PREEMPT RT Sun Apr 1 20:25:01 CEST 2018
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] Memory policy: Data cache writeback
[    0.000000] cma: Reserved 8 MiB at 0x1b400000
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 113680
[    0.000000] Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:1D:65:20 vc_mem
.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  ignore_log_level dwc_otg.lpm_enable=0 console=tty1 console=ttyS0,115200 root=PARTUUID=cb4a1ae3-02 rootfstype=ex
t4 elevator=deadline fsck.repair=yes rootwait dwc_otg.fiq_fsm_enable=0 dwc_otg.fiq_enable=0 dwc_otg.nak_holdoff=0
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 434960K/458752K available (6643K kernel code, 595K rwdata, 2020K rodata, 440K init, 701K bss, 15600K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xdc800000 - 0xff800000   ( 560 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xdc000000   ( 448 MB)
[    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
[    0.000000]       .text : 0xc0008000 - 0xc068500c   (6645 kB)
[    0.000000]       .init : 0xc090c000 - 0xc097a000   ( 440 kB)
[    0.000000]       .data : 0xc097a000 - 0xc0a0ef54   ( 596 kB)
[    0.000000]        .bss : 0xc0a14b80 - 0xc0ac3fdc   ( 702 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 24294 entries in 72 pages
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000]  No expedited grace period (rcu_normal_after_boot).
[    0.000000]  Tasks RCU enabled.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000033] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000059] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000149] bcm2835: system timer (irq = 27)
[    0.000724] Console: colour dummy device 80x30
[    0.001846] console [tty1] enabled
[    0.001893] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060326] pid_max: default: 32768 minimum: 301
[    0.060835] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.060851] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.062185] Disabling memory control group subsystem
[    0.062359] CPU: Testing write buffer coherency: ok
[    0.063783] Setting up static identity map for 0x8200 - 0x8238
[    0.070478] Hierarchical SRCU implementation.
[    0.072918] devtmpfs: initialized
[    0.087674] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.088308] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.088411] futex hash table entries: 256 (order: 1, 8192 bytes)
[    0.089913] pinctrl core: initialized pinctrl subsystem
[    0.092037] NET: Registered protocol family 16
[    0.095425] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.104783] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.104870] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.105054] Serial: AMBA PL011 UART driver
[    0.108833] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.109833] uart-pl011 20201000.serial: could not find pctldev for node /soc/gpio@7e200000/uart0_pins, deferring probe
[    0.165995] bcm2835-dma 20007000.dma: DMA legacy API manager at dc80d000, dmachans=0x1
[    0.168911] SCSI subsystem initialized
[    0.169286] usbcore: registered new interface driver usbfs
[    0.169465] usbcore: registered new interface driver hub
[    0.169844] usbcore: registered new device driver usb
[    0.171580] raspberrypi-firmware soc:firmware: Attached to firmware from 2018-03-23 16:31
[    0.174502] clocksource: Switched to clocksource timer
[... 139 lines cut ...]
[    3.057113] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
[    3.057229] VFS: Mounted root (ext4 filesystem) readonly on device 179:2.
[    3.061571] devtmpfs: mounted
[    3.065824] Freeing unused kernel memory: 440K
[    3.065833] This architecture does not have kernel memory protection.
[    3.515351] Indeed it is in host mode hprt0 = 00001101
[    3.626817] systemd[1]: System time before build time, advancing clock.
[    3.793116] usb 1-1: New USB device found, idVendor=05ac, idProduct=1402
[    3.803188] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[... cut ...]
[    3.946472] systemd[1]: Detected architecture arm.
[   34.409263] systemd[1]: Set hostname to <raspberrypi>.
[   65.765741] systemd[1]: Reached target Swap.
[   95.845774] systemd[1]: Listening on udev Control Socket.
[  125.925461] systemd[1]: Listening on fsck to fsckd communication Socket.
[  156.006834] systemd[1]: Created slice User and Session Slice.
[  186.085876] systemd[1]: Started Forward Password Requests to Wall Directory Watch.
[  216.196165] systemd[1]: Listening on Journal Socket (/dev/log).

TiejunChina, 2018-04-02:
@jarodsmith This is another similar thing to #2424 since we can see "Disabling IRQ #53" here as well. @pelwell is working on this.

paul-1, 2018-07-18:
Just built 4.14.52-rt34, and it is now booting just fine on a RPi0 & RPi0W

TiejunChina, 2018-07-19:
@paul-1 Sounds good :)

FischC, 2018-07-26:
I have "raspberrypi 4.14.52-rt34-v7+ #1 SMP PREEMPT RT" running stable with kernel parameters "dwc_otg.fiq_enable=0 dwc_otg.fiq_fsm_enable=0" or device tree overlay "dtoverlay=dwc2", but audio recordings from my USB sound card (C-Media Electronics, Inc.) are of poor quality (stuttering, noise, etc.).
Setting the CPU governor to "performance" and "/usr/bin/chrt -a -r -p 99" on the process (e.g. areocrd) improves the situation al little but sound quality is still poor/not acceptable.
I have no problem living with approx. 30% CPU overhead but I need good sound quality. Any suggestions?

TiejunChina, 2018-07-26:
@FischC Did this happen with our previous -rt kernel?  Anyway, I just update -rt to 4.14.58-rt34-v7.

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.14.58-rt34-v7+ #25 SMP PREEMPT RT Thu Jul 26 17:55:41 CST 2018 armv7l GNU/Linux

soundchekk, 2018-07-26:
Use  a standard kernel if you don't really know what you're doing.
This is not the place to discuss application issues. Especially not "soundquality" related subjects.

Trying to use a rt-kernel properly on the Pi is trying to tune a 20 strings guitar by ear.

E.g. Using a USB DAC and network at the same time usually causes severe hickups if the entire 
setup is not tuned properly. You might even never get it working properly. That gets much worse on higher samplerates (load).
By far the weakest spot of the PI the combined USB/networking chip and a single USB lane towards the CPU. That'll always cause major challenges.
Just lifting one priority to 99, basically locking up all the rest of the PI, is pretty much useless. 

There's more. Not to be discussed over here I'd say. Except there'd be a real issue associated.

FischC, 2018-08-23:
@TiejunChina

I've updated to Linux raspberrypi 4.14.59-rt37-v7+ #4 SMP PREEMPT RT, removed the fiq stuff and now the kernel doesn't freeze that quickly.
But with high network traffic (iperf) on a USB Wifi adapter I can still force kernel freezes, etc.. I'm not sure how to address this issue then? Create a new ticket?

FischC, 2018-09-06:
@TiejunChina I've tested with Linux "raspberrypi 4.14.66-rt40-v7+ #6 SMP PREEMPT RT Tue Aug 28 13:54:36 CEST 2018 armv7l GNU/Linux" and I can still reproduce the problem. Shall I create a new ticket?

TiejunChina, 2018-09-10:
@FischC Please, go ahead.

TiejunChina, 2018-09-10:
I'd like to close this issue since we have had our -rt branch for a while.

closed 2018-09-10 by TiejunChina