raspberrypi/linux #2245: KGDB SESSION ( problem with start)

piter1993vc, 2017-10-26:
Hello,
Does anybody know how to run properly kgdb session on serial session with vmlinux file and magic_sysrq trigger?
I have kernel 4.9.58 (and aufs 4.9) on raspberry pi 3 B this is my cmdline.txt line:
elevator=deadline  console=serial0,38400 kgdboc=serial0,38400  bootmenutimeout=10
in config.txt i have enable_uart=1.
When I boot my raspberry with kgdboc=serial0,38400  then device froozen ( without kgdbwait )
Without kgdboc=serial0,38400 device start correctly. The problem is appearing in addidtion on kernel 4.13 with aufs 4.13.
I think that the problem is links with boot log:
 sdhost-bcm2835 3f202000.sdhost: could not get clk, deferring probe
mmc-bcm2835 3f300000.mmc: could not get clk, deferring probe
bcm2835-aux-uart 3f215040.serial: could not get clk: -517
This problem was not appearing on kernel 4.4.50 with aufs 4.4.
Maybe the problem is with mutiplatform ( global changes in kernel for bcm ). I don't know how to boot correctly boot kgdb and debug kernel panic after insmod my local gpio driver.
this driver don't has problem on kernel 4.4.50.

This is problem with gpio  driver:
[  103.127141] Unable to handle kernel paging request at virtual address b94ef034
[  103.149096] pgd = 80004000
[  103.157299] [b94ef034] *pgd=35c43811, *pte=00000000, *ppte=00000000
[  103.176340] Internal error: Oops: 7 [#1] SMP ARM

duanwujie, 2017-10-27:
the 4.9.58 (and aufs 4.9) with kgdboc=serial0,38400 are working  for kgdb?

piter1993vc, 2017-10-27:
No. If has been working then i didn't write this post.

JamesH65, 2017-12-05:
It would be worth posting this question on the forum to get more people to see it.

piter1993vc, 2017-12-05:
Okey. Thanks for you good advice.

JamesH65, 2018-06-27:
Closing as this issue is not kernel related.

closed 2018-06-27 by JamesH65