raspberrypi/linux #2249: Stretch Lite (2017-09-07) WiFi not working on Pi Zero W

HankB, 2017-10-28:
After performing a clean install of 2017-09-07-raspbian-stretch-lite.img and adding a wpa_supplicant.conf file (and ssh) file to boot, the Wifi does not come up. Information on configuring this is near the end of https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=191252&p=1227953#p1227953 and present contents of /etc/wpa_supplicant/wpa_supplicant.conf are
```
pi@raspberrypi:~ $ sudo cat /etc/wpa_supplicant/wpa_supplicant.conf 
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
	ssid="IoT_Test"
	psk="not the real password"
	key_mgmt=WPA-PSK
}
pi@raspberrypi:~ $ 
```
If the SD card is then plugged into a Pi 3B the WiFi then works. Following 'apt update and 'apt-upgrade' the WiFi now works when the card is swapped back to the Pi Zero W.

My intent with this bug report is to either discover an error in my WiFi configuration or request that this test case be added to the testing performed before the nest Stretch Lite release,

HankB, 2017-11-03:
Additional info: Problem seems to relate to the firmware versions on the Pi Zero W. Following firmware exhibits the problem.

```
hbarta@niwot:~ $ dmesg|grep -i firmware
[    0.266997] raspberrypi-firmware soc:firmware: Attached to firmware from 2017-07-03 14:17
[    7.213852] brcmfmac: Firmware version = wl0: Aug  7 2017 00:46:29 version 7.45.41.46 (r666254 CY) FWID 01-f8a78378
hbarta@niwot:~ $ 
```
After updating firmware the original problem no longer manifests.

```
pi@raspberrypi:~ $ dmesg|grep -i firmware
[    0.269160] raspberrypi-firmware soc:firmware: Attached to firmware from 2017-08-08 12:01
[   19.768083] brcmfmac: Firmware version = wl0: Aug  7 2017 00:46:29 version 7.45.41.46 (r666254 CY) FWID 01-f8a78378
pi@raspberrypi:~ $ 
```

JamesH65, 2017-12-05:
@XECDesign Is the firmware specified now standard? Odd that the firmware versions do appear to be the same....

Can this be closed?

XECDesign, 2017-12-05:
The firmware version has been bumped since the 2017-09-07 image, but I don't know if that has the fix or what the problem even is. I test headless wifi setup on a pi zero w before releasing images, and this isn't something I've seen.

It can be closed if @HankB is happy that the new images work as expected for him. Otherwise, we'd need to know which firmware or kernel commit has the fix.

HankB, 2017-12-05:
>> Odd that the firmware versions do appear to be the same....
[    0.266997] raspberrypi-firmware soc:firmware: Attached to firmware from 2017-07-03 14:17
[    0.269160] raspberrypi-firmware soc:firmware: Attached to firmware from 2017-08-08 12:01

What is this?

>> Otherwise, we'd need to know which firmware or kernel commit has the fix.

Please look three posts up for the firmware versions involved.
I will repeat that in order to resolve this I had to:
1) Update Raspbian Lite on another Pi that had working networking. (IIRC that was a Pi 2 B using Ethernet.)
2) Update firmware on the Pi Zero using the updated Raspbian (Lite) on the SD card.

Had I not had another Pi to use, I would have been stuck with a non-functional Pi Zero W.

I'm satisfied that I have all of my units working at present. It is up to the Foundation to decide if the testing of new releases should include all extant variations on the product. It seems to me that is not presently the case.

XECDesign, 2017-12-05:
> I'm satisfied that I have all of my units working at present. It is up to the Foundation to decide if the testing of new releases should include all extant variations on the product. It seems to me that is not presently the case.

It does. All releases (since the zero w has been released) work without wifi problems on the pi zero w in front of me.

> Please look three posts up for the firmware versions involved.

I meant the exact rpi-update revision that the problem was fixed in, but that's okay.
There are quite a few commits in the range you gave, but this one could've been the fix:
https://github.com/raspberrypi/linux/pull/2126

In any case, we can't test for an issue that only some people see and we don't. Unfortunately, some issues can only be caught after the image is release.

topherbuckley, 2018-01-16:
I'm having the same issue as @HankB, but on a Pi 3 B. Wired headless SSH works, but will not connect to WIFI.  As this was a month or so ago and possibly a different bug, let me know if it is better to start a new issue. 

Using same steps as [here](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=191252&start=25#p1220111)

RASPBIAN STRETCH LITE
Version:November 2017
Release date:2017-11-29
Kernel version:4.9

HankB, 2018-01-16:
@topherbuckley This issue is specific to a Pi Zero W. I was visiting a couple weeks ago and initial attempts to get my Pi 3 on local WiFi were not successful. I did not pursue a solution as we were only there for a couple days. You might try asking on one of the support forums to see if someone can help.

jakemagee, 2018-01-16:
Looks like the issue has been resolved... can you close it @HankB ?

@topherbuckley , if your still having issues then I would suggest opening a new ticket (since, as HankB pointed out, this ticket was specifically for a Pi Zero W).  I use the following first line in my wpa_supplicant.conf file instead... "ctrl_interface=/var/run/wpa_supplicant".  That other "ctrl_interface" line looks suspicious to me.  Also, assuming that your access point is setup correctly, you might make sure that the wpa_supplicant.conf file is in unix format (not dos)... not sure if it causes problems or not.

HankB, 2018-01-16:
Closed as not feasible to fix - limited future impact.

closed 2018-01-16 by HankB