raspberrypi/linux #2250: Question about VCHIQ kernel module support

lategoodbye, 2017-10-29:
One point on my upstream VCHIQ TODO list is fixing kernel module support. Currently it's possible to build the driver as a kernel module, but it doesn't properly clean up all resources on unloading (e.g. kthreads).

My idea is to create a patch like this:

```
diff --git a/drivers/staging/vc04_services/interface/vchiq_arm/vchiq_core.c b/drivers/staging/vc04_services/interface/vchiq_arm/vchiq_core.c
index 486be99..1dbae48 100644
--- a/drivers/staging/vc04_services/interface/vchiq_arm/vchiq_core.c
+++ b/drivers/staging/vc04_services/interface/vchiq_arm/vchiq_core.c
@@ -2095,7 +2095,7 @@ slot_handler_func(void *v)
 
 	DEBUG_INITIALISE(local)
 
-	while (1) {
+	while (!kthread_should_stop()) {
 		DEBUG_COUNT(SLOT_HANDLER_COUNT);
 		DEBUG_TRACE(SLOT_HANDLER_LINE);
 		remote_event_wait(state, &local->trigger);
@@ -2185,7 +2185,7 @@ recycle_func(void *v)
 	VCHIQ_STATE_T *state = (VCHIQ_STATE_T *) v;
 	VCHIQ_SHARED_STATE_T *local = state->local;
 
-	while (1) {
+	while (!kthread_should_stop()) {
 		remote_event_wait(state, &local->recycle);
 
 		process_free_queue(state);
@@ -2203,7 +2203,7 @@ sync_func(void *v)
 	VCHIQ_HEADER_T *header = (VCHIQ_HEADER_T *)SLOT_DATA_FROM_INDEX(state,
 		state->remote->slot_sync);
 
-	while (1) {
+	while (!kthread_should_stop()) {
 		VCHIQ_SERVICE_T *service;
 		int msgid, size;
 		int type;
```

Are there any concerns about this, before i start working on this?

pelwell, 2017-10-30:
My concern would be that the VPU is not expecting the service to be restarted so VCHIQ would be a one-shot module. Is there no way to declare a module as not unloadable?

lategoodbye, 2017-10-30:
In this case we better declare the driver as builtin.

pelwell, 2017-10-30:
Really? I can see a useful difference between an optional driver that can't be unloaded and one which is built in. In reality, how many modules are ever unloaded?

lategoodbye, 2017-10-30:
Sure there is a difference, but a proper kernel module should be able to unload. I could ask on kernel newbies about a one-shot function.

lategoodbye, 2017-10-30:
Btw is there something like a point of no return during VCHIQ driver init, where we shouldn't kill those kernel threads?

lategoodbye, 2017-11-05:
According to Greg KH we cannot prevent driver unloading and this issue should be fixed in the VPU firmware instead.

JamesH65, 2017-12-05:
Seems unlikely that the VPU firmware will be changed to do this. Any thoughts on progressing or closing this issue?

lategoodbye, 2017-12-07:
Would be nice to get answer to my question on October 30 otherwise this could be closed.

pelwell, 2017-12-07:
The connection process is symmetrical - vchiq_core.c is essentially the same on the VPU and the ARMs - so you can see that they both send a CONNECT message and wait for a CONNECT message in return, after which they are connected and will not try to connect again. If that doesn't answer your question you will have to rephrase it.

lategoodbye, 2017-12-07:
Thanks, this is sufficient.

closed 2017-12-07 by lategoodbye