raspberrypi/linux #2251: hciuart / hciattach cannot be restarted

lucasvdp, 2017-10-30:
you cannot stop the hciuart service and then start it again. This is needed to go to a different run level and back.
On first boot hciattach switches the bluetooth device to a higher serial baud rate. When running hciattach a second time it seems to try to connect to the bluetooth device at the old baud rate (115200). This will not work because the bluetooth device is still configured at the higher baud rate.

workaround: change hciuart.service or btuart to use a bautrate of 115200

possible solutions I can think off:
- try also the higher baud rate in hciattach
- switch back to 115200 when stopping hciattach
- somehow reset the bluetooth chip (does happen on reboot) maybe by disabling clock to gpio43?

JamesH65, 2017-10-30:
@pelwell Will this be affected by recent changes?

lucasvdp, 2017-10-30:
It is an old issue, I used to "solve" this by changing the baud rate in /lib/systemd/system/hciuart.service. In Stretch you need to edit /usr/bin/btuart.

pelwell, 2017-10-30:
You can do better than that by resetting the modem. I have a script called `btreset` that wiggles the correct GPIO lines:
```
sudo killall hciattach
if grep -a Zero /proc/device-tree/model; then
  raspi-gpio set 45 op dl
  sleep 1
  raspi-gpio set 45 op dh
else
  /opt/vc/bin/vcmailbox 0x38041 8 8 128 0
  sleep 1
  /opt/vc/bin/vcmailbox 0x38041 8 8 128 1
fi
sleep 4
sudo btuart
```
The 3B case is more tricky because the Bluetooth reset GPIO is on the GPIO expander.

We could add the script to the standard image, but I can't help thinking there might be a cleaner way to achieve this.

XECDesign, 2017-10-30:
Perhaps a vcgencmd?

pelwell, 2017-10-30:
Something along those lines, unless there is a DT method for describing serial-attached BT modems - perhaps using the new serdev mechanism.

JamesH65, 2017-12-05:
Is there likely to be anything coming from this issue, or should it be closed? @

beren12, 2018-03-27:
This is interesting, could we knock together an init script for those of us who don't use systemd? seems like start could just be calling bturat after bluetooth comes up, and down/restart could use your code pelwell.

JamesH65, 2018-06-27:
I'm inclined to close within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

JamesH65, 2018-08-31:
Closing due to lack of activity. Please request to be reopened if you feel this issue is still relevant.

closed 2018-08-31 by JamesH65

samueleforconi, 2020-02-03:
Hi,
referring to the message written by @pelwell 

> You can do better than that by resetting the modem. I have a script called `btreset` that wiggles the correct GPIO lines:
>
> ```
> sudo killall hciattach
> if grep -a Zero /proc/device-tree/model; then [...]

I have checked this script on a Raspberry 3B and it works correctly (the bluetooth reset GPIO is toggled and the bluetooth subsystem can be restarted successfully), but on a Raspberry Pi 3B+ it doesn't work, maybe the bluetooth reset pin is not the same as in the 3B. So I expect that this script will not work even for RPi4...
@pelwell Could you please provide the bluetooth reset commands also for RPi 3B+ and RPi 4? I'm actually using RPi to build headless BLE receivers and the reset functionality for the bluetooth peripheral is really important for products like these...

pelwell, 2020-02-03:
Testing here shows the reset script not working for me either. Please post your findings at this more recent (and still open) issue: https://github.com/raspberrypi/linux/issues/2832

henzef, 2020-02-24:
Please reopen. This is not fixed and no sufficent workaround exists for 3b+ and 4b

JamesH65, 2020-02-24:
So #2832 is not an acceptable alternative? Would prefer to only have one issue open if they are indeed duplicates.

henzef, 2020-02-24:
#2832 does not describe the issue I have, however having a working reset of the bluetooth chip via gpios would completely solve my problems.