raspberrypi/linux #2255: /proc/cpuinfo reporting Hardware and Revision incorrected

jeffersonchua, 2017-11-03:
Something changed a few days ago in the kernel ...

3 days ago ...
# cat /proc/cpuinfo
        Hardware        : BCM2835
        Revision        : a22082
        Serial          : 00000000de1e6ef7

Now ...
# cat /proc/cpuinfo
        Hardware        : Generic DT based system
        Revision        : 0000
        Serial          : 00000000da20a410

The only difference is after cloning ..
commit 349e0ba59807962ab1c796e1c54313d824759973 (HEAD -> rpi-4.14.y)

Am I missing something? Running on pi3.

Thanks,
Jeff

jeffersonchua, 2017-11-03:
Reverting this patch resolved the problem  ...

commit f96f5f8b0b824bbacf14b8c38ba18f349615ce2d
Author: Phil Elwell <pelwell@users.noreply.github.com>
Date:   Thu Nov 2 09:16:17 2017 +0000

    bcm2710: Refactor bcm2710.dtsi using bcm2837.dtsi (#2236)

    The 4.14 tree includes bcm2837.dtsi - use it as a basis for
    the downstream bcm2710.dtsi, which only contains a few tweaks.

    See: https://github.com/raspberrypi/linux/issues/2235

    Signed-off-by: Phil Elwell <phil@raspberrypi.org>

pelwell, 2017-11-03:
The new .dtsi was just a little too lean and mean. The bcm2385 board support code doesn't recognise `brcm,bcm2837` as a valid compatible string - adding it as a valid option also seems to work but has potentially wider consequences, so instead I've restored `brcm,bcm2836` as a fallback in the .dtsi.

jeffersonchua, 2017-11-03:
>  bcm2710: Fix compatible string in bcm2710.dtsi 

That fixed it!

Thank you!
Jeff

JamesH65, 2017-12-05:
@jeffersonchua Can this now be closed?

jeffersonchua, 2017-12-05:
On Tue, Dec 5, 2017 at 5:58 PM, James Hughes <notifications@github.com>
wrote:

> @jeffersonchua <https://github.com/jeffersonchua> Can this now be closed?
>

Yes. Fixed. Thank you!

Jeff

closed 2017-12-05 by pelwell