raspberrypi/linux #2256: Raspbian kernel update breaks network bridging

jvonau, 2017-11-03:
root@box:/opt/iiab/iiab# apt list -a raspberrypi-kernel
Listing... Done
raspberrypi-kernel/stable,now 1.20171029-1 armhf [installed,automatic]

root@box:/opt/iiab/iiab# uname -a
Linux box.lan 4.9.41-v7+ #1023 SMP Tue Aug 8 16:00:15 BST 2017 armv7l GNU/Linux

root@box:/opt/iiab/iiab# brctl addbr br1
add bridge failed: Package not installed

[forum reference](https://www.raspberrypi.org/forums/viewtopic.php?t=196623)

asavah, 2017-11-05:
Have you rebooted after updating the kernel ?
I don't think so.
You are running  4.9.41-v7 kernel while 1.20171029-1 is 4.9.59

jvonau, 2017-11-05:
Thanks, that was it, didn't make the 20171029-1 -> 4.9.59 connection when looking at apt list as the 4.9.59 reversion number is not part of the package name. Sorry for the noise.

closed 2017-11-05 by jvonau