raspberrypi/linux #2257: Write / read camera OTP memory

mikemccabe, 2017-11-03:
It would be very handy to be able to assign serial numbers to camera modules, to track focus, defects, etc.  (We have an installation that uses thousands of cameras.)

I've read in various places that the ov5647 has 32 bytes of one-time programmable memory, this seems like a perfect place to put a serial, or of course calibration information.  Is there a way to make this writable/readable from the pi?

pelwell, 2017-11-03:
It should already be accessible from the Pi in the sense that, provided you don't try and use the standard camera APIs at the same time, you should be able to use the other (VPU) I2C interface from the ARM. The OTP is at offsets 0x3d00 0x3d1f, with a few control registers at 0x3d20 and 0x3d21. There's a datasheet [here](https://cdn.sparkfun.com/datasheets/Dev/RaspberryPi/ov5647_full.pdf).

JamesH65, 2017-12-05:
@mikemccabe Are you happy to close this issue?

mikemccabe, 2017-12-05:
Marking closed - thanks!

closed 2017-12-05 by mikemccabe