raspberrypi/linux #2258: Include CONFIG_USB_DWCOTG as module fails

malikolivier, 2017-11-05:
I included CONFIG_USB_DWCOTG as a module.

However, I get a compile error as below:

```
ERROR: "DWC_TASK_HI_SCHEDULE" [drivers/usb/host/dwc_otg/dwc_otg.ko] undefined!
```

I included by defconfig here: https://gist.github.com/malikolivier/abd6d96bd559476c39a12efd86ada63e

I don't know yet if it's something unique to the linux kernel for raspberrypi. Does anyone have some idea?
I suspect that DWC_TASK_HI_SCHEDULE, which is defined in `drivers/usb/host/dwc_common_port/dwc_common_linux.c` is not included when CONFIG_USB_DWCOTG is built as a module.

pelwell, 2017-11-05:
The standard Raspberry Pi configurations build dwc_otg as an in-kernel module, since it is required for an NFS root filing system. If you (or anyone) have a patch to make it work as a module then we would probably merge it, but until then if you want a module-based USB driver for Raspberry Pi then I would suggest CONFIG_USB_DWC2=m.

malikolivier, 2017-11-06:
Thanks for the feedback.
 
I was trying to reduce the boot time as much as possible, and it happens that dwc_otg_driver_init  is taking around 0.5ms on my config (compute module). That's why I thought about including USB support as a module.

I'll try you suggestions!

malikolivier, 2017-11-07:
It seems that USB fails to initialize with CONFIG_USB_DWC2 included (whether it built-in or as a module).

```
root@pi:~# lsusb --verbose
unable to initialize libusb: -99
```

```
root@pi:~# cat /lib/modules/$(uname -r)/modules.builtin | grep usb
kernel/drivers/usb/common/usb-common.ko
kernel/drivers/usb/core/usbcore.ko
kernel/drivers/usb/dwc2/dwc2.ko
```

pelwell, 2017-11-07:
You need to select the alternate driver with the dwc2 overlay:
```
pi@raspberrypi:~ $ dtoverlay -h dwc2
Name:   dwc2

Info:   Selects the dwc2 USB controller driver

Usage:  dtoverlay=dwc2,<param>=<val>

Params: dr_mode                 Dual role mode: "host", "peripheral" or "otg"

        g-rx-fifo-size          Size of rx fifo size in gadget mode

        g-np-tx-fifo-size       Size of non-periodic tx fifo size in gadget
                                mode
```
If you are down to caring about 500us delays, I hope you have set `initial_turbo=30` (or some similar time in seconds). You may also be able to make use of the `sd_overclock` dtparam, e.g. `dtparam=sd_overclock=62` or `dtparam=sd_overclock=83`.

malikolivier, 2017-11-07:
Thanks, it does work by setting the device tree to use the dwc2 USB driver. I did not think about it.

`dwc2` is way faster to initialize! Its initcall takes 80ms. Do you have any reason why you use dwc_otg as the default driver? Is dwc2 somewhat not recommended?

> If you are down to caring about 500us delays, I hope you have set initial_turbo=30 (or some similar time in seconds). You may also be able to make use of the sd_overclock dtparam, e.g. dtparam=sd_overclock=62 or dtparam=sd_overclock=83.

I tried initial_turbo=30 and sd_overclock=83 out but did not see any noticeable difference.
Either would give almost the same amount in systemd-analyze.

```
root@pi:~# systemd-analyze                                                                                                                                                                           
Startup finished in 822ms (kernel) + 1.821s (userspace) = 2.644s
```

pelwell, 2017-11-07:
dwcotg has been heavily optimised to use the ARM FIQ for the most frequent interrupts (of which there are many). The upstream dwc2 driver will struggle with certain device classes and traffic patterns.

malikolivier, 2017-11-07:
True, you seem right. One of the devices plugged by USB hanged several times when using dwc2. I went back to dwcotg and it went back to normal.

Are there some kind of online documents explaining what you just said in more details to which I can refer to?

pelwell, 2017-11-07:
The information will be scattered across a number of github and Raspberry Pi Forum posts.

Another option is to leave the driver built in but disabled, then to activate it at run time using the `dtoverlay` command. I added an `enable` parameter to the existing `dwc-otg` and found that it seems to work as expected. However, some time after enabling USB - sometimes about 20 seconds later but occasionally immediately - I get a kernel panic from within the dwc_otg_fig_fsm handler.

XECDesign, 2017-11-07:
Lots of good info here, although a fair bit of it is out of date:
https://www.raspberrypi.org/forums/viewtopic.php?t=70437

malikolivier, 2017-11-09:
Thank you very much! Will take a look to the available literature.

BTW, the build succeeds with CONFIG_USB_DWCOTG included as a module after the following patch is applied. I am only including the missing functions inside the module.

```patch
From 0b92d430b4ca9b07dc2d011ae3f0ab774b532308 Mon Sep 17 00:00:00 2001
From: Malik Olivier Boussejra <malik@boussejra.com>
Date: Thu, 9 Nov 2017 04:40:35 +0900
Subject: [PATCH] dwc_otg: Fix compile error when building as module

Fix the following error:

ERROR: "DWC_TASK_HI_SCHEDULE" [drivers/usb/host/dwc_otg/dwc_otg.ko]
undefined!
scripts/Makefile.modpost:91: recipe for target '__modpost' failed
make[1]: *** [__modpost] Error 1
Makefile:1199: recipe for target 'modules' failed
make: *** [modules] Error 2

---
 drivers/usb/host/dwc_otg/Makefile | 1 +
 1 file changed, 1 insertion(+)

diff --git a/drivers/usb/host/dwc_otg/Makefile b/drivers/usb/host/dwc_otg/Makefile
index 4872b8113b68..544a56ee6257 100644
--- a/drivers/usb/host/dwc_otg/Makefile
+++ b/drivers/usb/host/dwc_otg/Makefile
@@ -37,6 +37,7 @@ dwc_otg-objs	+= dwc_otg_pcd_linux.o dwc_otg_pcd.o dwc_otg_pcd_intr.o
 dwc_otg-objs	+= dwc_otg_hcd.o dwc_otg_hcd_linux.o dwc_otg_hcd_intr.o dwc_otg_hcd_queue.o dwc_otg_hcd_ddma.o
 dwc_otg-objs	+= dwc_otg_adp.o
 dwc_otg-objs	+= dwc_otg_fiq_fsm.o
+dwc_otg-objs	+= ../dwc_common_port/dwc_common_linux.o
 ifneq ($(CONFIG_ARM64),y)
 dwc_otg-objs	+= dwc_otg_fiq_stub.o
 endif
-- 
2.11.0
```

I currently do not have a PI near me to test against it. Will test later this week.
I am not really knowledgeable with Kbuild, and maybe this solution is not good from a coding standard perspective. I would be glad if I can have your opinion.

pelwell, 2017-11-09:
I'm happy in principle to take that patch (with a further patch that only includes the extra object file when building a module), but the module has to be usable when loaded - I'm concerned that the kernel panics I saw may be a side effect of the delayed start.

malikolivier, 2017-11-11:
The module seems unusable when loaded. I get the following kernel panic with dwc_otg built as a module.

```
[   10.837133] Bad mode in prefetch abort handler detected
[   10.842401] Internal error: Oops - bad mode: 0 [#1] ARM
[   10.847627] Modules linked in: evdev hid_generic usbhid hid dwc_otg fixed
[   10.854452] CPU: 0 PID: 1 Comm: systemd Not tainted 4.9.58-d2-v0.1 #1
[   10.860890] Hardware name: BCM2835
[   10.864292] task: dac58000 task.stack: dac46000
[   10.869097] PC is at dwc_otg_fiq_fsm+0x0/0x1f0 [dwc_otg]
[   10.874410] LR is at 0x1
[   10.876944] pc : [<bf02bec0>]    lr : [<00000001>]    psr: 000001d1
[   10.876944] sp : dac47ec8  ip : 00000000  fp : dac47f3c
[   10.888422] r10: 00000011  r9 : 000a0000  r8 : 00100000
[   10.893644] r7 : 00000000  r6 : daf1d240  r5 : dae58c00  r4 : bf02bec0
[   10.900168] r3 : 00000000  r2 : 00020000  r1 : 00000008  r0 : daf66400
[   10.906695] Flags: nzcv  IRQs off  FIQs off  Mode FIQ_32  ISA ARM  Segment user
[   10.914003] Control: 00c5387d  Table: 1ae24008  DAC: 00000055
[   10.919746] Process systemd (pid: 1, stack limit = 0xdac46188)
[   10.925577] Stack: (0xdac47ec8 to 0xdac48000)
[   10.929938] 7ec0:                   daf66400 00000008 00020000 00000000 bf02bec0 dae58c00
[   10.938120] 7ee0: daf1d240 00000000 00100000 000a0000 00000011 dac47f3c 00000000 dac47ec8
[   10.946303] 7f00: 00000001 bf02bec0 000001d1 ffffffff 000a0000 000a0000 ffffff9c daecb000
[   10.954485] 7f20: fffff000 c000f944 dac46000 00000000 dac47f4c dac47f40 c00f7df4 c00f7c38
[   10.962668] 7f40: dac47f94 dac47f50 c00da7d0 c00f7dcc dac47f74 dadff820 dadff820 00020000
[   10.970850] 7f60: 00000000 00000004 00000100 00000001 01c6b540 00000008 00000008 00000142
[   10.979032] 7f80: c000f944 dac46000 dac47fa4 dac47f98 c00da8f4 c00da6d8 00000000 dac47fa8
[   10.987214] 7fa0: c000f7a0 c00da8ec 01c6b540 00000008 ffffff9c be881860 000a0000 00000000
[   10.995396] 7fc0: 01c6b540 00000008 00000008 00000142 00000007 00000000 fffffffd be88189c
[   11.003578] 7fe0: 00000040 be880f80 b6c29cd8 b6c80778 60000010 ffffff9c 00000000 00000000
[   11.011749] Backtrace: 
[   11.014223] [<c00f7c2c>] (__alloc_fd) from [<c00f7df4>] (get_unused_fd_flags+0x34/0x38)
[   11.022235]  r10:00000000 r9:dac46000 r8:c000f944 r7:fffff000 r6:daecb000 r5:ffffff9c
[   11.030063]  r4:000a0000 r3:000a0000
[   11.033647] [<c00f7dc0>] (get_unused_fd_flags) from [<c00da7d0>] (do_sys_open+0x104/0x1f0)
[   11.041917] [<c00da6cc>] (do_sys_open) from [<c00da8f4>] (SyS_openat+0x14/0x18)
[   11.049232]  r9:dac46000 r8:c000f944 r7:00000142 r6:00000008 r5:00000008 r4:01c6b540
[   11.056990] [<c00da8e0>] (SyS_openat) from [<c000f7a0>] (ret_fast_syscall+0x0/0x34)
[   11.064652] Code: 83a00001 e89da800 e3a00000 e89da800 (e1a0c00d) 
[   11.070747] ---[ end trace e945173083866ff6 ]---
[   11.075807] Bad mode in prefetch abort handler detected
[   11.081056] Internal error: Oops - bad mode: 0 [#2] ARM
[   11.086279] Modules linked in: evdev hid_generic usbhid hid dwc_otg fixed
[   11.093104] CPU: 0 PID: 74 Comm: agetty Tainted: G      D         4.9.58-d2-v0.1 #1
[   11.100758] Hardware name: BCM2835
[   11.104159] task: daede300 task.stack: daed6000
[   11.108689] PC is at 0xdaec9604
[   11.111827] LR is at 0xb6f29c40
[   11.114968] pc : [<daec9604>]    lr : [<b6f29c40>]    psr: 600001d1
[   11.114968] sp : daed7fb0  ip : 0002a02c  fp : 00000000
[   11.126446] r10: 0002a224  r9 : 0000006f  r8 : bed3d92b
[   11.131668] r7 : 00000003  r6 : 0002a226  r5 : b6fd3f90  r4 : daec9604
[   11.138192] r3 : 00000000  r2 : 00000001  r1 : 00000008  r0 : daf66400
[   11.144719] Flags: nZCv  IRQs off  FIQs off  Mode FIQ_32  ISA ARM  Segment user
[   11.152026] Control: 00c5387d  Table: 1af48008  DAC: 00000055
[   11.157769] Process agetty (pid: 74, stack limit = 0xdaed6188)
[   11.163599] Stack: (0xdaed7fb0 to 0xdaed8000)
[   11.167956] 7fa0:                                     daf66400 00000008 00000001 00000000
[   11.176139] 7fc0: daec9604 b6fd3f90 0002a226 00000003 bed3d92b 0000006f 0002a224 00000000
[   11.184322] 7fe0: 0002a02c daed7fb0 b6f29c40 daec9604 600001d1 ffffffff 1bffa861 1bffac61
[   11.192492] Backtrace: no frame pointer
[   11.196335] Code: 00000000 00000000 00000000 00000000 (00000000) 
[   11.202428] ---[ end trace e945173083866ff7 ]---
[   11.207436] Bad mode in prefetch abort handler detected
[   11.212680] Internal error: Oops - bad mode: 0 [#3] ARM
[   11.217903] Modules linked in: evdev hid_generic usbhid hid dwc_otg fixed
[   11.224723] CPU: 0 PID: 50 Comm: systemd-journal Tainted: G      D         4.9.58-d2-v0.1 #1
[   11.233158] Hardware name: BCM2835
[   11.236558] task: dae1a300 task.stack: daea2000
[   11.241087] PC is at 0xdaec95cc
[   11.244226] LR is at 0xb6c92c7c
[   11.247368] pc : [<daec95cc>]    lr : [<b6c92c7c>]    psr: 800001d1
[   11.247368] sp : daea3fb0  ip : 00000000  fp : 000005e8
[   11.258845] r10: 00000800  r9 : b6f55220  r8 : b6d08be4
[   11.264068] r7 : 00000003  r6 : 00000012  r5 : 01f0ee78  r4 : daec95cc
[   11.270593] r3 : 00000000  r2 : 00000800  r1 : 00000008  r0 : daf66400
[   11.277118] Flags: Nzcv  IRQs off  FIQs off  Mode FIQ_32  ISA ARM  Segment user
[   11.284425] Control: 00c5387d  Table: 1ae8c008  DAC: 00000055
[   11.290167] Process systemd-journal (pid: 50, stack limit = 0xdaea2188)
[   11.296779] Stack: (0xdaea3fb0 to 0xdaea4000)
[   11.301137] 3fa0:                                     daf66400 00000008 00000800 00000000
[... cut ...]
[   11.330906] Code: 00000000 00000000 00000000 00000000 (00000000) 
[   11.336998] ---[ end trace e945173083866ff8 ]---
[   11.346015] systemd: 10 output lines suppressed due to ratelimiting
[   11.352511] Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
[   11.352511] 
[   11.361662] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
[   11.361662]
```

Maybe the best idea now is  a patch to prevent the user from setting CONFIG_USB_DWCOTG to "m"...

pelwell, 2017-11-11:
I'd be happy with that - sorry this didn't work out for you.

malikolivier, 2017-11-11:
If you are OK, you can take that patch:

```patch
From b0aafa7da820b739816e787ec58b1c2d69a3365c Mon Sep 17 00:00:00 2001
From: Malik Olivier Boussejra <malik@boussejra.com>
Date: Sat, 11 Nov 2017 23:33:46 +0900
Subject: [PATCH] USB_DWCOTG: Disable building dwc_otg as a module

When dwc_otg is built as a module, build will fail with the following
error:

ERROR: "DWC_TASK_HI_SCHEDULE" [drivers/usb/host/dwc_otg/dwc_otg.ko] undefined!
scripts/Makefile.modpost:91: recipe for target '__modpost' failed
make[1]: *** [__modpost] Error 1
Makefile:1199: recipe for target 'modules' failed
make: *** [modules] Error 2

Even if the error is solved by including the missing
DWC_TASK_HI_SCHEDULE function, the kernel will panic when loading
dwc_otg.

As a workaround, simply prevent user from building dwc_otg as a module
as the current kernel does not support it.

See details in https://github.com/raspberrypi/linux/issues/2258.

---
 drivers/usb/host/Kconfig | 5 +----
 1 file changed, 1 insertion(+), 4 deletions(-)

diff --git a/drivers/usb/host/Kconfig b/drivers/usb/host/Kconfig
index f3cecb6..fa39573 100644
--- a/drivers/usb/host/Kconfig
+++ b/drivers/usb/host/Kconfig
@@ -763,7 +763,7 @@ config USB_HWA_HCD
 	  will be called "hwa-hc".
 
 config USB_DWCOTG
-	tristate "Synopsis DWC host support"
+	bool "Synopsis DWC host support"
 	depends on USB && (FIQ || ARM64)
 	help
 	  The Synopsis DWC controller is a dual-role
@@ -772,9 +772,6 @@ config USB_DWCOTG
 	  Enable this option to support this IP in host controller mode.
 	  If unsure, say N.
 
-	  To compile this driver as a module, choose M here: the
-	  modules built will be called dwc_otg and dwc_common_port.
-
 config USB_IMX21_HCD
        tristate "i.MX21 HCD support"
        depends on ARM && ARCH_MXC
-- 
2.7.4
```

JamesH65, 2017-12-05:
@malikolivier @pelwell Can this now be closed?

malikolivier, 2017-12-05:
Yes, it can be closed! Thanks!

closed 2017-12-05 by pelwell