raspberrypi/linux #2260: Kernel Crash when using dwc_otg.speed=1 on an RPI3 with Stretch

dl1esk, 2017-11-08:
Hi there,

i use the Repeater Software Svxlink ( https://github.com/sm0svx/svxlink ) in Combination with an Raspberry PI 3 and an USB Soundcard.
When i force the USB Port to 1.1 i become an Kernel Oops and the system hangs up with the following Messages:

Message from syslogd@DB0CA at Nov 7 22:58:00 ...
kernel:[ 2277.323701] Internal error: Oops: 17 [#1] SMP ARM

Message from syslogd@DB0CA at Nov 7 22:58:00 ...
kernel:[ 2277.345574] Process svxlink (pid: 875, stack limit = 0xb66bc210)

Message from syslogd@DB0CA at Nov 7 22:58:00 ...
kernel:[ 2277.346891] Stack: (0xb66bde80 to 0xb66be000)

Message from syslogd@DB0CA at Nov 7 22:58:00 ...
kernel:[ 2277.348218] de80: b65b3f00 00000006 b66bdef4 b66bde98 8017f230 802725 70 60000113 80b7ad40

Message from syslogd@DB0CA at Nov 7 22:58:00 ...
kernel:[ 2277.350892] dea0: 80c79814 80c0f5c0 b65b36c0 ba360f80 b920e070 80c723 47 80b7f080 80c0312c

Message from syslogd@DB0CA at Nov 7 22:58:00 ...
kernel:[ 2277.353601] dec0: 80c031d0 80c0fb40 b66bdef4 80c020a4 00000009 b66bc0 38 00000100 b9c05180

Message from syslogd@DB0CA at Nov 7 22:58:00 ...
kernel:[ 2277.356309] dee0: 00000024 00000200 b66bdf5c b66bdef8 8010169c 8017ec d8 801011e8 b9c88100

Message from syslogd@DB0CA at Nov 7 22:58:00 ...
kernel:[ 2277.359048] df00: 00000001 00400040 80c02d00 0003043f 00000008 80c7b9 80 80c726d8 80c0312c

Message from syslogd@DB0CA at Nov 7 22:58:00 ...
kernel:[ 2277.361927] df20: 80b793b8 80c02080 b66bdef8 0000000a 80178484 80b7ce c0 00000000 00000000

Message from syslogd@DB0CA at Nov 7 22:58:00 ...
kernel:[ 2277.364944] df40: 00000001 b9c05180 00000000 ffffffff b66bdf6c b66bdf 60 80122cf8 8010151c

Message from syslogd@DB0CA at Nov 7 22:58:00 ...
kernel:[ 2277.368087] df60: b66bdf94 b66bdf70 8017371c 80122bf8 b66bdfb0 80c046 70 40000010 ffffffff

Message from syslogd@DB0CA at Nov 7 22:58:00 ...
kernel:[ 2277.371383] df80: 10c5383d 10c5387d b66bdfac b66bdf98 8010150c 801736 b8 80c040a4 767cdba4

Message from syslogd@DB0CA at Nov 7 22:58:00 ...
kernel:[ 2277.374752] dfa0: 00000000 b66bdfb0 8071aa0c 80101470 015b6c48 7ef8fd 18 00000000 7ef8fe04

Message from syslogd@DB0CA at Nov 7 22:58:00 ...
kernel:[ 2277.378202] dfc0: 00000007 015b6c48 00000003 00000008 00000065 000000 00 ffffffff 00000004

Message from syslogd@DB0CA at Nov 7 22:58:00 ...
kernel:[ 2277.381699] dfe0: 00000008 7ef8fc44 00000009 767cdba4 40000010 ffffff ff e1a00007 0a00008c

----------------------------------------------------------------------------------------------------------
The same happens on an Raspberry Pi Zero, with the Raspberry Pi 2 it doesnt happen.
How can i narrow down the Problem ?!

73 Sebastian

P33M, 2017-11-09:
Can you provide more of the log output? That's half of the OOPS. Start from where "cut here" appears in the log through to "end trace"

dl1esk, 2017-11-09:
i got that Message on my Console, so thats all what i have, because the Raspi doesnt respond anymore when this happens...
The Logfiles are written into an Ramdisk to save my SD-Card...
Maybe i can switch back that Logfiles will be written on Disk that i can provide an complete Oops Log.
Or do u have another solution to narrow down the problem?!

P33M, 2017-11-09:
A serial console attached to the GPIO UART is the preferred method of getting information out when the Pi crashes. Do you have a USB-serial adapter that you can use?

Note that you will need to put `enable_uart=1` in /boot/config.txt.

dl1esk, 2017-11-09:
i got a Log now, something with out of memory...

Nov  9 15:29:38 DB0CA kernel: [ 2054.335622] Unable to handle kernel NULL pointer dereference at virtual address 00000023
Nov  9 15:29:38 DB0CA kernel: [ 2054.339545] pgd = b73b4000
Nov  9 15:29:38 DB0CA kernel: [ 2054.341491] [00000023] *pgd=393a9835, *pte=00000000, *ppte=00000000
Nov  9 15:29:38 DB0CA kernel: [ 2054.343601] Internal error: Oops: 17 [#1] SMP ARM
Nov  9 15:29:38 DB0CA kernel: [ 2054.345660] Modules linked in: cmac bnep hci_uart btbcm bluetooth evdev snd_usb_audio snd_hwdep snd_usbmidi_lib snd_rawmidi snd_seq_device brcmfmac brcmutil cfg80211 rfkill snd_bcm2835 snd_pcm snd_timer snd bcm2835_gpiomem uio_pdrv_genirq uio fixed ip_tables x_tables ipv6
Nov  9 15:29:38 DB0CA kernel: [ 2054.355191] CPU: 0 PID: 339 Comm: python Not tainted 4.9.59-v7+ #1047
Nov  9 15:29:38 DB0CA kernel: [ 2054.357715] Hardware name: BCM2835
Nov  9 15:29:38 DB0CA kernel: [ 2054.360166] task: b9160ec0 task.stack: b92d8000
Nov  9 15:29:38 DB0CA kernel: [ 2054.362739] PC is at eventpoll_release_file+0x2c/0x84
Nov  9 15:29:38 DB0CA kernel: [ 2054.365323] LR is at mutex_lock+0x1c/0x64
Nov  9 15:29:38 DB0CA kernel: [ 2054.367871] pc : [<802bcffc>]    lr : [<80717760>]    psr: a0000013
Nov  9 15:29:38 DB0CA kernel: [ 2054.367871] sp : b92d9ef8  ip : b92d9ee0  fp : b92d9f1c
Nov  9 15:29:38 DB0CA kernel: [ 2054.373117] r10: 00000008  r9 : b6fca9f0  r8 : b9453f94
Nov  9 15:29:38 DB0CA kernel: [ 2054.375720] r7 : b6f913b8  r6 : b9598110  r5 : fffffff3  r4 : b9453f00
Nov  9 15:29:38 DB0CA kernel: [ 2054.378371] r3 : 00000023  r2 : 00000000  r1 : 00000008  r0 : 00000000
Nov  9 15:29:38 DB0CA kernel: [ 2054.381018] Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Nov  9 15:29:38 DB0CA kernel: [ 2054.383761] Control: 10c5383d  Table: 373b406a  DAC: 00000055
Nov  9 15:29:38 DB0CA kernel: [ 2054.386442] Process python (pid: 339, stack limit = 0xb92d8210)
Nov  9 15:29:38 DB0CA kernel: [ 2054.389082] Stack: (0xb92d9ef8 to 0xb92da000)
Nov  9 15:29:38 DB0CA kernel: [ 2054.391738] 9ee0:                                                       b9453f94 b9453f00
Nov  9 15:29:38 DB0CA kernel: [ 2054.396906] 9f00: b6fca9f0 b9598110 b6f913b8 00000000 b92d9f5c b92d9f20 80272764 802bcfdc
Nov  9 15:29:38 DB0CA kernel: [ 2054.402040] 9f20: 00000000 00000000 b9453f00 b9453f08 b92d9f54 b91613a8 80c7bee0 00000000
Nov  9 15:29:38 DB0CA kernel: [ 2054.407153] 9f40: b9160ec0 80108244 b92d8000 00000000 b92d9f6c b92d9f60 8027281c 802725d0
Nov  9 15:29:38 DB0CA kernel: [ 2054.412333] 9f60: b92d9f8c b92d9f70 8013ac34 80272810 b92d8000 b92d8010 80108244 b92d9fb0
Nov  9 15:29:38 DB0CA kernel: [ 2054.417619] 9f80: b92d9fac b92d9f90 8010b838 8013ab74 0098c748 76e38bec 00000000 00000006
Nov  9 15:29:38 DB0CA kernel: [ 2054.420507] 9fa0: 00000000 b92d9fb0 801080e8 8010b778 00000000 00000444 76e387a8 76d65fc4
Nov  9 15:29:38 DB0CA kernel: [ 2054.423500] 9fc0: 0098c748 76e38bec 00000000 00000006 00971050 002ca000 0098c748 76c94330
Nov  9 15:29:38 DB0CA kernel: [ 2054.426642] 9fe0: 00000000 7eae0580 76d67928 76d65fd4 20000010 00000006 00000000 00000000
Nov  9 15:29:38 DB0CA kernel: [ 2054.429863] [<802bcffc>] (eventpoll_release_file) from [<80272764>] (__fput+0x1a0/0x1e8)
Nov  9 15:29:38 DB0CA kernel: [ 2054.433164] [<80272764>] (__fput) from [<8027281c>] (____fput+0x18/0x1c)
Nov  9 15:29:38 DB0CA kernel: [ 2054.434895] [<8027281c>] (____fput) from [<8013ac34>] (task_work_run+0xcc/0xfc)
Nov  9 15:29:38 DB0CA kernel: [ 2054.436669] [<8013ac34>] (task_work_run) from [<8010b838>] (do_work_pending+0xcc/0xd0)
Nov  9 15:29:38 DB0CA kernel: [ 2054.440089] [<8010b838>] (do_work_pending) from [<801080e8>] (slow_work_pending+0xc/0x20)
Nov  9 15:29:38 DB0CA kernel: [ 2054.443519] Code: eb1169d4 e5b83094 e1580003 e2435030 (e5933000) 
Nov  9 15:29:38 DB0CA kernel: [ 2054.447930] ---[ end trace 701f16df15c4ce47 ]---
Nov  9 15:30:15 DB0CA kernel: [ 2091.254505] kthreadd invoked oom-killer: gfp_mask=0x27000c0(GFP_KERNEL_ACCOUNT|__GFP_NOTRACK), nodemask=0, order=1, oom_score_adj=0
Nov  9 15:30:15 DB0CA kernel: [ 2091.262476] kthreadd cpuset=/ mems_allowed=0
Nov  9 15:30:15 DB0CA kernel: [ 2091.266612] CPU: 1 PID: 2 Comm: kthreadd Tainted: G      D         4.9.59-v7+ #1047
Nov  9 15:30:15 DB0CA kernel: [ 2091.275073] Hardware name: BCM2835
Nov  9 15:30:15 DB0CA kernel: [ 2091.279166] [<8010fb3c>] (unwind_backtrace) from [<8010c058>] (show_stack+0x20/0x24)
Nov  9 15:30:15 DB0CA kernel: [ 2091.287194] [<8010c058>] (show_stack) from [<80456764>] (dump_stack+0xd4/0x118)
Nov  9 15:30:15 DB0CA kernel: [ 2091.291151] [<80456764>] (dump_stack) from [<8026d2ac>] (dump_header+0x9c/0x1f4)
Nov  9 15:30:15 DB0CA kernel: [ 2091.298550] [<8026d2ac>] (dump_header) from [<80210b14>] (oom_kill_process+0x3e0/0x4e4)
Nov  9 15:30:15 DB0CA kernel: [ 2091.305772] [<80210b14>] (oom_kill_process) from [<80210f7c>] (out_of_memory+0x124/0x334)
Nov  9 15:30:15 DB0CA kernel: [ 2091.312610] [<80210f7c>] (out_of_memory) from [<80216120>] (__alloc_pages_nodemask+0xd7c/0xe58)
Nov  9 15:30:15 DB0CA kernel: [ 2091.319467] [<80216120>] (__alloc_pages_nodemask) from [<8011ab38>] (copy_process.part.5+0xec/0x17b0)
Nov  9 15:30:15 DB0CA kernel: [ 2091.326260] [<8011ab38>] (copy_process.part.5) from [<8011c38c>] (_do_fork+0xc8/0x408)
Nov  9 15:30:15 DB0CA kernel: [ 2091.332992] [<8011c38c>] (_do_fork) from [<8011c73c>] (kernel_thread+0x40/0x48)
[... cut ...]
Nov  9 15:33:03 DB0CA kernel: [    9.969493] Bluetooth: HCI UART protocol Three-wire (H5) registered
Nov  9 15:33:03 DB0CA kernel: [    9.969692] Bluetooth: HCI UART protocol Broadcom registered
Nov  9 15:33:03 DB0CA kernel: [   10.139203] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
Nov  9 15:33:03 DB0CA kernel: [   10.139210] Bluetooth: BNEP filters: protocol multicast
Nov  9 15:33:03 DB0CA kernel: [   10.139223] Bluetooth: BNEP socket layer initialized
Nov  9 15:33:15 DB0CA kernel: [   22.063744] random: crng init done

P33M, 2017-11-09:
The OOM kill backtrace is unlikely to be related to the OTG speed parameter. Do you have any other settings in config.txt or cmdline.txt that are not standard?

dl1esk, 2017-11-09:
only this one....
on my raspberry Pi2 i could use that Parameter in combination with the svxlink software, there i had some Audio Dropouts sometimes but no out of Memory and Kernel oops...
the rest in my cmdline.txt is standard...
I want to use that because when i use Usb 2.0 i have a much bigger Audio Delay from Soundcard Input to Soundcard Output via the svxlink Software...
With Usb 1.1 i have no Delay...

dl1esk, 2017-11-09:
and this happens only when i use the OTG Speed Parameter

P33M, 2017-11-09:
The only bearing the dwc_otg driver could have on a crash like this would either be
- Memory corruption, writing to out-of-bounds locations
- Subtly altering timing such that a different bug is exposed elsewhere in the kernel

What happens if you add the parameter `dwc_otg.fiq_fsm_enable=0` to cmdline.txt (and reboot)?

dl1esk, 2017-11-09:
with the OTG Speed parameter on  ?

P33M, 2017-11-09:
Yes, with `dwc_otg.speed=`.

dl1esk, 2017-11-09:
Just for u to understand what Svxlink does...
the Mic Input is opened the whole Time...
and the Earphone Output or Speaker Output is just opened when there is a Speech signal on the input, so the output is switched on/off many times while the input is open the whole time

dl1esk, 2017-11-10:
@P33M in this combination it works well.....
do u have an Idea what the problem is ?!
i observed that sometimes when the Output ( earphone/ speaker ) be opened to send audio through that the first 5-10  secs or more dont come through. That just happens sometimes...but will be interesting why that happens because after that time the audio comes through and everything is fine.

P33M, 2017-11-10:
I think I know what might be causing issues in full-speed mode. The frame counter in FS mode increments every 1ms, whereas in high-speed mode it increments every 125uS. There are certain paths in the FIQ and dwc_otg driver that make the assumption we're in high-speed mode when a delay is necessary, those assumptions are not valid *anyway* when in FS mode - the FIQ code is more or less redundant, aside from suppressing unnecessary SOF interrupts.

As a definitive test, does it fail to crash with `dwc_otg.fiq_fsm_mask=0xB`?

dl1esk, 2017-11-10:
That u don`t misunderstand me....
With USB 2.0 ( Sound Card Chip can only 1.1 ) i have a Delay between Input<>svxlink<>output
It`s an amateur radio repeater, the voice what comes over the input is routed directly to the output.
the output pipe will be opened / closed after every conversation participant, then opened agn when the next one is speaking. With USB 1.1(dwc_otg.speed=1 )i have NO delay but sometimes when somebody starts speaking the first 5-10 secs are gone won`t be routed to the output, the output hangs then.
After that time everything resumes to normal operation...
I will try that Feature now and give u the result when i tested it.
tnx for your help...

dl1esk, 2017-11-10:
dwc_otg.fiq_fsm_mask=0xB & dwc_otg.speed=1 is running now since 3 hrs without any  Crash...
what is dwc_otg.fiq_fsm_mask=0xB for ?!

P33M, 2017-11-10:
The FSM mask is a bitfield that controls which types of USB transfers are handled by the FIQ code.

Negating bit 2 means that Isochronous transfers are not performed by the FIQ code - which, as it produces a different result - implies that there is a bug somewhere in that codepath in combination with dwc_otg.speed=1.

What's the reason for using speed=1? Is it just that the delay between opening the USB sound card output and the first sound being played is unacceptable?

dl1esk, 2017-11-10:
Yes the delay is too big sometimes ..,.that when u speak over the Repeater and stop talking, so the radio switches to rx ...then u can hear ur last spoken word >1000ms
that happens not everytime but very often...
and i also have the feeling that since i use the raspi 3...with Usb 2.0 sometimes some words get lost...seems that something get synced to the right time and then one word or some letters are lost....
and with using speed=1 it happens like described that sometimes one speaking passage get lost, just when the signal comes to close audio pipe from the program u can hear 1-2 letters from that passage...dunno why but no audio delay...

dl1esk, 2017-11-14:
@P33M any Ideas?

P33M, 2017-11-14:
Your initial issue describes a crash with dwc_otg.speed=1. Changing a setting makes the crash go away (so there is a bug in the FIQ code being suppressed) but aside from the general lack of network performance, what else is the issue with dwc_otg.speed=1?

dl1esk, 2017-11-14:
That sometimes Audio got lost for one troughput..
pipe open no audio..pipe closed u get a few ms of the spoken Audio...
That happens not everytime but quite a lot...
it seems that this happens when u open/close the pipe quickly

aplocher, 2018-04-18:
Hello, I've had this Pi3 setup with Octoprint and MJPG-Streamer to handle 3d printing and cam streaming to monitor it remotely for many months.  It was pretty out-of-date so last night I did a `apt-get update` / `apt-get dist-upgrade` and everything was seemingly working fine until about an hour ago I received a similar error with that "Internal error: Oops: 17 [#1] SMP ARM" message in my remote SSH terminal:

```
Message from syslogd@BC-REPRAP-01 at Apr 18 04:46:09 ...
 kernel:[13872.033349] Internal error: Oops: 17 [#1] SMP ARM

Message from syslogd@BC-REPRAP-01 at Apr 18 04:46:09 ...
 kernel:[13872.033723] Process mjpg_streamer (pid: 14663, stack limit = 0x9eaca210)

Message from syslogd@BC-REPRAP-01 at Apr 18 04:46:09 ...
 kernel:[13872.033735] Stack: (0x9eacbd78 to 0x9eacc000)

Message from syslogd@BC-REPRAP-01 at Apr 18 04:46:09 ...
 kernel:[13872.033747] bd60:                                                       000005b4 806d1930

Message from syslogd@BC-REPRAP-01 at Apr 18 04:46:09 ...
 kernel:[13872.033770] bd80: 000206fc 9eacbe88 a66db840 000206fc 9eacbe34 9eacbda0 806d1950 806ce34c

Message from syslogd@BC-REPRAP-01 at Apr 18 04:46:09 ...
 kernel:[13872.033787] bda0: 000005b4 9eacbe6c 9eacbdcc 9eacbdb8 80c069c4 80d0a08c 80c049c0 9eaca020

Message from syslogd@BC-REPRAP-01 at Apr 18 04:46:09 ...
 kernel:[13872.033804] bdc0: 00000000 9eacbdd0 00000000 00000000 00000001 00000000 9eacbe04 00000200

Message from syslogd@BC-REPRAP-01 at Apr 18 04:46:09 ...
 kernel:[13872.033820] bde0: 9eacbe90 a66db990 000005b4 00000000 00000000 000005b4 7fffffff 9eacbe08

Message from syslogd@BC-REPRAP-01 at Apr 18 04:46:09 ...
 kernel:[13872.033837] be00: 801235c4 801e0000 a66db840 a66db840 0003e764 9eacbe88 00000000 9eacbf08

Message from syslogd@BC-REPRAP-01 at Apr 18 04:46:09 ...
 kernel:[13872.033853] be20: a6517600 00000000 9eacbe54 9eacbe38 806d247c 806d1654 806d2444 a66db840

Message from syslogd@BC-REPRAP-01 at Apr 18 04:46:09 ...
 kernel:[13872.033870] be40: 00000000 00000000 9eacbe74 9eacbe58 806fe570 806d2450 00000001 9eacbe84

Message from syslogd@BC-REPRAP-01 at Apr 18 04:46:09 ...
 kernel:[13872.033889] be60: 9eacbef0 00000000 9eacbe84 9eacbe78 80657c84 806fe534 9eacbedc 9eacbe88

Message from syslogd@BC-REPRAP-01 at Apr 18 04:46:09 ...
 kernel:[13872.033906] be80: 80657d2c 80657c6c 00000000 00000000 00000001 000206fc 0001e068 9eacbee8

Message from syslogd@BC-REPRAP-01 at Apr 18 04:46:09 ...
 kernel:[13872.033922] bea0: 00000001 00400040 00000000 00000000 00000000 9eacbf08 a6517600 9eacbf80

Message from syslogd@BC-REPRAP-01 at Apr 18 04:46:09 ...
 kernel:[13872.033939] bec0: 00000000 00000000 0003e764 9eaca000 9eacbf4c 9eacbee0 8028a930 80657ca0

Message from syslogd@BC-REPRAP-01 at Apr 18 04:46:09 ...
 kernel:[13872.033955] bee0: 0003e764 00000006 75002008 0003e764 00000001 00000000 0003e764 9eacbee8

Message from syslogd@BC-REPRAP-01 at Apr 18 04:46:09 ...
 kernel:[13872.033971] bf00: 00000001 00400040 a6517600 00000000 00000000 00000000 00000000 00000000

Message from syslogd@BC-REPRAP-01 at Apr 18 04:46:09 ...
 kernel:[13872.033987] bf20: 00000000 00000000 a6517600 0003e764 a6517600 75002008 9eacbf80 80108224

Message from syslogd@BC-REPRAP-01 at Apr 18 04:46:09 ...
 kernel:[13872.034007] bf40: 9eacbf7c 9eacbf50 8028ab50 8028a85c 802aa234 802a9a10 a6517600 a6517601

Message from syslogd@BC-REPRAP-01 at Apr 18 04:46:09 ...
 kernel:[13872.034024] bf60: 0003e764 75002008 80108224 9eaca000 9eacbfa4 9eacbf80 8028ada0 8028aaa8

Message from syslogd@BC-REPRAP-01 at Apr 18 04:46:09 ...
 kernel:[13872.034040] bf80: 00000000 00000000 00023120 0003e764 74efe0e8 00000004 00000000 9eacbfa8

Message from syslogd@BC-REPRAP-01 at Apr 18 04:46:09 ...
 kernel:[13872.034057] bfa0: 80108080 8028ad58 00023120 0003e764 00000007 75002008 0003e764 00000000

Message from syslogd@BC-REPRAP-01 at Apr 18 04:46:09 ...
[... 121 lines cut ...]
Apr 18 04:46:09 BC-REPRAP-01 kernel: [13872.034007] bf40: 9eacbf7c 9eacbf50 8028ab50 8028a85c 802aa234 802a9a10 a6517600 a6517601
Apr 18 04:46:09 BC-REPRAP-01 kernel: [13872.034024] bf60: 0003e764 75002008 80108224 9eaca000 9eacbfa4 9eacbf80 8028ada0 8028aaa8
Apr 18 04:46:09 BC-REPRAP-01 kernel: [13872.034040] bf80: 00000000 00000000 00023120 0003e764 74efe0e8 00000004 00000000 9eacbfa8
Apr 18 04:46:09 BC-REPRAP-01 kernel: [13872.034057] bfa0: 80108080 8028ad58 00023120 0003e764 00000007 75002008 0003e764 00000000
Apr 18 04:46:09 BC-REPRAP-01 kernel: [13872.034074] bfc0: 00023120 0003e764 74efe0e8 00000004 74efe530 76f0a24c 00000138 00040f57
Apr 18 04:46:09 BC-REPRAP-01 kernel: [13872.034090] bfe0: 00000000 74efe0b8 00000000 76ed9314 80000010 00000007 00000000 00000000
Apr 18 04:46:09 BC-REPRAP-01 kernel: [13872.034119] [<806ce384>] (tcp_push) from [<806d1950>] (tcp_sendmsg_locked+0x308/0xdfc)
Apr 18 04:46:09 BC-REPRAP-01 kernel: [13872.034144] [<806d1950>] (tcp_sendmsg_locked) from [<806d247c>] (tcp_sendmsg+0x38/0x4c)
Apr 18 04:46:09 BC-REPRAP-01 kernel: [13872.034167] [<806d247c>] (tcp_sendmsg) from [<806fe570>] (inet_sendmsg+0x48/0xe4)
Apr 18 04:46:09 BC-REPRAP-01 kernel: [13872.034190] [<806fe570>] (inet_sendmsg) from [<80657c84>] (sock_sendmsg+0x24/0x34)
Apr 18 04:46:09 BC-REPRAP-01 kernel: [13872.034211] [<80657c84>] (sock_sendmsg) from [<80657d2c>] (sock_write_iter+0x98/0xcc)
Apr 18 04:46:09 BC-REPRAP-01 kernel: [13872.034233] [<80657d2c>] (sock_write_iter) from [<8028a930>] (__vfs_write+0xe0/0x138)
[... cut ...]
Apr 18 04:46:09 BC-REPRAP-01 kernel: [13872.034320] Code: e155000c 03a05000 e2117902 1a000039 (e5d50028)
Apr 18 04:46:09 BC-REPRAP-01 kernel: [13872.034335] ---[ end trace 4fafc113ac0136da ]---

```

I don't seem to have dwc_otg.speed=1

When I did the apt-get, I believe I saw it say it was replacing the January rpi kernel (201801??) with March's (201803??) if that helps.  I don't recall if I was on a 4.9 kernel previously, but I'm guessing I was (if that was the kernel in January).  Whatever is causing it seems to be related to the update I did.

Thanks

pelwell, 2018-04-18:
We have today pushed an update to the Raspbian kernel package that includes a fix for this upstream TCP bug, so if you update and upgrade again you should no longer get this Oops.

aplocher, 2018-04-18:
You guys are on it!  You rock thanks.  Updating now and will report back if I see this again.

JamesH65, 2018-06-27:
@aplocher Anything to report? If now ok, please close the issue.

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

aplocher, 2018-06-27:
Sorry about that, the update fixed everything.  As stated previously, you guys rock 👍 

But I'm not the originator of this issue, so I can't close it.

JamesH65, 2018-06-27:
I can!! Bwahahahahahahah etc

closed 2018-06-27 by JamesH65