raspberrypi/linux #2261: Relationship between timer frequency (HZ_*) and usbhid.jspoll kernel parameter?

vanfanel, 2017-11-08:
Hi,

The raspberry pi "official" kernel comes with HZ_100 configured, but at the same time it's possible to pass usbhid.jspoll=2 on cmdline.txt.
Is there a relationship between the kernel timer frequency (HZ_* configuration item, ranging from 100Hz to 1000Hz) and the usbhid.jspoll kernel parameter?
The usbhid.jspoll kernel parameter is a recent parameter that allows to read from USB HID devices at a given rate. Passing usbhid.jspoll=2 causes the kernel to poll USB HID devices at 500Hz.
However, it seems to me that, in order to be able to poll at 500Hz, the HZ_* config option should be HZ_500 as minimum. Am I right on this?

pelwell, 2017-11-08:
If something wants to poll faster than the standard tick rate it requests a timer callback, which can usually be specified to microsecond resolution (although there is no guarantee of microsecond accuracy).

vanfanel, 2017-11-08:
@pelwell: so I can set, let's say, HZ_100, pass usbhid.jspoll=2 and the USB HID device will still be able to poll at 500Hz anyway using that timer callback??

pelwell, 2017-11-08:
I haven't looked at that specific parameter, but I think you can pretty much ignore the HZ setting these days, except perhaps for time-slicing between compute-bound threads.

pelwell, 2017-11-08:
See https://elinux.org/High_Resolution_Timers - Linux has run on hrtimers for years now.

vanfanel, 2017-11-09:
@pelwell 
Thanks a lot for the responses and the link, now I think I understand.. :+1:

closed 2017-11-09 by vanfanel