raspberrypi/linux #2262: No sound thru headphones jack with raspbian/stretch

echevresm, 2017-11-11:
Raspi-config and others tricks doesn't succeed to change the playback route, instead they set the main volume. I think that's because they don't send the right parameters to amixer. Instead of using "amixer cset numid=3 1" they should use "amixer -c 0 cset numid=3 1".

pelwell, 2017-11-14:
The normal Raspbian audio routing is working as expect on my installation. Can you give a bit more information about your setup, especially any additional audio hardware and ALSA configuration files you have? The output of `aplay --list-devices` would be helpful.

echevresm, 2017-11-14:
I have not a particular installation, just a Pi 3 board with anything else.
With Raspbian Jessie, the audio routing works as expected but with Stretch, the new Raspbian release, the audio routing doesn't work, the audio is directed through hdmi with the same installation and without a standard possibility to change this routing. The raspi-config script once modified with 
amixer -c 0 cset numid=3 "$AUDIO_OUT"
works as expected.
The output of aplay --list-devices is (sorry, it's in french): 
**** Liste des Périphériques Matériels PLAYBACK ****
carte 0: ALSA [bcm2835 ALSA], périphérique 0: bcm2835 ALSA [bcm2835 ALSA]
  Sous-périphériques: 8/8
  Sous-périphérique #0: subdevice #0
  Sous-périphérique #1: subdevice #1
  Sous-périphérique #2: subdevice #2
  Sous-périphérique #3: subdevice #3
  Sous-périphérique #4: subdevice #4
  Sous-périphérique #5: subdevice #5
  Sous-périphérique #6: subdevice #6
  Sous-périphérique #7: subdevice #7
carte 0: ALSA [bcm2835 ALSA], périphérique 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Sous-périphériques: 1/1
  Sous-périphérique #0: subdevice #0

pelwell, 2017-11-16:
Any suggestions, @spl237 ?

spl237, 2017-11-16:
These amixer commands work to switch the audio output for me under Stretch, without the need to add the -c 0 arguments.

Could you please post the contents of /home/pi/~.asoundrc, if that file exists on your system? Many thanks.

echevresm, 2017-11-16:
I have not a ~.asound file in my system, but I tried a new Stretch installation out of the box and it worked as expected. So, I am trying to search what did I do to get the issue. At the begining of my first Strech installation I was connected to a TV receiver with audio throught HDMI, and so I became aware of the issue much later.

The timidity release of Strech is a pretty old one. I had to install the last release from source. It didn't work well. I tried others music softwares coming with the distro and they didn't work as well, except Sonic Pi. As I wrote all what I did, I'm going to do the same operations step by step in order to see what's wrong.

echevresm, 2017-11-18:
I confirm that's just the compilation of timidity 2.14 which is the origin of the problem. First, I had to install libjack-jackd2-dev, libasound2-dev, libghc-gtk-dev and then : 
./configure --enable-audio=alsa,jack --enable-interface=gtk,alsaseq
make
sudo make install
And after that, timidity didn't work as well !
I did this compilation under Mint 18.1 without any problem.

closed 2017-11-18 by echevresm