raspberrypi/linux #2264: Bluetooth hci0 kernel issue

jswim788, 2017-11-13:
I run a python based Bluetooth scanner script to ping phones.  With this running, every few weeks I get a message like this in the syslog and the RPi gets sort of hung (can't access it over the network and end up needing to power cycle it).

Nov 11 04:53:14 raspberrypi kernel: [3679030.821581] Bluetooth: hci0: Frame reassembly failed (-84)

The uname -a is: Linux raspberrypi 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux

Any idea what I can do about this to debug further?  Moving to Stretch is possible, but not trivial.  Is there a good reason to believe it is fixed in Stretch?  I've seen some posts suggesting disabling the internal Bluetooth and going with an external Bluetooth dongle.  I shouldn't need to do that, but is it a good workaround?

thanks for any advice.

pelwell, 2017-11-13:
Please update to kernel 4.9.56 or later (`sudo rpi-update` should be safe at the moment), because this includes fixes for two issues with the UART driver that could cause data loss. I've found Bluetooth to be much more reliable with these patches.

jswim788, 2017-11-15:
That failed much faster than before - died in less than 2 hours.  Here's some syslog information.  Would anything else be helpful?

Linux raspberrypi 4.9.61-v7+ #1049 SMP Fri Nov 10 15:32:51 GMT 2017 armv7l GNU/Linux

Nov 13 22:26:12 raspberrypi kernel: [ 5986.500274] Bluetooth: hci0: Frame reassembly failed (-84)
Nov 13 22:26:22 raspberrypi kernel: [ 5996.547341] ------------[ cut here ]------------
Nov 13 22:26:22 raspberrypi kernel: [ 5996.547371] WARNING: CPU: 1 PID: 3482 at fs/sysfs/dir.c:31 sysfs_warn_dup+0x78/0x88
Nov 13 22:26:22 raspberrypi kernel: [ 5996.547377] sysfs: cannot create duplicate filename '/devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:11'
Nov 13 22:26:22 raspberrypi kernel: [ 5996.547381] Modules linked in: bnep hci_uart btbcm bluetooth cfg80211 rfkill cdc_acm snd_usb_audio snd_hwdep snd_usbmidi_lib evdev snd_rawmidi snd_seq_device snd_bcm2835 snd_pcm snd_timer snd bcm2835_gpiomem uio_pdrv_genirq fixed uio i2c_dev fuse ipv6
Nov 13 22:26:22 raspberrypi kernel: [ 5996.547483] CPU: 1 PID: 3482 Comm: kworker/u9:0 Not tainted 4.9.61-v7+ #1049
Nov 13 22:26:22 raspberrypi kernel: [ 5996.547487] Hardware name: BCM2835
Nov 13 22:26:22 raspberrypi kernel: [ 5996.547643] Workqueue: hci0 hci_rx_work [bluetooth]
Nov 13 22:26:22 raspberrypi kernel: [ 5996.547673] [<8010fb3c>] (unwind_backtrace) from [<8010c058>] (show_stack+0x20/0x24)
Nov 13 22:26:22 raspberrypi kernel: [ 5996.547690] [<8010c058>] (show_stack) from [<80457664>] (dump_stack+0xd4/0x118)
Nov 13 22:26:22 raspberrypi kernel: [ 5996.547707] [<80457664>] (dump_stack) from [<8011d348>] (__warn+0xf8/0x110)
Nov 13 22:26:22 raspberrypi kernel: [ 5996.547719] [<8011d348>] (__warn) from [<8011d3a8>] (warn_slowpath_fmt+0x48/0x50)
Nov 13 22:26:22 raspberrypi kernel: [ 5996.547736] [<8011d3a8>] (warn_slowpath_fmt) from [<802eda64>] (sysfs_warn_dup+0x78/0x88)
Nov 13 22:26:22 raspberrypi kernel: [ 5996.547755] [<802eda64>] (sysfs_warn_dup) from [<802edb60>] (sysfs_create_dir_ns+0x94/0xa4)
Nov 13 22:26:22 raspberrypi kernel: [ 5996.547770] [<802edb60>] (sysfs_create_dir_ns) from [<8045a110>] (kobject_add_internal+0xb4/0x30c)
Nov 13 22:26:22 raspberrypi kernel: [ 5996.547783] [<8045a110>] (kobject_add_internal) from [<8045a540>] (kobject_add+0x50/0x94)
Nov 13 22:26:22 raspberrypi kernel: [ 5996.547799] [<8045a540>] (kobject_add) from [<804f8f68>] (device_add+0xe8/0x57c)
Nov 13 22:26:22 raspberrypi kernel: [ 5996.547946] [<804f8f68>] (device_add) from [<7f27425c>] (hci_conn_add_sysfs+0x3c/0x60 [bluetooth])
Nov 13 22:26:22 raspberrypi kernel: [ 5996.548243] [<7f27425c>] (hci_conn_add_sysfs [bluetooth]) from [<7f262e94>] (hci_event_packet+0xc54/0x2d14 [bluetooth])
Nov 13 22:26:22 raspberrypi kernel: [ 5996.548523] [<7f262e94>] (hci_event_packet [bluetooth]) from [<7f255920>] (hci_rx_work+0x170/0x240 [bluetooth])
Nov 13 22:26:22 raspberrypi kernel: [ 5996.548671] [<7f255920>] (hci_rx_work [bluetooth]) from [<80136788>] (process_one_work+0x158/0x454)
Nov 13 22:26:22 raspberrypi kernel: [ 5996.548689] [<80136788>] (process_one_work) from [<80136ae8>] (worker_thread+0x64/0x5b8)
Nov 13 22:26:22 raspberrypi kernel: [ 5996.548705] [<80136ae8>] (worker_thread) from [<8013c9cc>] (kthread+0x10c/0x124)
Nov 13 22:26:22 raspberrypi kernel: [ 5996.548721] [<8013c9cc>] (kthread) from [<80108148>] (ret_from_fork+0x14/0x2c)
Nov 13 22:26:22 raspberrypi kernel: [ 5996.548728] ---[ end trace f99518072c5fcd92 ]---
Nov 13 22:26:22 raspberrypi kernel: [ 5996.548736] ------------[ cut here ]------------
Nov 13 22:26:22 raspberrypi kernel: [ 5996.548746] WARNING: CPU: 1 PID: 3482 at lib/kobject.c:240 kobject_add_internal+0x294/0x30c
Nov 13 22:26:22 raspberrypi kernel: [ 5996.548752] kobject_add_internal failed for hci0:11 with -EEXIST, don't try to register things with the same name in the same directory.

pelwell, 2017-11-15:
I don't think I've seen that particular pattern of errors before. Which script are you using?

jswim788, 2017-11-15:
I'm using the Bluetooth presence scanner located here: https://github.com/daemondazz/vera-presence-scanner

pelwell, 2017-11-16:
I'm running a cut-down, standalone version of the scanner, and I can see similar errors (although so far just the reassembly failures). However, I only see them on Pi3B rev 1.2.

The rev 1.2 Pi3B lacks the flow control signals to the Bluetooth modem (we ran out of pins), but the rev 1.3 board drops the BT PCM interface and hooks up the flow control. I think what you are seeing is occasional data loss when the FIFOs overflow, something which is hard to avoid in all circumstances. You may be able to lessen the problem by reducing the baudrate on the modem - try editing /usr/bin/btuart, replacing the 921600 with 460800 and rebooting.

jswim788, 2017-11-16:
I do have the rev 1.2 Pi3B, revision is a02082.  I will try the lower baud rate.  However, this takes weeks to months to crash, so it will be some time before I can say it helped.

Would the problem be solved by switching to an external USB based Bluetooth adapter?

thanks for you help

pelwell, 2017-11-16:
It probably would. The USB interface is likely to be more reliable than the UART with no flow control, and having a dedicated aerial for Bluetooth is usually a good thing.

JamesH65, 2017-12-05:
Alternative to a USB BT adaptor would be a revision 1.3 Pi which has the improved flow control.

@jswim788 Are you happy to close this?

jswim788, 2017-12-08:
I've tried the USB BT with limited success.  I haven't observed the problem
(but that might takes a long time to recur), yet functionally it isn't as
good as the built-in BT - the range is weaker.  I will work on trying
another and/or switching to the newer 1.3 Pi.  As far as this bug is
concerned I think it is okay to close.  Thanks for your help.

dingausmwald, 2018-01-15:
Where to find this new Pi revision? ZeroW is still on 1.1 and Zero has no bt so no alternative to bt dongle?!

JamesH65, 2018-01-15:
PiZero 1.1 does have the HW flow control, for the Pi3 you need the 1.3 revision to have HW flow control.

pelwell, 2018-01-15:
Yes, Pi ZeroW (what James meant) has HW flow control to the modem, but only rev 1.3 of the Pi 3 has it.

JamesH65, 2018-01-16:
Doh!

stschake, 2018-03-28:
Did the revision 1.3 changes make it into the RPi 3B+? From the device tree, it doesn't look like that one has HW flow control either.

pelwell, 2018-03-28:
Yes, the 3B+ has flow control - the DTB is currently patched by the firmware, but now the 3B+ has a dedicated .dts there's no reason not to make it explicit.

JamesH65, 2018-06-27:
Closing this issue as questions answered/issue resolved.

closed 2018-06-27 by JamesH65

HariGopalG, 2018-12-20:
JamesH65, shall we get any information on root cause for this issue?

JamesH65, 2018-12-20:
On original Pi boards with BT, the BT chip was connected to a uart that had no HW flow control. The latest versions change to a UART with HW flow control. That in combination with the fixing of a bug in the UART driver itself has made the BT much more reliable.

HariGopalG, 2018-12-22:
In which version of kernel the UART driver issue has got resolved?