raspberrypi/linux #2268: No rbd module

Pea13, 2017-11-14:
Hi,

While playing with ceph on Raspian Stretch, i noticed that rbd's module is not compiled.
% zgrep RBD /proc/config.gz 
CONFIG_BLK_DEV_DRBD=m
# CONFIG_DRBD_FAULT_INJECTION is not set
# CONFIG_BLK_DEV_RBD is not set

% uname -a
Linux ceph01 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux

Is there a reason to not enable it ? Is that possible to change this for the future releases ?
Regards,

JamesH65, 2017-12-05:
Do you have any figures for the impact on kernel performance and size? It also bring in CEPH_LIB as well. They can brought in as modules, so perhaps its not that critical, but storage requirements would be good to know before adding.

Pea13, 2017-12-05:
No, i have no ideas about the impact on size and performance. I was just wondering why the rbd module wasn't built by default. There are no known issues ?

shreddedbacon, 2018-02-04:
This would be very handy to have rbd.

JamesH65, 2018-06-27:
Do you have any figures for impact on kernel size and performance? Given a limited resources device like the Pi we like to ensure that kernel changes for smaller use cases do not have a detrimental impact.

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

closed 2018-08-31 by JamesH65