raspberrypi/linux #2271: Bluetooth Keyboard stopped working after upgrading to latest kernel.

daniejstriata, 2017-11-16:
I have a Samsung Bluetooth Keyboard (VG-KBD1000) that works fine on an older Jessie install of Raspbian but stopped working when I upgraded to Stretch. The error is that the connection to the device timed out as can be seen in the bluetooth service output.

I connect and pair the Samsung device successfully as I would running Jessie but the device itself does not work. I get an error in Blueman when I try to force a connection.

[image]

[image]
Currently using Kernel: 4.9.61-v7+

output 1:
pi@raspberrypi:~ $ sudo systemctl status bluetooth
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2017-11-16 10:42:17 SAST; 25min ago
     Docs: man:bluetoothd(8)
 Main PID: 2444 (bluetoothd)
   Status: "Running"
   CGroup: /system.slice/bluetooth.service
           └─2444 /usr/lib/bluetooth/bluetoothd

Nov 16 10:44:48 raspberrypi bluetoothd[2444]: connect error: Host is down (112)
Nov 16 10:48:08 raspberrypi bluetoothd[2444]: Can't get HIDP connection info
Nov 16 10:48:14 raspberrypi bluetoothd[2444]: connect error: Host is down (112)
Nov 16 10:49:17 raspberrypi bluetoothd[2444]: Can't get HIDP connection info
Nov 16 10:49:38 raspberrypi bluetoothd[2444]: D0:66:7B:53:D7:87: error updating services: **Connection timed out (110)**
Nov 16 10:50:04 raspberrypi bluetoothd[2444]: Can't get HIDP connection info
Nov 16 10:50:09 raspberrypi bluetoothd[2444]: connect error: Host is down (112)
Nov 16 10:52:02 raspberrypi bluetoothd[2444]: Can't get HIDP connection info
Nov 16 10:53:50 raspberrypi bluetoothd[2444]: Can't get HIDP connection info
Nov 16 10:53:55 raspberrypi bluetoothd[2444]: connect error: Host is down (112)

Output 2:
$ dmesg | grep Bluetooth
[    5.498167] Bluetooth: Core ver 2.22
[    5.498299] Bluetooth: HCI device and connection manager initialized
[    5.498329] Bluetooth: HCI socket layer initialized
[    5.498348] Bluetooth: L2CAP socket layer initialized
[    5.498405] Bluetooth: SCO socket layer initialized
[   93.750485] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   93.750498] Bluetooth: BNEP filters: protocol multicast
[   93.750520] Bluetooth: BNEP socket layer initialized
[   93.860295] Bluetooth: RFCOMM TTY layer initialized
[   93.860330] Bluetooth: RFCOMM socket layer initialized
[   93.860369] Bluetooth: RFCOMM ver 1.11
[  213.476651] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[  213.476691] Bluetooth: HIDP socket layer initialized

pelwell, 2017-11-16:
Raspbian Stretch does not use Blueman, so I think your upgrade has left you with a strange mixture of software. If you have a spare microSD card, test your keyboard with a clean image from our [Downloads page](https://www.raspberrypi.org/downloads/raspbian/).

daniejstriata, 2017-11-16:
I installed Blueman to test with. I'll remove Blueman but is also willing to help identify the Bluetooth issue with the current kernel. Seems to be related to 4.9.x as I tried a couple. Going to 4.14 fixed it.

pelwell, 2017-11-16:
There should be very few Pi-specific differences between the 4.9 and 4.14 kernels, but it is possible that something in the kernel Bluetooth support has changed.

Our BRANCH=next kernel (`sudo BRANCH=next rpi-update`) has just moved to 4.14 (perhaps that's where you got yours), so we are looking to update over the next few months and we appreciate any testing that users can do for us. If 4.14 no longer exhibits the problem then the solution is going to be to upgrade - we are unlikely to spend much time diagnosing the problem but would back-port a fix if one was identified.

daniejstriata, 2017-11-16:
I also believe is it specific to the Samsung Keyboard I'm using otherwise everyone with a bluetooth Keyboard would've had this problem. Hopefully whatever got added with the 4.9.x kernels do not get added in later releases of 4.14.x

If you can give me the specifics to help pinpoint and identify the issue with 4.9.x I will gladly assist otherwise this can be closed as I found a work-around.

pelwell, 2017-11-16:
If you are happy with the workaround then I would prefer not to put further effort into the 4.9 kernel.

daniejstriata, 2017-11-16:
Happy here! Thanks

closed 2017-11-16 by daniejstriata