raspberrypi/linux #2274: Wlan0 interface missing

vanfanel, 2017-11-17:
Hi,

It seems that, with latest kernels/firmwares, I am missing the wlan0 interface after doing:
`sudo modprobe brcmfmac
`
It used to work well, but now I got in dmesg:

```
pi@raspberrypi:~/src/eduke32 $ dmesg
[  373.191993] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[  373.204503] usbcore: registered new interface driver brcmfmac
[  373.210099] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.bin failed with error -2
[  374.269564] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[  375.279560] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50

```

So I went and downloaded brcmfmac43430-sdio.bin to /lib/firmware/brcm/brcmfmac43430-sdio.bin.
Now, dmesg won't have any complains from modprobe, but wlan0 is still absent.
Any ideas on what is going on here, please?

pelwell, 2017-11-17:
1. Which hardware platform is this?
2. What are the kernel and firmware versions (`vcgencmd version`)?
3. Which OS/distribution are you running?
4. Have you always had to use `sudo modprobe brcmfmac`? It ought to be loaded automatically - do you know why it isn't for you?
5. Was the firmware absent before you copied it there? Do you have any idea why?

vanfanel, 2017-11-17:
@pelwell : I have just fixed it. I was missing  /lib/firmware/brcm/brcmfmac43430-sdio.txt too, but modprobe will silently load the module anyway and the interface was still missing. Bad thing to do, if you ask me.
Anyhow, getting /lib/firmware/brcm/brcmfmac43430-sdio.txt fixed it.

Anyhow, for future reference (and in case you can include the firmware by default):

1. I am on a Raspberry Pi 3

2. vcgencmd version says:
```
pi@raspberrypi:~ $ vcgencmd version
Nov  3 2017 17:00:25 
Copyright (c) 2012 Broadcom
version 1bcf9152e3b2ab5c67dcb9c7c1f6d9256d9a6f38 (clean) (release)
```
3. It's Raspbian LITE without systemd. I manually load/unload the modules I need for networking.

4. Yes, I always use a personal script to connect to WiFi networks manually.

5. The firmware was absent, yes: in fact, the /lib/firmware/bcrm directory didn't exist before I created it. I don't know why.

pelwell, 2017-11-17:
> It's Raspbian LITE without systemd.

Who creates this distribution? Raspbian (including Lite) uses systemd on Jessie and Stretch.

> The firmware was absent, yes: in fact, the /lib/firmware/bcrm directory didn't exist before I created it. I don't know why.

/lib/firmware/brcm is present in a standard Raspbian Jessie/Stretch Lite image, so my theory is that whatever removed systemd also removed the firmware.

vanfanel, 2017-11-17:
@pelwell : I removed most of systemd stuff, it's simply Raspbian LITE, not another specific distro.

pelwell, 2017-11-17:
If you can show me a Raspbian Lite image less than a year old without the /lib/firmware/brcm directory then feel free to re-open this issue.

closed 2017-11-17 by pelwell

vanfanel, 2017-11-17:
It may have been my fault, since it's a modified system, so don't worry. And sorry for making you wasting time on this. Thanks :)

mimmoLinino, 2018-09-06:
I have a similar issue using a CM3 connected to a wifi module WSDSB-104GNI(BT).
At startup I see on dmesg the following messages:

> [ 1382.362690] mmc1: queuing unknown CIS tuple 0x80 (2 bytes)
> [ 1382.365644] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
> [ 1382.368583] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
> [ 1382.373856] mmc1: queuing unknown CIS tuple 0x80 (7 bytes)
> [ 1382.543237] mmc1: new high speed SDIO card at address 0001
> [ 1382.612847] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
> [ 1382.619101] brcmfmac: brcmf_fw_map_chip_to_name: using brcm/brcmfmac43430-sdio.bin for chip 0x00a9a6(43430) rev 0x000001
> [ 1382.619326] usbcore: registered new interface driver brcmfmac
> [ 1383.755363] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
> [ 1384.825463] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50

the wlan0 interface is missing. oin order to enable it I need to remove and then insert the brcmfmac module using the commands:
```
modprobe -r brcmfmac
modprobe brcmfmac
```
After that I see the folling messaged on dmesg:

> [ 1531.373109] usbcore: deregistering interface driver brcmfmac
> [ 1538.204465] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
> [ 1538.211679] brcmfmac: brcmf_fw_map_chip_to_name: using brcm/brcmfmac43430-sdio.bin for chip 0x00a9a6(43430) rev 0x000001
> [ 1538.211937] usbcore: registered new interface driver brcmfmac
> [ 1539.389533] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Oct 23 2017 03:55:53 version 7.45.98.38 (r674442 CY) FWID 01-e58d219f
> [ 1539.390198] brcmfmac: brcmf_c_preinit_dcmds: CLM version = API: 12.2 Data: 7.11.15 Compiler: 1.24.2 ClmImport: 1.24.1 Creation: 2014-05-26 10:53:55 Inc Data: 9.10.39 Inc Compiler: 1.29.4 Inc ClmImport: 1.36.3 Creation: 2017-10-23 03:47:14 
> [ 1539.695418] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
> [ 1539.695578] brcmfmac: power management disabled
> [ 1540.626286] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready

How can I solve this issue?

zmrow, 2018-12-03:
I'm also seeing the same behavior as @mimmoLinino on a RPi 3B.