raspberrypi/linux #2278: Kernel 4.14 - Enable  HID_ASUS [=m] in defconfigs

paul-1, 2017-11-18:
Asus branded keyboard (Including USB keyboards) have been moved from hid-chicony to hid-asus

I've enabled the module locally and compiled kernel, keyboard works fine with the new module.

JamesH65, 2017-12-04:
@paul-1 What are you asking for here?

JamesH65, 2017-12-04:
@popcornmix @pelwell Any objections?

pelwell, 2017-12-04:
It's a shame, because the Asus module is over twice the size of the Chicony equivalent, but we don't want to lose support we used to have.

Done.

paul-1, 2017-12-04:
Agreed on the bloat comment, since the asus module includes all the code of the gaming keyboards, which I doubt anyone is using on the rpi.    My keyboard in question was just a simple standard 102 key board that happened to come with a Asus eebox. 

Thanks.

closed 2017-12-05 by paul-1