raspberrypi/linux #2280: Add option to disable NIC LEDs

Rayn0r, 2017-11-19:
Is it possible to adapt a patch found [here](http://blog.everpi.net/2013/11/patch-para-controlar-os-leds-ethernet-do-raspberrypi.html ) so that the NIC's LEDs can be turned off as well?

An updated version of this patch can be found [here](https://www.anginf.de/?p=433)

popcornmix, 2017-11-19:
This can be done without kernel changes. See:
https://www.raspberrypi.org/forums/viewtopic.php?t=72070

Rayn0r, 2017-11-19:
@popcornmix 
Thank you very much for the hint.

I ended up using this piece code: https://mockmoon-cybernetics.ch/git/gitlist/index.php/lan951x-led-ctl.git/
I only had to edit the Makefile so that linking worked:

```
--- Makefile.org        2017-11-19 20:38:24.793069225 +0100
+++ Makefile    2017-11-19 20:38:37.283042876 +0100
@@ -38,5 +38,5 @@
 # Specific programs
 lan951x-led-ctl:       src/lan951x-led-ctl.o
        @$(ECHO) "\t==> Linking objects to output file $@"
-       @$(CC) $(CFLAGS) $(LDFLAGS) $+ -o $@
+       @$(CC) $+ $(CFLAGS) $(LDFLAGS) -o $@
        @$(STRIP) $@
```
Running `./lan951x-led-ctl --fdx=0 --lnk=0 --spd=0` did the trick and turned off the LEDs on my RPI3.

closed 2017-11-19 by Rayn0r