raspberrypi/linux #2283: /proc/cpu/alignment is absent

Fullmetal8ender, 2017-11-22:
Hi. I build this kernel, and notice that /proc/cpu/alignment is absent.
I need to enable SIGBUS signal to the user process performing the unaligned access.
Do I need to reconfigure kernel for /proc/cpu/alignment arrive, or there are another way to  change the alignment trap behavior?

pelwell, 2017-11-22:
/proc/cpu/alignment is present on builds using the Raspberry Pi defconfigs (bcm2709_defconfig and bcmrpi_defconfig). In order for the procfs entry to appear, CONFIG_PROC_FS must be defined, but otherwise it should be automatic for ARMs:
```
config ALIGNMENT_TRAP
	bool
	depends on CPU_CP15_MMU
	default y if !ARCH_EBSA110
```
Notice the default value.

Fullmetal8ender, 2017-11-22:
Does this works for aarch64 (rpi3)?
I create config with `make ARCH=arm64 bcmrpi3_defconfig O=../out` and there is no ALIGNMENT_TRAP.
Maybe aarch64 does not support CPU_CP15_MMU?

pelwell, 2017-11-22:
You should have mentioned that yours was a 64-bit build - alignment trap is not available on any arm64 kernel.

closed 2017-11-22 by pelwell

blucey, 2021-03-22:
For what it's worth, I ran into this same problem this evening.  Running a 64 bit kernel on a Pi 4.  I spent a bunch of time debugging it only to come to the same conclusions as this thread.  Not sure if anyone needs information on the specific line of code causing it, but I was able to narrow it down for that when running borg init after building with debug on and optimizations off.

Anyhow, pulling the latest xxhash.c and xxhash.h from the xxhash repo itself and dropping that on top of the ones in the borg repo seems to have resolved the issue for me.