raspberrypi/linux #2284: RPi3: APT over SSH returns

gazoo4, 2017-11-22:
We were using RPi2 devices with private APT repository for distribution of our custom software and it was working well. With the newly bought RPi3s we have the following issue unfortunately.

`user@client:~$ sudo apt-get update
Hit:1 http://mirrordirector.raspbian.org/raspbian stretch InRelease
Hit:2 http://archive.raspberrypi.org/debian stretch InRelease
Get:3 ssh://repo/home/user/repo stretch InRelease [1,458 B]
Ign:3 ssh://repo/home/user/repo stretch InRelease            
Get:4 ssh://repo/home/user/repo stretch Release [921 B]
Ign:4 ssh://repo/home/user/repo stretch Release
Ign:5 ssh://repo/home/user/repo stretch/main all Packages
Ign:6 ssh://repo/home/user/repo stretch/main armhf Packages
Ign:7 ssh://repo/home/user/repo stretch/main Translation-en_GB
Ign:8 ssh://repo/home/user/repo stretch/main Translation-en
Ign:5 ssh://repo/home/user/repo stretch/main all Packages
Ign:6 ssh://repo/home/user/repo stretch/main armhf Packages
Ign:7 ssh://repo/home/user/repo stretch/main Translation-en_GB
Ign:8 ssh://repo/home/user/repo stretch/main Translation-en
Ign:5 ssh://repo/home/user/repo stretch/main all Packages
Ign:6 ssh://repo/home/user/repo stretch/main armhf Packages
Ign:7 ssh://repo/home/user/repo stretch/main Translation-en_GB
Ign:8 ssh://repo/home/user/repo stretch/main Translation-en
Ign:5 ssh://repo/home/user/repo stretch/main all Packages
Get:6 ssh://repo/home/user/repo stretch/main armhf Packages [412 B]
Ign:6 ssh://repo/home/user/repo stretch/main armhf Packages
Ign:7 ssh://repo/home/user/repo stretch/main Translation-en_GB
Ign:8 ssh://repo/home/user/repo stretch/main Translation-en
Ign:5 ssh://repo/home/user/repo stretch/main all Packages
Ign:6 ssh://repo/home/user/repo stretch/main armhf Packages
Ign:7 ssh://repo/home/user/repo stretch/main Translation-en_GB
Ign:8 ssh://repo/home/user/repo stretch/main Translation-en
Ign:5 ssh://repo/home/user/repo stretch/main all Packages
Get:6 ssh://repo/home/user/repo stretch/main armhf Packages [560 B]
Err:6 ssh://repo/home/user/repo stretch/main armhf Packages
  Data socket timed out
Ign:7 ssh://repo/home/user/repo stretch/main Translation-en_GB
Reading package lists... Done
W: The repository 'ssh://repo/home/user/repo stretch Release' does not have a Release file.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: Failed to fetch ssh://repo/home/user/repo/dists/stretch/main/binary-armhf/Packages  Data socket timed out
E: Some index files failed to download. They have been ignored, or old ones used instead.`

However the missing file apt-get complained about is there and can be pulled easily:
`user@client:~$ sudo scp repo:/home/s/repo/dists/stretch/main/binary-armhf/Packages .
Packages        100%  560    12.0KB/s   00:00`

The repository is accessed via ssh. This is the content of the /etc/apt/sources.list.d/customsoft.list:
`deb ssh://repo/home/user/repo stretch main
#deb http://172.23.1.56:8080 stretch main`

The ssh configuration on the client side in ~root/.ssh/config is the following:
`Host repo
HostName 172.23.1.56
User user
Port 2222`

The keys are distributed and passwordless connection `ssh repo` works well. The file/folder structure has been built with `reprepro` and it looks the following way (to be sure I don't deal with a permission problem on my test box I did chmod -R 777 across the whole repo folder):
`user@server:~$ ls -ltR repo/
repo/:
total 32
drwxrwxrwx 2 user user     4096 Nov 21 12:10 db
drwxrwxrwx 3 user user     4096 Nov 21 12:10 dists
drwxrwxrwx 3 user user     4096 Nov 21 12:10 pool
drwxrwxrwx 2 user user     4096 Nov 21 12:10 conf
-rwxrwxrwx 1 user user     1749 Nov 21 12:08 customsoft.gpg.key

repo/db:
total 88
-rwxrwxrwx 1 user user 16384 Nov 21 12:10 checksums.db
-rwxrwxrwx 1 user user 16384 Nov 21 12:10 references.db
-rwxrwxrwx 1 user user    31 Nov 21 12:10 version
-rwxrwxrwx 1 user user 20480 Nov 21 12:10 release.caches.db
-rwxrwxrwx 1 user user 16384 Nov 21 12:10 packages.db
-rwxrwxrwx 1 user user 16384 Nov 21 12:10 contents.cache.db

repo/dists:
total 4
drwxrwxrwx 3 user user 4096 Nov 21 12:10 stretch

repo/dists/stretch:
total 16
-rwxrwxrwx 1 user user 1458 Nov 21 12:10 InRelease
-rwxrwxrwx 1 user user  488 Nov 21 12:10 Release.gpg
drwxrwxrwx 3 user user 4096 Nov 21 12:10 main
-rwxrwxrwx 1 user user  921 Nov 21 12:10 Release

repo/dists/stretch/main:
total 4
drwxrwxrwx 2 user user 4096 Nov 21 12:10 binary-armhf

repo/dists/stretch/main/binary-armhf:
total 12
-rwxrwxrwx 1 user user 560 Nov 21 12:10 Packages
-rwxrwxrwx 1 user user 412 Nov 21 12:10 Packages.gz
-rwxrwxrwx 1 user user 132 Nov 21 12:10 Release

repo/pool:
total 4
drwxrwxrwx 3 user user 4096 Nov 21 12:10 main

repo/pool/main:
total 4
drwxrwxrwx 3 user user 4096 Nov 21 12:10 c

repo/pool/main/c:
total 4
drwxrwxrwx 2 user user 4096 Nov 21 12:10 customsoft

repo/pool/main/c/customsoft:
total 956
-rwxrwxrwx 1 user user 975002 Nov 21 12:10 customsoft_0.1.1_armhf.deb

repo/conf:
total 8
-rwxrwxrwx 1 user user 170 Nov 21 12:10 distributions
-rwxrwxrwx 1 user user  44 Nov 21 11:36 options`

Also, I tried to host the repo in Apache (using the second entry in the /etc/apt/sources.list.d/customsoft.list shown above) to confirm the repository has the correct structure:
`user@server:~$ sudo cat /etc/apache2/sites-enabled/000-default.conf 
<VirtualHost *:80>
        ...
        DocumentRoot /home/user/repo
        ...
        <Directory /home/user/repo>
                Require all granted
                Options Indexes FollowSymLinks Multiviews
        </Directory>
</VirtualHost>`

This worked well:
`user@client:~$ sudo apt-get update
Hit:1 http://172.23.1.56:8080 stretch InRelease
Hit:2 http://archive.raspberrypi.org/debian stretch InRelease                                                    
Hit:3 http://mirrordirector.raspbian.org/raspbian stretch InRelease                                              
Reading package lists... Done`

Our PRD repo where we spotted the problem is hosted on RPi2, our testbox repo where we are able to reproduce the problem is on Debian 9 (but compiled naturally for armhf architecture).

The same problem has been reported in RPi user forum [https://www.raspberrypi.org/forums/viewtopic.php?t=196852](here).

Thank you for any guidance in advance.

pelwell, 2017-11-22:
Can you confirm the following:
1. The Pi is the server in the system.
2. The same image (same card, ideally) works in a Pi 2 but fails as described in a Pi 3.
3. The same client is used in both tests.

By the way, use three backticks (` ``` ` ) on a line of its own for block code quotes.

gazoo4, 2017-11-22:
Thanks for your quick reply.

1. Pi 3 is the client. The server is a Debian stretch.
2. I just did the following test: 
  - Pi 3 + 2017-09-07-raspbian-stretch-lite = failure as per the above
  - same device + same card + 2017-07-05-raspbian-jessie-lite + same config = working OK
3. see above

pelwell, 2017-11-22:
4. Are you using wired or wireless networking on the Pi 3?
5. Does the Pi 3 have an adequate power supply? After a failure, run `vcgencmd get_throttled` and report the output.

pelwell, 2017-11-22:
```
Reading package lists... Done
W: The repository 'ssh://repo/home/user/repo stretch Release' does not have a Release file.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: Failed to fetch ssh://repo/home/user/repo/dists/stretch/main/binary-armhf/Packages  Data socket timed out
E: Some index files failed to download. They have been ignored, or old ones used instead.
```
Can you confirm whether you see the same failure with the stretch image in a Pi 2?
Is the `/home/s/` vs. `/home/user/` difference in the output above significant?

This really doesn't sound like a kernel problem, and I have no experience of Debian packaging. @XECDesign may be able to shed some light, though.

gazoo4, 2017-11-22:
4. Wired networking.
5. I'm using the officially bundled power supply. I was running `vcgencmd get_throttled` in a loop every 1 second and storing the result in the file while `apt-get update` was running in parallel. The output was always `throttled=0x0`
`s` is the username for this test, for the sake of readability I've expanded that to `user` in my bugreport.

pelwell, 2017-11-22:
OK - until then we should assume that this is an OS difference rather than a platform difference, and that the Pi 2 will also fail.

gazoo4, 2017-11-22:
The issue has been  confirmed on Pi 2 running Stretch. So Stretch is the common denominator.

XECDesign, 2017-11-22:
Does the problem occur if the client is running Debian rather than Raspbian?

If the issue is inherited from Debian, it should be reported here: https://www.debian.org/Bugs/Reporting

gazoo4, 2017-11-23:
I was unable to reproduce the problem on Debian Stretch. The scenario worked well there (and on Jessie as well).
```
user@debian:~$ sudo apt -v
apt 1.4.8 (amd64)
```
```
user@raspberrypi:~$ sudo apt -v
apt 1.4.8 (armhf)
```
The issue appeared on version 1.4.7 as well.

XECDesign, 2017-11-27:
I am seeing this as well, but I'm having trouble connecting it to anything in the raspberrypi.org userland. I can dig a bit further when time allows.

gazoo4, 2017-12-04:
Just adding some information gathered on my end - on a clean raspbian jessie install on a client side I installed gcc-6 and g++-6 from stretch repo (with their dependencies) and the issue appeared on this jessie installation as well. For the list of all pulled packages and their versions please see the [apt.log](https://github.com/raspberrypi/linux/files/1528221/apt.log).

XECDesign, 2017-12-04:
That's useful information. Perhaps it's a libc issue then. Might make more sense to move this to https://github.com/RPi-Distro/repo/issues/

gazoo4, 2017-12-10:
Some more debugging information I collected today. I recorded the commands issued by apt during the ssh `apt-get update` session when accessing the repository (using [https://github.com/open-ch/log-user-session](log-user-session)). I found one subtle, but likely important difference between apt client on jessie ([jessie.log](https://github.com/raspberrypi/linux/files/1545973/jessie.log)) and apt client on stretch ([stretch.log](https://github.com/raspberrypi/linux/files/1545981/stretch.log)). In both cases they issue a dd command at the beginning:
`dd if=/home/user/repo/dists/jessie/InRelease bs=2048 skip=0 2> /dev/null`
However stretch issues the command with `skip=1`. What the man-page of dd says about this argument: `skip=N skip N ibs-sized blocks at start of input`.

I executed both cases manually from console and the stretch version failed to pull any data from the repository. Please find the result attached (including the STDERR output summarizing how much data was transferred) 
[dd_execution.log](https://github.com/raspberrypi/linux/files/1545984/dd_execution.log)
. Apparently the `skip=1` argument is causing the issues we have.

gazoo4, 2017-12-15:
If you'd like me to move the topic to https://github.com/RPi-Distro/repo/issues/ if you could point me to the directions explaining how please. Thank you.

pelwell, 2017-12-15:
There is no official way to move an issue to another repo. I suggest you create a new issue, copying the main points across and linking to this issue for the details, then close this one.

gazoo4, 2017-12-15:
As suggested above, let's follow this in the RPi-Distro/repo section.

closed 2017-12-15 by gazoo4