raspberrypi/linux #2285: Kernel oops in VC4 driver

jorisvergeer, 2017-11-23:
Hi,

I am doing some intensive work in QT5.9.1+webengine (I got the qt binaries from the next raspbian version and compiled qtwebengine myself). I am sure there are memory leaks in some point. 

But I don't think the kernel is allowed to do oopses, even when out of memory (or rather, could not allocate memory).

Anyway, I attached a kernel log with 3 oopses in it. 
I hope if someone with knowledge of the VC4 drivers can look into it.

[kernel.log](https://github.com/raspberrypi/linux/files/1498322/kernel.log)

jorisvergeer, 2017-11-23:
After updating the kernel to the latest update, only the first oops remains (about 14 sec into boot)

pelwell, 2017-11-24:
If the remaining "oops" is the WARNING in drm_handle_vblank then @anholt is aware and a future update should remove it. However, I'm happy that the situation being warned about only occurs during startup and is not harmful (it's caused by receiving an interrupt before one of the software components is configured).

JamesH65, 2017-12-04:
@pelwell OK to close then?

pelwell, 2017-12-04:
Traditionally it is the person who raises the issue that closes it, but fine by me.

closed 2017-12-04 by pelwell